micropython/micropython #6772: librtc missing

ehtnevets, 2021-01-15:
I'm building esp32 with esp idf V4. It seems the makefile is not building librtc thus I'm getting xtensa-esp32-elf/bin/ld: cannot find -lrtc.

Is there a flag I need to set?

psi-cmd, 2021-02-05:
Same problem.

dpgeorge, 2021-02-05:
It should build correctly.  You need to make sure you have the right IDF version with the submodules sync'd.  See the `tools/ci.sh` script `ci_esp32_idf4_setup` and `ci_esp32_idf4_build` functions for more info.

ehtnevets, 2021-02-06:
I forgot what I did but I managed to compile with the make-accepted idf 4x version. However, running the binary on Heltec Wifi Kit 32 has issue with Wifi (unreliable connection and super slow) and somewhat memory hoag. 

@dpgeorge , I tried your cmake fork with idf 4.2. I can't be happier. Instant wifi reconnect and around 10+K extra free ram available. I also managed to compile with idf 4.3, but it uses up almost all the ram and renders the binary unusable. 

For now I'm happy with the cmake fork with idf 4.2. Thanks for the fork.

dpgeorge, 2021-04-14:
CMake builds are now in master.

closed 2021-04-14 by dpgeorge