micropython/micropython #6773: uncaught exception in Timer(1) interrupt handler MemoryError:

Irvingao, 2021-01-15:
Hello, here is my problem:
`import pyb
import ps2
import encoder
from pyb import LED, Timer

from machine import I2C,Pin
from oled import SSD1306_I2C

def mainTimer_cb(cb):
    '''
    #函数功能：用于总体程序的逻辑定时功能
    '''
    global count
    LED(3).toggle()
    count += 1
    print(count)
    if (count%8) == 0: 
        LED(2).toggle() 
    elif (count%20) == 0:
        count = 0
    elif (count%5):
        oled_draw(count)

def oled_draw(text):
    oled.text(str(text),20,0)

if __name__ == '__main__':
    count = 0
    
    mainTimer = Timer(1, freq=2, callback=mainTimer_cb) 
    
    i2c = I2C(scl = Pin("X9"),sda = Pin("X10"),freq = 10000) 
    oled = SSD1306_I2C(128, 64, i2c) #创建oled对象

    oled.text("count:",0,0)
    oled.text("i love u!",20,40)
    oled.show()

    while True:
        i = 0`

when i ran this code in pyb, it always got the memory error. So  i wanna ask how can i write my main code in callback function? Because i can do this by C in STM32, just so confused to the error.

dlech, 2021-01-15:
See the docs about writing interrupt handlers: http://docs.micropython.org/en/latest/reference/isr_rules.html

Deleting the `print()` might fix it.

Irvingao, 2021-01-16:
I still can't figure it out by deleting the `print()`. What's more, i found  i got same error when i use oled in interrupt handlers:

```
def display(cb):
    '''
    函数功能：oled所有信息的显示功能
    '''
    oled.fill(0)
    # oled.text("count:",0,0)
    # text = str(count)
    # oled.text(text,50,0)
    # oled.show()
    # print(text)

if __name__ == '__main__':
    # 初始化
    count = 0
    mainTimer = Timer(1, freq=10, callback=mainTimer_cb) # 主程序逻辑控制定时器，频率为1000，每次计时1ms
    
    i2c = I2C(scl = Pin("X9"),sda = Pin("X10"),freq = 400000) #软件I2C
    oled = SSD1306_I2C(128, 64, i2c) #创建oled对象
    oledTimer = Timer(2, freq=10, callback=display)
```
So can i expand the interrupt area by some ways?? I think the error is caused by the memory out of limit, but my flash still have a lot space. So how can i move the interrupt area to the empty flash area?

peterhinch, 2021-01-16:
The problem is likely to be `oled_draw(count)` allocating RAM. The doc cited by @dlech explains this. The solution is to use `micropython.schedule` to implement a soft interrupt handler. Another possible approach is to use `uasyncio` rather than interrupts.

Please note that this is the MicroPython bug tracker. Support queries should be raised in [the forum](http://forum.micropython.org/index.php).