micropython/micropython #6777: MPZ arithmetic errors with DIG_SIZE == MPZ_DIG_SIZE

mipy123, 2021-01-17:
There are some (potentially dangerous) arithmetic errors often occurring with long 100% CPU delays with the Unix port defaults (cannot test other ports) when dealing with large numbers.

CPython:
`>>> (2**256 - 2**32)**2 // (2**256 - 2**32)
115792089237316195423570985008687907853269984665640564039457584007908834672640`

MicroPython 1.13 (takes approximately 1 minute):
`>>> (2**256 - 2**32)**2 // (2**256 - 2**32)
1766847064778390606685032887445983096815437806911733293014036620377194496`

MicroPython 1.13 (immediate, with DIG_SIZE = MPZ_DIG_SIZE - 1):
`>>> (2**256 - 2**32)**2 // (2**256 - 2**32)
115792089237316195423570985008687907853269984665640564039457584007908834672640`

Is it safe to change the default DIG_SIZE to MPZ_DIG_SIZE - 1?

dpgeorge, 2021-02-04:
Thanks for the report.  I can reproduce it.  This is indeed a serious bug, which seems to be an edge case in the long division.

A simpler reproduction, which is broken on unix (64-bit) and stm32 (32-bit, on a PYBD-SF6):
```
>>> (2**48-1)**2 // (2**48-1)
65535
```
True answer is 281474976710655

dpgeorge, 2021-02-04:
See #6844 for a fix that fixes the cases above.  The fix needs further testing.

commit 0a599385 (2021-02-08): py/mpz: Fix overflow of borrow in mpn_div.

For certain operands to mpn_div, the existing code path for
`DIG_SIZE == MPZ_DBL_DIG_SIZE / 2` had a bug in it where borrow could still
overflow in the `(x >= *n || *n - x <= borrow)` branch, ie
`borrow + x - (mpz_dbl_dig_t)*n` overflows the borrow variable.  In such
cases the subsequent right-shift of borrow would not bring in the overflow
bit, leading to an error in the result.  An example division that had
overflow when MPZ_DIG_SIZE = 16 is `(2 ** 48 - 1) ** 2 // (2 ** 48 - 1)`.

This is fixed in this commit by simplifying the code and handling th [... cut]
PR #6844 (2021-02-08): py/mpz: Fix overflow of borrow in mpn_div.
This fixes #6777.  But there might be a cleaner way to fix it, and there might also be additional overflow of the borrow that is not caught by this fix.

Edit: there is a cleaner fix, which is now applied in this PR.
closed 2021-02-08 by dpgeorge