micropython/micropython #6778: Needs tweak to build on ARM M1 (Apple Silicon)

jcw, 2021-01-17:
This platform is very new, here's the build problem I see:

```
mpy-cross (master)$ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/genhdr
GEN build/genhdr/mpversion.h
GEN build/genhdr/moduledefs.h
GEN build/genhdr/qstr.i.last
../py/persistentcode.c:857:2: error: mp_raw_code_save_file not implemented for this
      platform
```

Appending "` || defined (__arm64__)`" to the `#if` in line 831 fixes it.

After that, the rest is as on Intel MacOS:

```
$ ./micropython
MicroPython v1.13-268-gf7aafc062-dirty on 2021-01-17; darwin version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>>
```

dlech, 2021-01-17:
See #6775

jcw, 2021-01-17:
Cool, I'll close this. That didn't show up when I searched for "M1" in issues.

closed 2021-01-17 by jcw