micropython/micropython #6783: esp32: Audio encoder FLAC/MP3 wanted

emard, 2021-01-18:
HI!

I'd like if possible the audio compression encoder in micropythong.
FLAC encoder would be ideal, but MP3 is ok too.

I'd like to store audio from microphone to SD card, compressed
One example code in C is here

https://github.com/myvobot/esp32_mp3_encoder

UnexpectedMaker, 2021-02-03:
This is a feature request not an issue. Maybe ask for this on the forums?

jonnor, 2024-08-18:
Hi. This kind of functionality can - and should - be implemented outside of core MicroPython. For example, as user C modules or native C module. There is a discussion thread about this here: https://github.com/orgs/micropython/discussions/11307
Proposing to close this issue in favor of the discussion thread.

closed 2024-08-18 by robert-hh