micropython/micropython #6795: knock-off LAN8270 board on STM32F407VET6 black board

LouDnl, 2021-01-25:
I've been trying to get a knock-off LAN8270 board to work on my STM32F407VET6 black board.
I have this board `https://os.mbed.com/users/hudakz/code/STM32F407VET6_Hello/`
Which I have compiled MicroPython for using this config `https://github.com/mcauser/BLACK_F407VE`

MicroPython works just fine but I cannot get the LAN8270 board to enable.
The network.LAN module doesn't take an arguments on STM32 and when I try to enable the network this is what I get:
```
>>> import network
>>> l = network.LAN()
>>> l.active(1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 110] ETIMEDOUT
```
I checked if the LAN8270 board isn't faulty by connecting it to an ESP32 MicroPython board and it works like it should.
```
>>> import network
>>> from machine import Pin
>>> lan = network.LAN(mdc=Pin(16), mdio=Pin(17), power=None, id=None, phy_addr=1, phy_type=network.PHY_LAN8720)
>>> lan.active(True)
True
>>> lan.ifconfig()
('192.168.0.107', '255.255.255.0', '192.168.0.1', '192.168.0.1')
```

Checked config files, pin definitions and everything else I could find. Can't seem to find an answer on how to fix it.
Any tips on what I could be doing wrong?

Reference: https://github.com/mcauser/BLACK_F407VE/issues/6#issuecomment-766772522

Meekdai, 2021-02-25:
I got the same problem on STM32F407ZGT6 board, but on STM32H7 board it works well.

LouDnl, 2021-02-26:
> I got the same problem on STM32F407ZGT6 board, but on STM32H7 board it works well.

Good to see that I'm not the only one, but there is no solution yet.

wangshujun-tj, 2022-04-06:
You can remove the pull-up resistance of the 10 pin of the lan8720. Most of the purchased modules have the default phyaddr set to 1407 and the MPY firmware needs to be 0 by default

I'm trying to modify the firmware and add the parameter input of phyaddr

ukicomputers, 2024-01-17:
Check PR #13422. There was invalid clock config.

roizano, 2024-03-31:
> I got the same problem on STM32F407ZGT6 board, but on STM32H7 board it works well.

Can you share configure on stm32h7?
How you make it work with LAN8720?

robert-hh, 2024-03-31:
Below is a set of Pins I used with a STM32F750 and STM32F743 to test LAN8720 support :
```

// Ethernet via RMII
#define MICROPY_HW_ETH_MDC          (pin_C1)
#define MICROPY_HW_ETH_MDIO         (pin_A2)
#define MICROPY_HW_ETH_RMII_REF_CLK (pin_A1)
#define MICROPY_HW_ETH_RMII_CRS_DV  (pin_A7)
#define MICROPY_HW_ETH_RMII_RXD0    (pin_C4)
#define MICROPY_HW_ETH_RMII_RXD1    (pin_C5)
#define MICROPY_HW_ETH_RMII_TX_EN   (pin_B11)
#define MICROPY_HW_ETH_RMII_TXD0    (pin_B12)
#define MICROPY_HW_ETH_RMII_TXD1    (pin_B13)
```
The python snippet to start it is:
```
import network, time
lan=network.LAN(phy_addr=1, phy_type=network.LAN.PHY_LAN8720)
lan.active(True)
while not lan.isconnected():
    time.sleep_ms(50)
    print(".", end="")
print("\r\n", lan.ifconfig())
```

roizano, 2024-04-01:
> Below is a set of Pins I used with a STM32F750 and STM32F743 to test LAN8720 support :
> 
> ```
> 
> // Ethernet via RMII
> #define MICROPY_HW_ETH_MDC          (pin_C1)
> #define MICROPY_HW_ETH_MDIO         (pin_A2)
> #define MICROPY_HW_ETH_RMII_REF_CLK (pin_A1)
> #define MICROPY_HW_ETH_RMII_CRS_DV  (pin_A7)
> #define MICROPY_HW_ETH_RMII_RXD0    (pin_C4)
> #define MICROPY_HW_ETH_RMII_RXD1    (pin_C5)
> #define MICROPY_HW_ETH_RMII_TX_EN   (pin_B11)
> #define MICROPY_HW_ETH_RMII_TXD0    (pin_B12)
> #define MICROPY_HW_ETH_RMII_TXD1    (pin_B13)
> ```
> 
> These pin have to be connected to the respective pins of the LAN8270. The Python snippet to start it is:
> 
> ```
> import network, time
> lan=network.LAN(phy_addr=1, phy_type=network.LAN.PHY_LAN8720)
> lan.active(True)
> while not lan.isconnected():
>     time.sleep_ms(50)
>     print(".", end="")
> print("\r\n", lan.ifconfig())
> ```

Do you have any special HW requirements for the LAN8720? any resistor to remove? I have the sample from Waveshare.
Would you have a full board setup somewhere in GitHub so I could compare it to what I compiled from?

robert-hh, 2024-04-01:
I use a Sparkfun breakout as well. Maybe your board uses a different phy_addr. You can try phy_addr = 0. Besides that no resistors were added or changed. Attached are the board files for the WEACT STM32H743 board I used. They are mostly made by Matt Trentini. I just added the Ethernet part.

[WEACT_STM32H743.zip](https://github.com/micropython/micropython/files/14824789/WEACT_STM32H743.zip)

roizano, 2024-04-02:
> I use a Sparkfun breakout as well. Maybe your board uses a different phy_addr. You can try phy_addr = 0. Besides that no resistors were added or changed. Attached are the board files for the WEACT [...]

Hi Robert,
thank you for your help!
finally I managed to get IP addresses: ('0.0.0.0', '255.255.255.0', '192.168.0.1', '8.8.8.8').. Is that right?
I tried to modify the IP with:
```
lan.ifconfig(('192.168.0.4', '255.255.255.0', '192.168.0.1', '8.8.8.8'))
```
then I run again the command:
```
>>> lan.ifconfig(('192.168.0.4', '255.255.255.0', '192.168.0.1', '8.8.8.8'))
>>> print("\r\n", lan.ifconfig())

 ('192.168.0.4', '255.255.255.0', '192.168.0.1', '8.8.8.8')
>>>

```
But when I ping it from my PC no reply:
```
C:\Users\koko1>ping 192.168.0.4

Pinging 192.168.0.4 with 32 bytes of data:
Request timed out.
Request timed out.
Request timed out.
Request timed out.

Ping statistics for 192.168.0.4:
    Packets: Sent = 4, Received = 0, Lost = 4 (100% loss),

C:\Users\koko1>
```
What am I doing wrong?

ukicomputers, 2024-04-02:
Hi @roizano!

If the you got as defualt address is `0.0.0.0`, that means some communication is between MCU and LAN chip. Could you write here clock rates for your MCU, and what lights you see on ethernet port (I suppose green)? That same problem happened to me, and I sucessfully fixed it (but MPY did not want to accept PR, just note for others, its not that problem with MCU not giving proper clock to LAN, probably, not confirmed).

Thanks!

robert-hh, 2024-04-02:
> '0.0.0.0', '255.255.255.0', '192.168.0.1', '8.8.8.8').. Is that right?

These are just the default settings of the driver, and if you change them, the chaged values are reported. That does not indicate the the interface is active.
You can check the LEDs at the RJ45 jack. When you power up the LAN8720 with Ethernet wire attached, they should show activity. When you then start the MicroPython LAN interface with nic.active(True), the LEDs should turn off for a second or two. If this does not happen, then there is a communication problem at the MDIO interface. With a simple logic analyzer you can check the MDC and MDIO signals. MDC must be below 2.5MHz.
If you have an oscilloscope, you cam as well check if the RefClk signal is present (25MHz). RefClk is generated by the LAN8720 interface and supplied to the MCU. The TX/RX wiring runs at ~50Mhz and is pretty sensitive. So use short good quality wires.

roizano, 2024-04-02:
> Hi @roizano!
> 
> If the you got as defualt address is `0.0.0.0`, that means some communication is between MCU and LAN chip, but something other is wrong (could you just post what `lan.status()` returns, and also what `lan.ifconfig("dhcp")` returns). Could you also write here clock rates for your MCU, and what lights you see on ethernet port (I suppose green)? That same problem happened to me, and I sucessfully fixed it (but MPY did not want to accept PR, just note for others, its not that problem with MCU not giving proper clock to LAN, probably, not confirmed).
> 
> Thanks!

Hi,
see what I got:
```
>>> lan.status()
2
>>> lan.ifconfig("dhcp")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: timeout waiting for DHCP to get IP address
>>>
```
I used clock configuration:
```
// Clock config
#define MICROPY_HW_CLK_PLLM (5)
#define MICROPY_HW_CLK_PLLN (160)
#define MICROPY_HW_CLK_PLLP (2)
#define MICROPY_HW_CLK_PLLQ (4)
#define MICROPY_HW_CLK_PLLR (2)
#define MICROPY_HW_CLK_PLLVCI (RCC_PLL1VCIRANGE_1)
#define MICROPY_HW_CLK_PLLVCO (RCC_PLL1VCOWIDE)
#define MICROPY_HW_CLK_PLLFRAC (0)

#define MICROPY_HW_CLK_PLL3M (25)
#define MICROPY_HW_CLK_PLL3N (240)
#define MICROPY_HW_CLK_PLL3P (2)
#define MICROPY_HW_CLK_PLL3Q (5)
#define MICROPY_HW_CLK_PLL3R (2)
#define MICROPY_HW_CLK_PLL3VCI (RCC_PLL3VCIRANGE_1)
#define MICROPY_HW_CLK_PLL3VCO (RCC_PLL3VCOWIDE)
#define MICROPY_HW_CLK_PLL3FRAC (0)

// Bus clock divider values
#define MICROPY_HW_CLK_AHB_DIV          (RCC_HCLK_DIV2)
#define MICROPY_HW_CLK_APB1_DIV         (RCC_APB1_DIV2)
#define MICROPY_HW_CLK_APB2_DIV         (RCC_APB2_DIV2)
#define MICROPY_HW_CLK_APB3_DIV         (RCC_APB3_DIV2)
#define MICROPY_HW_CLK_APB4_DIV         (RCC_APB4_DIV2)

// 32kHz crystal for RTC
#define MICROPY_HW_RTC_USE_LSE (1)
#define MICROPY_HW_RTC_USE_US  (0)
```
didn't work... and I tried also:
```
// The board has an 25MHz HSE, the following gives 480MHz CPU speed
#define MICROPY_HW_CLK_PLLM         (5)
#define MICROPY_HW_CLK_PLLN         (192)
#define MICROPY_HW_CLK_PLLP         (2)
#define MICROPY_HW_CLK_PLLQ         (4)
#define MICROPY_HW_CLK_PLLR         (2)
#define MICROPY_HW_CLK_PLLVCI       (RCC_PLL1VCIRANGE_1)
#define MICROPY_HW_CLK_PLLVCO       (RCC_PLL1VCOWIDE)
#define MICROPY_HW_CLK_PLLFRAC      (0)

// The USB clock is set using PLL3
#define MICROPY_HW_CLK_PLL3M        (5)
#define MICROPY_HW_CLK_PLL3N        (48)
#define MICROPY_HW_CLK_PLL3P        (2)
#define MICROPY_HW_CLK_PLL3Q        (5)
#define MICROPY_HW_CLK_PLL3R        (2)
#define MICROPY_HW_CLK_PLL3VCI      (RCC_PLL3VCIRANGE_1)
#define MICROPY_HW_CLK_PLL3VCO      (RCC_PLL3VCOWIDE)
#define MICROPY_HW_CLK_PLL3FRAC     (0)
```
also no success ...

roizano, 2024-04-02:
> These are just the default settings of the driver, and if you change them, the chaged values are reported. That does not indicate the the interface is active. You can check the LEDs at the RJ45 [...]

Hi Robert,
only when I run the lan.active(True) the LED turn off for one second and then orange is working all the time and green kind of blinking but nothing rather then that ...

ukicomputers, 2024-04-02:
Ok, as I expected. I'm not expert in H7 series, but I will send correct clock config if i get hands on it. Could you also measure voltage DIRECTLY on LAN chip on pin XTAL1? Btw, this is my schematic for using LAN chip with my devboard:[image]
Just ignore there PA8, it is connected to PA1.

ukicomputers, 2024-04-02:
Very worth looking #13406

robert-hh, 2024-04-02:
@roizano Which board do you use?

roizano, 2024-04-04:
> @roizano Which board do you use?

Hi,
I'm using STM32H743 from WeAct Studio.

The clock is as configured in stm32h7xx_hal.h
```
// Oscillator values in Hz
#define HSE_VALUE (25000000)
#define LSE_VALUE (32768)
#define EXTERNAL_CLOCK_VALUE (12288000)

```
the module of LAN8720 is from :

http://www.waveshare.com/wiki/LAN8720%20ETH%20Board

ukicomputers, 2024-04-04:
After all things, seems like it's not problem to the clock, because link to board sent by @roizano contains an internal 50MHz clock source (on board). Left issues:
- Invalid clock config for STM32H7 (my personal recommendation is to download [NECTO Studio](http://mikroe.com/necto) `6.1x` and for in it there is tested clock configuration for normal usage, ETHERNET, USB and for CAN BUS for almost any MCU from ST to Texas Instruments, and also, for every MCU that I tried MPY with (currently only ST F4x and F7x series and some PIC18), that config from IDE worked from first try, and this is not any ad even if I was MikroE intern, I really like it)
- Maybe your board is faulty (as @robert-hh said, you can check REFCLK pin, for example, maybe crystal is not working as expected)
- Or you got Chinese clone (first thing that Waveshare's website says)

roizano, 2024-04-06:
> After all things, seems like it's not problem to the clock, because link to board sent by @roizano contains an internal 50MHz clock source (on board). Left issues:
>
> * Invalid clock config for [...]

Hi @ukicomputers ,
The same HW worked for me in the STM32CubeIDE and ping me without any issues.
@robert-hh Maybe The issue is in the PC( Do I need to configure it manually or firewall configuration) ? what SW do you have in the PC side?  How you checked this?

robert-hh, 2024-04-06:
I have the board, PC and router connected to a switch. The router runs a DHPC service.
You can as well configure board and PC manually. You only have to take care, that both are configured to the same UP address range, like 192.168.0.x with netmask 255.255.255.0.
For basic communication not specific software is needed. Ping should work.

Does the green LED on the interface show activity? If yes, then the LAN wiring works at the lower protocol levels.

roizano, 2024-04-06:
> I have the board, PC and router connected to a switch. The router runs a DHPC service. You can as well configure board and PC manually. You only have to take care, that both are configured to the [...]

The green LED blinks for a second and then off completely .

I will buy new STM32H743VIT6 from Weact studio as this might have issues.

robert-hh, 2024-04-06:
That configuration is right:
`lan=network.LAN(phy_addr=1, phy_type=network.LAN.PHY_LAN8720)`

I have the same EThernet breakout, with RefClk being supplied from the LAN8720. If the wiring is right, the green LED should start flickering as soon as there is power to the LAN8720. Green LED means LAN activity.

roizano, 2024-04-06:
> That configuration is right: `lan=network.LAN(phy_addr=1, phy_type=network.LAN.PHY_LAN8720)`
> 
> I have the same EThernet breakout, with RefClk being supplied from the LAN8720. If the wiring is right, the green LED should start flickering as soon as there is power to the LAN8720. Green LED means LAN activity. Orange LED on indicates the network speed. How do you connect the LAN8720 to your Ethernet?

This is the setup:

[image]

robert-hh, 2024-04-06:
So you have a direct connection to the PC. Then there ist most likely not DHCP, and you have to set a manual IP address for the LAN8720. You have to try, whether lan.ifconfig() is to be called before or after lan.active("True").

lan.ifconfig(("192.168.0.99", "255.255.255.0", "192.168.0.1",  "8.8.8.8"))

With the IP address manually set, lan.status() should be 3. Then you should be able to "ping" the LAN8720 from the PC.  You should not need a cross-cable.

roizano, 2024-04-06:
> So you have a direct connection to the PC. Then there is most likely no DHCP server active, and you have to set a manual IP address for the LAN8720. You have to try, whether lan.ifconfig() is to be called before or after lan.active("True").
> 
> lan.ifconfig(("192.168.0.99", "255.255.255.0", "192.168.0.1", "8.8.8.8"))
> 
> With the IP address manually set, lan.status() should be 3. Then you should be able to "ping" the LAN8720 from the PC. You should not need a cross-cable.

@robert-hh 
this is what I did:
```
>>> import network, time
>>> lan=network.LAN(phy_addr=1, phy_type=network.LAN.PHY_LAN8720)
>>> lan.active(True)
>>> lan.ifconfig(('192.168.0.4', '255.255.255.0', '192.168.0.1', '8.8.8.8'))
>>> lan.isconnected()
True
>>> lan.ifconfig()
('192.168.0.4', '255.255.255.0', '192.168.0.1', '8.8.8.8')
>>> lan.status()
3
>>>
```
when I ping:
```
C:\>ping 192.168.0.99

Pinging 192.168.0.99 with 32 bytes of data:
Reply from 192.168.0.100: Destination host unreachable.
Request timed out.
Request timed out.
Request timed out.

Ping statistics for 192.168.0.99:
    Packets: Sent = 4, Received = 1, Lost = 3 (75% loss),
```
over Wireshark I can see that I'm trying to check if there is respond from 192.168.0.99 but no respond...
BTW, The green LED is working only when I'm pinging

roizano, 2024-04-09:
@robert-hh  - until I will get my new STM32H743VIT6.
I found that I have STM32H750VB  with 128Kb memory.
I compiled it and found that there are two bins that need to be flashed.
Do you know how to flash them?
[image]

robert-hh, 2024-04-09:
Actually I used the same build process and board files as for the H743 receiving a single firmware.bin file with a size of 550k. For flashing I just put the board into bootloader mode by pushing the boot0 switch and then reset, and used the 
`make BOARD=<board_name>  USE_PYDFU=0 deploy` target and dfu-util as the downloader. Within the REPL, the board will call itself H743, but that is just the name given in the mpconfigboard.h file.

roizano, 2024-04-10:
> Actually I used the same build process and board files as for the H743 receiving a single firmware.bin file with a size of 550k. For flashing I just put the board into bootloader mode by pushing the boot0 switch and then reset, and used the `make BOARD=<board_name> USE_PYDFU=0 deploy` target and dfu-util as the downloader. Within the REPL, the board will call itself H743, but that is just the name given in the mpconfigboard.h file. The board I have is from WEACT with a H750VBT6 chip. Edit: Reading the data sheets I am quite surprised that the H750VB board works, It should have only 128kB of flash at address 0x8000000. But it seems to have more, like 1MB. Trying to access memory bank 2 causes a hard reset.

Hi @robert-hh ,
My BIN and DFU files are 397kB and when I'm flashing them using DFU the board not responding at all . 
when pressing reset the board not respond but when I'm pressing one time on the BOOT0 it will go automatically to DFU mode.

robert-hh, 2024-04-10:
By data sheet the H750 requires a different flash layout than the H743. The fact that the H743 firmware runs on my H750 is pure chance and cannot be expected.
The H750VB has only 128k flash at address 0x8000000, opposite to the H743VI which has 1MB. 

So for the H750 the code has to be split in two parts. Some of that, especially the boot up code, at address 0x8000000, the other parts at address 0x9000000 in the external QSPI flash. The flash file system size has to be decreased, the startup code has to be aligned to enable the external flash for code execution and the flash file system has to be decreased.

So better wait for the new H743.

roizano, 2024-04-13:
> By data sheet the H750 requires a different flash layout than the H743. The fact that the H743 firmware runs on my H750 is pure chance and cannot be expected. The H750VB has only 128k flash at [...]

Hi @robert-hh ,
The new product is finally here and all working!
Thank you!

roizano, 2024-04-13:
@robert-hh  - The new product is here and all is working :) I dropped the usage of STM32H750

robert-hh, 2024-04-13:
That sounds good. It would be interesting what is wrong with the other board. You could try the Ethernet Pins on that board by using them as GPIO output and see, if toggling them appears at the respective board connector pins. If that all works, the MCU may be defective.

jonnor, 2024-08-18:
Sounds like the problem got fixed. Proposing to close this issue :)

closed 2024-08-18 by robert-hh