micropython/micropython #6798: windows port: no module named ussl

bammi, 2021-01-26:
I built windows port from source using visual studio 2019, full success during the build, can execute micropython, interact with REPL passes tests all per README.s here. Went through and tools/bootstrap_upip.sh successfully using cygwin on windows. If i try to run upip now, i always get 
 /cygdrive/D/bammi/src/micropython/ports/windows/micropython.exe -m upip.py -help
Traceback (most recent call last):
  File "upip.py", line 109, in <module>
ImportError: no module named 'ussl'

I cant see anthing resembling ussl in the library either.
Does ussl not exist in the windows port.

thanks
bammi

stinos, 2021-01-26:
ussl is not currently implemented for the windows port indeed

Question: is the use of bootstrap_upip.sh suggested in the documentation or so? Asking because upip.py and upip_tarfile.py are actually in the same directory so instead of installing Cygwin just to run a bash file you could just have copied upip*.py or even used them directly.

Note: this issue tracker is mostly for issues, more general questions like this belong on https://forum.micropython.org. Where people can also suggest solutions.

bammi, 2021-01-26:
Thank you for the confirmation 

I have Cygwin installed and use it daily, so I didn’t install it for this purpose

I will discuss on uPython forum, sounded like a issue to me since something as basic/ central like upip will not work without ussl   As suggested will post in forums there and see what response I receive. 
Thanks again 
Cheers,
Bammi

closed 2021-01-27 by bammi