micropython/micropython #6799: Add means for accessing locals when MicroPython is built with sys.settrace

aivarannamaa, 2021-01-26:
`sys.settrace` is a great step towards having a usable MicroPython debugger, but as long as it's not possible to query local variables, people can't replace their print-debugging with a debugger based on `sys.settrace`.

My proposal is to somehow expose locals when `sys.settrace` is enabled. It doesn't need to be a `Dict[str, Any]`. If the exact rules are published for mapping local names to positions in an array, the debugger can still present a conventional view to the user. It's alright if some locals (eg. created by constant expressions) don't end up in this array at all, as long as the rules are clear.

peterhinch, 2021-01-28:
A debugger would be a huge step forward for MicroPython.

stinos, 2021-01-28:
Definitely, but I looked into retrieving locals the other day and that didn't look like a trivial job because of how that part of the runtime is optimized.  Could be that I missed something though.

wolfc01, 2021-02-03:
> A debugger would be a huge step forward for MicroPython.

Yes, I agree. The first baby steps of a poor mans debugger can be found in https://github.com/wolfc01/micropython_debugger.

JohannesMaierhofer, 2023-11-08:
Any updates on this? Even the most simple stop and step line by line often helps a lot.

andrewleech, 2023-11-08:
I've got a basic port of pdb working https://github.com/micropython/micropython-lib/pull/499 which can be used over the terminal. It doesn't really provide access to locals because the names of them is not stored at all in the bytecode. It would be possible to expose them by index though which I've thought of doing (ie. local variable "0", "1", "2"... )

I haven't progressed it any time recently though, got distracted looking into how I'd integrate it into vscode :-D