micropython/micropython #6802: Safari on iOS and MacOS would not open page served on AP socket ESP8266

m-stefanski, 2021-01-26:
I am trying to host a page using ESP8266:

```
ap = network.WLAN(network.AP_IF)
ap.active(True)
ap.config(essid=WIFI_SSID, password=WIFI_PASSWORD)

while ap.active() == False:
  pass

def web_page():
  html = '<html><head></head><body><h1>Hello world!</h1></body></html>'
  return html

s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.bind(('', 80))
s.listen(5)

while True:
  conn, addr = s.accept()
  print('Got a connection from %s' % str(addr))
  request = conn.recv(1024)
  print('Content = %s' % str(request))
  response = web_page()
  conn.send(response)
  conn.close()
```

I can successfully connect to the WIFI, and using any other browser on MacOS than Safari - I can connect and see the page. However, on Safari (both iOS and macOS) I get "Safari cannot open the page because the network connection was lost". 

Requests are showing up in log (3 requests each refresh):

```
Got a connection from ('192.168.4.2', 52934)
Content = b'GET / HTTP/1.1\r\nHost: 192.168.4.1\r\nUpgrade-Insecure-Requests: 1\r\nAccept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8\r\nUser-Agent: Mozilla/5.0 (iPhone; CPU iPhone OS 14_3 like Mac OS X) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0.2 Mobile/15E148 Safari/604.1\r\nAccept-Language: en-gb\r\nAccept-Encoding: gzip, deflate\r\nConnection: keep-alive\r\n\r\n'
```

Curious thing, is that DNS is set to address 208.67.222.222', which I am not able to change:

```
ap.ifconfig()
('192.168.4.1', '255.255.255.0', '192.168.4.1', '208.67.222.222')
```

I was reading that safari might act up when DNS are not set properly. But I am not sure this is the real issue. 

Does anybody have he same problem?

fabseeeee, 2022-11-28:
i have the same problem ... with google chrome on windows :(

what's your solution?

jonnor, 2024-09-13:
This might be due to very old TLS configuration being used. But more details would be needed to reproduce and debug this.

dpgeorge, 2024-09-16:
> This might be due to very old TLS configuration being used.

Yes, that could be it.  And maybe related to the browser issuing `Upgrade-Insecure-Requests: 1\r\n`.

I suggest switching to an ESP32 board.

closed 2024-09-16 by dpgeorge