micropython/micropython #6803: uasyncio problem on Raspberry Pi pico

peterhinch, 2021-01-27:
This script works, but only if the "Waiting" print statement exists. If you comment it out, the script hangs until you press ctrl-c, when it prints "Done". 
```python
import uasyncio as asyncio

done = False

async def test():
    global done
    await asyncio.sleep(5)
    done = True

async def report():
    asyncio.create_task(test())
    while not done:
        await asyncio.sleep(1)
        print('Waiting')
    print('Done')

asyncio.run(report())
```

dpgeorge, 2021-01-27:
I cannot reproduce.  Deleting the "Waiting" line, it does exit after 5 seconds.

robert-hh, 2021-01-27:
Neither me. Code stops after 5 sec. I am using the Jan 23 code compiled from the PR, pasted to a file, uploaded to the device and started with import.

peterhinch, 2021-01-27:
That is odd. I've repeated the test using **miniterm** in place of **rshell** with the same results. The script runs both on a Pyboard and on ESP32, on the same PC using **rshell**.

Any thoughts? Because I'm baffled at the moment.
> MicroPython de1239b6a-dirty on 2021-01-27; Raspberry Pi Pico with RP2040

Firmware built this morning. The only change is to **_boot.py** to mount the filesystem on **/flash** - trying to diagnose a separate **rshell** issue.

peterhinch, 2021-01-27:
I've now built from the PR with the same outcome. In all these tests I executed using ctrl-e paste.

dpgeorge, 2021-01-27:
Very strange indeed... I have no suggestions right now.  Maybe we can wait until the PR is merged and there are official daily builds of the rp2 port.

peterhinch, 2021-01-27:
On my daily walk the following _reductio ad absurdum_ test occurred to me.
```python
MicroPython v1.13-291-gf4aef253a-dirty on 2021-01-27; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>> 
>>> import time
>>> time.sleep(5)
```
The outcome was that after 5 seconds no prompt appeared. I waited another 5 seconds then hit Enter. A prompt then appeared.

I am clearly having communications problems and perhaps this accounts for [my rshell issue](https://forum.micropython.org/viewtopic.php?f=21&t=9646&start=10#p54067). Puzzling as these PC's are rock solid with Pyboards, ESP32 and ESP8266.

But I'm happy to wait.

peterhinch, 2021-01-27:
I've now tried my last two PC's. On these (ancient desktops running fairly recent versions of Mint) the `time.sleep(5)` test worked - i.e. produced a prompt after 5 seconds. Odd that it's PC/Linux version dependent.

None of my four machines can reliably talk to the filesystem via rshell: access is essentially read-only. In the course of testing I must have briefly got write access as I managed to create a file and edit it once. Now, on any machine, I can edit it but trying to save changes causes a hang. I guess that's one for Dave, who's waiting on Pico hardware.

I'm out of ideas here, so I'll abandon the Pico for the moment.

robert-hh, 2021-01-27:
My state is the same as yours. rhshell works only partially. mpr works fine.

dpgeorge, 2021-01-28:
> The outcome was that after 5 seconds no prompt appeared. I waited another 5 seconds then hit Enter. A prompt then appeared.

Hmm, this sounds like a USB VCP issue: it could be that a USB packet is being lost, or some buffers are not being flushed.  There are 4 erratas for the RP2040 related to USB which might be related to this.

Can you please do the following test:
- run the import time, time.sleep(5) test
- after 5 seconds of nothing happening, press "a"; if nothing happens then press "enter"
- paste the full results here

peterhinch, 2021-01-28:
Nothing very interesting happened. As soon as I pressed the "a" key it issued a prompt followed by "a".
```python
MPY: soft reboot
MicroPython v1.13-291-gf4aef253a-dirty on 2021-01-27; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>> import time
>>> time.sleep(5)
>>> a
```

mcauser, 2021-01-28:
I came across something similar with `time.sleep()` in circuitpython on a Wio terminal (samd51)
https://github.com/adafruit/circuitpython/issues/3829

I thought `time.sleep()` was causing the board to hang, but instead, just wasn't showing the next `>>>` after it ran.
After the sleep duration, pressing `a` (or `down arrow`) makes the `>>>` appear on the next line.

kadamski, 2021-02-18:
I'm having similar issue on my system. I'm trying to run the extmod/utime_res.py test but it always hangs. If I change the TEST_TIME from 2500 to 2000, it works OK. I did some tests and: when running:
```
utime.sleep_ms(2490); print("one")
```
I'm getting the output "one" and the ">>>" prompt after those 2.5s but if I run:
```
utime.sleep_ms(2500); print("one")
```
The console "hangs" (no matter how long I wait) and I only get "one" and a problem after I press some key on the keyboard.

The consol "hangs" also on this code:
```
for i in range (1, 11):
    utime.sleep_ms(250)
```

but works fine when i add some print in the loop (in this case everything works fine even if i do hundreds of iterations):

```
for i in range (1, 11):
    utime.sleep_ms(250)
    print(i)
```

peterhinch, 2021-03-31:
In my case this PR has indeed fixed the problems.

closed 2021-03-31 by peterhinch

lurch, 2021-03-31:
That's great news, sorry it took us a while to get round to fixing it.