micropython/micropython #6806: Why the example changes CC during the build?

PeterBorisenko, 2021-01-27:
Hello!
My goal is to launch micropython upon the freertos application.
The main purpose of this is to provide to the customer a simple way to extend application.
So I have started from compiling embedded example inside the micropython tree.

Steps to reproduce:
1. add CC definition to the makefile:
```diff
@@ -1,8 +1,10 @@
 MPTOP = ../..
 CFLAGS = -std=c99 -I. -I$(MPTOP) -DNO_QSTR
 LDFLAGS = -L.
 
+CC = arm-none-eabi-gcc
+
```
2. change Makefile.upylib:
```diff
@@ -131,19 +131,22 @@ MAIN_C = main.c
 SRC_C = $(addprefix ports/unix/,\
 	$(MAIN_C) \
 	gccollect.c \
 	unix_mphal.c \
 	input.c \
-	file.c \
 	modmachine.c \
 	modos.c \
 	moduselect.c \
 	alloc.c \
 	coverage.c \
 	fatfs_port.c \
 	$(SRC_MOD) \
 	)
+
+SRC_C += $(addprefix extmod/,\
+    vfs_posix_file.c \
+    )
```
3. fix the definition of `mp_import_stat()` in the `hello-embed.c` file:
```diff
@@ -63,11 +63,11 @@ int main() {
     if (execute_from_str(str)) {
         printf("Error\n");
     }
 }
 
-uint mp_import_stat(const char *path) {
+mp_import_stat_t mp_import_stat(const char *path) {
     return MP_IMPORT_STAT_NO_EXIST;
 }
```
4. run `make`

The output:
```console
Traceback (most recent call last):
  File "X:\PROJECTS\Other\libs&test\uPython\py\makeqstrdefs.py", line 188, in <module>
In file included from ./mpconfigport.h:1,
                 from ../../py/mpconfig.h:62,
                 from ../../py/obj.h:31,
                 from ../../py/objstr.h:29,
                 from ../../py/objstringio.c:31:
./mpconfigport_minimal.h:128:10: fatal error: alloca.h: No such file or directory
 #include <alloca.h>
          ^~~~~~~~~~
    preprocess()
  File "X:\PROJECTS\Other\libs&test\uPython\py\makeqstrdefs.py", line 62, in preprocess
compilation terminated.
    for output in p.imap(pp(flags), chunks):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.496.0_x64__qbz5n2kfra8p0\lib\multiprocessing\pool.py", line 870, in next
    raise value
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.496.0_x64__qbz5n2kfra8p0\lib\multiprocessing\pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "X:\PROJECTS\Other\libs&test\uPython\py\makeqstrdefs.py", line 46, in run
    return subprocess.check_output(args.pp + flags + files)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.496.0_x64__qbz5n2kfra8p0\lib\subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.496.0_x64__qbz5n2kfra8p0\lib\subprocess.py", line 524, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['gcc', '-E', '-I.', '-I..', '-I../..', '-I../../ports/unix', '-Ibuild', '-Wall', '-Werror', '-Wpointer-arith', '-Wuninitialized', '-std=gnu99', '-DUNIX', '-DFFCONF_H="lib/oofatfs/ffconf.h"', '-Os', '-U', '_FORTIFY_SOURCE', '-DNO_QSTR', '../../py/mpstate.c', '../../py/malloc.c', '../../py/gc.c', '../../py/pystack.c', '../../py/qstr.c', '../../py/vstr.c', '../../py/mpprint.c', '../../py/unicode.c', '../../py/mpz.c', '../../py/reader.c', '../../py/lexer.c', '../../py/parse.c', '../../py/scope.c', '../../py/compile.c', '../../py/emitcommon.c', '../../py/emitbc.c', '../../py/asmbase.c', '../../py/asmx64.c', '../../py/emitnx64.c', '../../py/asmx86.c', '../../py/emitnx86.c', '../../py/asmthumb.c', '../../py/emitnthumb.c', '../../py/emitinlinethumb.c', '../../py/asmarm.c', '../../py/emitnarm.c', '../../py/asmxtensa.c', '../../py/emitnxtensa.c', '../../py/emitinlinextensa.c', '../../py/emitnxtensawin.c', '../../py/formatfloat.c', '../../py/parsenumbase.c', '../../py/parsenum.c', '../../py/emitglue.c', '../../py/persistentcode.c', '../../py/runtime.c', '../../py/runtime_utils.c', '../../py/scheduler.c', '../../py/nativeglue.c', '../../py/pairheap.c', '../../py/ringbuf.c']' returned non-zero exit status 1.
../../py/mkrules.mk:104: recipe for target 'build/genhdr/qstr.i.last' failed
make[1]: *** [build/genhdr/qstr.i.last] Error 1
make[1]: Leaving directory 'X:/PROJECTS/Other/libs&test/uPython/examples/embedding'
make[1]: *** Deleting file 'build/genhdr/qstr.i.last'
make: *** [-lmicropython] Error 2
Makefile:10: recipe for target '-lmicropython' failed
"make" terminated with exit code 2. Build might be incomplete.
```

So I have no idea why `gcc` runs instead of predefined CC on the line 524.
Can anyone help me to find out the reason of this?

stinos, 2021-01-27:
That's because it uses the preprocessor `$(CPP)`. You shouldn't be manually overriding this I think (or else: all of them, there's also CXX etc) but use something like `make CROSS_COMPILE=arm-none-eabi-`

PeterBorisenko, 2021-01-27:
Thanks!
I knew it should have some way of selecting toolchain.
Is there any possibility of selecting target bsp using the same approach?

stinos, 2021-01-27:
I don't know what 'target bsp' is so no idea :)

PeterBorisenko, 2021-01-27:
ok, that was incorrect question. 
After I added CROSS_COMPILE variable I have the following error: 
`x:\libs\gcc-arm-none-eabi\7-2017-q4-major\arm-none-eabi\include\sys\dirent.h:10:2: error: #error "<dirent.h> not supported"`
Seems it's no so easy as described in Readme.md

stinos, 2021-01-27:
> Seems it's no so easy as described in Readme.md

Yes and no; this seems a problem with the specific platform/toolchain you're using, which lacks a functional dirent.h. However you specifically include extmod/vfs_posix.c which includes dirent.h. Apart from that modos.c also uses that. So: either remove those or supply a usable dirent.h (like the msvc port does, for instance).

PeterBorisenko, 2021-01-27:
Ok, I removed them for now. Later I will check do I really need dirent.h or not.
Next problem is to implement platform-dependend memory allocation service.
By the way is there any possibility to run embedded micropython statically (with predefined RAM section for py application)?

stinos, 2021-01-27:
Yes, have a look at how the other ports call `gc_init` etc.

Note this is an issue a.k.a. bug tracker, your questions are more suitable for https://forum.micropython.org

PeterBorisenko, 2021-01-27:
Ok. Thanks!

closed 2021-01-27 by PeterBorisenko