micropython/micropython #6816: USB problem on RPi pico

davehylands, 2021-01-29:
I have an issue with rshell that seems to be caused by the USB comms.

If I issue the command `cp ports/rp2/examples/* /pyboard` then rshell will hang, typically on the first file.

If I add some statements in the `recv_file_from_host` function in rshell (this is the part that runs on the pico) that opens a UART and logs some additional information to the UART, then the copies all succeed.

I tried adding 20 msec delays each place that there was log statement, but that doesn't seem to make things work either.

Sorry I don't have a simpler test case, but I figured I'd log the issue and see if I can create a simple program which reproduces the hang/crash.

Sometimes, things hang, and sometimes the serial port is forcibly closed (I suspect that this might be a crash on the pico side).

davehylands, 2021-01-30:
I've been able to get a version working with time.sleep_ms(20) statements and no UART calls. I think when I made the statement above I had a couple `time.sleep(20)` rather than `time.sleep_ms(20)` statements in there.

davehylands, 2021-01-30:
The version that works for me can be found on the pico branch here: https://github.com/dhylands/rshell/tree/pico (if you checkout the repository locally, then you can use ./r.py from the root directory to run the rshell from the local tree rather than the installed version.

Commenting out the time.sleep_ms(20) found here: https://github.com/dhylands/rshell/blob/9ab7443a523280576604b42455ad2f5243f50589/rshell/main.py#L1039 cause the cp to hang. In particular the host side seems to be hanging when it calls `follow`.

dpgeorge, 2021-02-01:
I haven't yet tried to reproduce this, but it sounds like it could be a problem with flow control (at the REPL/raw REPL  level).  If adding sleeps makes it work then it does sound like the RP2 is not keeping up with the data coming from the PC.

Do you know if rshell is using the new raw paste mode?  Using that might fix the problem (without the need for sleep_ms(20)).  If you use the latest pyboard.py then it should use raw paste mode.

davehylands, 2021-02-04:
I tried using the latest pyboard.py from the tools directory and I still get the issue. I don't think that the problem is flow control since rshell has its own flow control. I'll see if I can reproduce with something simple.

davehylands, 2021-02-05:
Investigating further, it seems to be hanging on the close portion of the with statement that is opening the file for writing. I put a single character write at the end of the with and another just after the with. The host receives the one at the end of the with but doesn't receive the one just after the with. There doesn't seem to be any problems just doing file I/O - I created a python file which contains all of the data as was being transferred and if I run that file by itself then it works perfectly fine. The sleep that makes rshell work is just before the write of the file data, which by itself is weird because the file it usually fails on is a 466 byte file, so the entire file fits in a single buffer. So it opens the file, writes a single ACK character to stdout, then reads the 466 bytes from stdin, does the 20 msec sleep, writes the data out, and closes the file (as part of the with statment). So the only thing I can think of is that the USB data read is somehow messing with the file I/O and adding the delay allows the USB data read to do its thing and then the file I/O doesn't get interfered with.

dpgeorge, 2021-02-06:
> So the only thing I can think of is that the USB data read is somehow messing with the file I/O and adding the delay allows the USB data read to do its thing and then the file I/O doesn't get interfered with.

Ok, so maybe it has something to do with flash writes interacting with IRQs.  Code (including IRQs) execute from external SPI flash and when flash is being written to as part of a filesystem write it'll block IRQs.  This should be working correctly but there's the possibility something is not working as it should in the system.

dpgeorge, 2021-02-08:
I just noticed that rp2 uses a 256 byte USB CDC TX/RX buffer size.  Don't know if that would make a difference to rshell (maybe it expects something bigger)?

lornajane, 2021-02-08:
I have smaller replication cases for this bug in rshell that I think might help here too (and thanks! The `pico` branch on the rshell repo fixed it), here's the file that fails:

```
# represents
options = range(16)
```

Removing an "s" from the comment produces a file that uploads successfully with rshell 0.0.28:

```
# represent
options = range(16)
```

robert-hh, 2021-02-08:
If I count right, the first text is 33 characters, the second 32.

davehylands, 2021-02-09:
I changed the buffer size to 128 and it still winds up hanging, just in a slightly different spot (this time its hung waiting for an ACK from the pico). So not related to the close per-say, but still some type of interaction between USB and file I/O

Veverest, 2022-04-30:
Hi @davehylands, I have run into a similar problem. Did you find a fix or a workaround for this?

dhylands, 2022-04-30:
I wound up adding a delay and reducing the buffer size to 32 chars. The latest version of rshell seems to work fine with the pico.