micropython/micropython #6817: rp2 PIO: Allow setting JMP pin

tjvr, 2021-01-30:
Hello! 👋

I've just gotten a Raspberry Pi Pico, and I'm experimenting with the PIO features.

I'm trying to port the [uart_rx example](https://github.com/raspberrypi/pico-examples/blob/master/pio/uart_rx/uart_rx.pio) from C to Python.

It looks like there's no way to set the pin used by the `JMP PIN` command from Python. I checked the [rp2_pio.c source](https://github.com/micropython/micropython/blob/469345e7285128739e2934e7934e107ffda79fc1/ports/rp2/rp2_pio.c), it looks like `sm_config_set_jmp_pin` is never called. 

Is this intentional, or is this something we could add? Happy to try and PR this if you'd like.

Please let me know if there's a better place to raise this!

tjvr, 2021-01-30:
To be clear, I realise #6791 was only merged yesterday, so it's very early days 😅

tjvr, 2021-01-30:
I had a little play around and opened #6820 as an attempt to resolve this. No obligation to merge it or anything! In particular:

* I don't know if the name (`jmp_pin`) is ideal
* It looks like we might want a way to set the pin on the Program itself (i.e. as an argument to the `asm_pio` decorator).

dpgeorge, 2021-01-31:
> It looks like we might want a way to set the pin on the Program itself

I'm not sure, I think it does make sense as you did it already, an argument to `StateMachine`.  That's because you can instantiate a program multiple times on different state machines, but each time it can use a different pin.

dpgeorge, 2021-02-02:
Implemented in 7a9027fd5d64868322b9a82b0ac47ac0eba0167d

closed 2021-02-02 by dpgeorge