micropython/micropython #6819: RP2 LAN: WIZNET5K driver

enesbcs, 2021-01-30:
Is it possible to migrate Wiznet5K SPI LAN driver from STM32 port to RP2 port?
As i see RP2 as a Cortex M0+ is very similar as an STM32LO.

Copied network_lan.c, network_wiznet5k.c and modnwwiznet5k.c (and necessary .h files),  RP2 port compiles ok, according to output:
```
...
[ 74%] Building C object CMakeFiles/firmware.dir/uart.c.obj
[ 74%] Building C object CMakeFiles/firmware.dir/network_lan.c.obj
[ 74%] Building C object CMakeFiles/firmware.dir/network_wiznet5k.c.obj
[ 75%] Building C object CMakeFiles/firmware.dir/modnwwiznet5k.c.obj
[ 75%] Building C object CMakeFiles/firmware.dir/mnt/ADAT/micropython/lib/pico-sdk/src/rp2_common/hardware_adc/adc.c.obj
...
[ 99%] Building C object CMakeFiles/firmware.dir/mnt/ADAT/micropython/lib/pico-sdk/src/rp2_common/pico_fix/rp2040_usb_device_enumeration/rp2040_usb_device_enumeration.c.obj
[100%] Linking CXX executable firmware.elf
   text    data     bss     dec     hex filename
 247612      84  201076  448772   6d904 /mnt/ADAT/micropython/ports/rp2/build/firmware.elf
[100%] Built target firmware

```
But network module still not presented after .uf2 uploading.

```
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
ImportError: no module named 'network'
```

Minimal changes needed "spi.h" -> "hardware/spi.h" and "pins.h" -> "extmod/virtpin.h" to be able to compile them.
I am not an expert C coder, just a micropython user, anyone have a clue what can be the problem?

dpgeorge, 2021-02-01:
> Is it possible to migrate Wiznet5K SPI LAN driver from STM32 port to RP2 port?

Yes!

> But network module still not presented after .uf2 uploading.

The `network` module needs to be added to `mpconfigport.h`.  But also a few other things are needed, in particular the `usocket` module, and probably some other tweaks here and there.

enesbcs, 2021-02-01:
> The `network` module needs to be added to `mpconfigport.h`. But also a few other things are needed, in particular the `usocket` module, and probably some other tweaks here and there.

Cool! As i seen STM32 has Makefile with conditional includes, but RP2 port created in pure cmake style. Originally i thought i can use specific Makefile parts from STM32 port, but now i see that it will not be enough.

enesbcs, 2021-04-07:
> The `network` module needs to be added to `mpconfigport.h`. But also a few other things are needed, in particular the `usocket` module, and probably some other tweaks here and there.

I've tried to tweak, but i am afraid it will require a lot more work, as STM32 spi.h and pin.h is not compatible with the RP2 branch, the W5500 driver will not compile at all. :( 
After two months of trying, I gave up permanently.

diasrobson, 2021-09-02:
@dpgeorge any change you can help us out I've been looking all over the place for a solution, there are a lot of people wanting to use the RP2 for weird IoT projects but are struggling due to this problem, it's a real dealbreaker for people trying to move over the micropython for IoT

robert-hh, 2021-09-02:
There is a PR open for the ARDUINO_CONNECT_2400, which next to the WiFi driver adds the network module and the TCP/IP stack. It also includes attempts to make the network module generic. Based on that it is just one more step to add the WIZNET5K driver. Are you sure that you need Ethernet support?  Or would WiFi also suit your needs?

enesbcs, 2021-11-06:
Finally there will be a usable Pico board with W5K.. we just need micropython support sometime.

https://www.hackster.io/news/wiznet-s-w5100s-evb-pico-clones-the-raspberry-pi-pico-but-adds-a-handy-ethernet-port-too-11125e568928

irinakim12, 2021-11-22:
@enesbcs  #8021  If this pull request approve , you can use the wiznet5k driver in ports/rp2

enesbcs, 2021-11-22:
> @enesbcs #8021 If this pull request approve , you can use the wiznet5k driver in ports/rp2 You can use the code in the link below until it becomes merge. https://github.com/irinakim12/micropython

Thanx, i can compile it but unfortunately i am unable to get any sign of work from a W5500 SPI modul, not even a green blink.

Test code:

```
import network
import machine

SPI1_SCK = 10
SPI1_TX = 11
SPI1_RX = 12
SPI1_CSn = 13
W5500_RSTn = 15
SPILINE = 1

ethernetRst = machine.Pin(W5500_RSTn,machine.Pin.OUT)
cs = machine.Pin(SPI1_CSn,machine.Pin.OUT)
spi_bus = machine.SPI(SPILINE,8000000,sck=machine.Pin(SPI1_SCK), mosi=machine.Pin(SPI1_TX), miso=machine.Pin(SPI1_RX), polarity=0, phase=0)
lan_if = network.WIZNET5K(spi_bus, cs, ethernetRst)
lan_if.ifconfig(('192.168.0.111','255.255.0.0','192.168.0.1','8.8.8.8'))
```

Results:

```
>>> lan_if.isconnected()
False
>>> lan_if.ifconfig()
('0.0.0.0', '0.1.2.3', '0.1.2.3', '8.8.8.8')

```

robert-hh, 2021-11-23:
you may have to call:

lan_if.active(True)

enesbcs, 2022-01-30:
It seems W5500 support stucked a little https://github.com/micropython/micropython/issues/8023

dpgeorge, 2022-06-03:
Wiznet is supported on rp2 since bca816f5ac36609212ab087789ee028d1d5d534d

closed 2022-06-03 by dpgeorge