micropython/micropython #6831: Any sample for Raspberry Pi Pico RTC?

pankaj-nikam, 2021-02-02:
Hi team,

Thank you for porting Micropython for Pico. Is there any sample for using RTC with Pico?

Thanks.

davehylands, 2021-02-03:
For the time being, machine.RTC() doesn't seem to be implemented on the Pico. This means that you need to access the registers directly. There is a forum post on doing this: https://forum.micropython.org/viewtopic.php?f=15&t=9706 and I just updated rshell (in the pico branch) to take do this as well: https://github.com/dhylands/rshell/commit/cd1b5522a9991bcc83e01eb4ddc6f5146b0a980c

pankaj-nikam, 2021-02-03:
Thanks for the update :) Let us hope that machine.RTC gets implemented soon 😄

aivarannamaa, 2021-02-03:
@dpgeorge, can you tell us whether `machine.RTC` is coming soon for Pico? (If not, then I'll implement the work-around described at https://forum.micropython.org/viewtopic.php?f=15&t=9706 in the next Thonny version, but a standard solution would be so much nicer.)

kadamski, 2021-02-18:
@aivarannamaa I've created a preliminary support for machine.RTC in https://github.com/micropython/micropython/pull/6928. Do you need anything else than a datetime() method?

aivarannamaa, 2021-02-18:
> @aivarannamaa I've created a preliminary support for machine.RTC in #6928. Do you need anything else than a datetime() method?

@kadamski, `.datetime(...)` would suffice for Thonny. Thank you!

Currently Thonny calls it like this 

`rtc.datetime(now.tm_year, now.tm_mon, now.tm_mday, now.tm_wday + 1, now.tm_hour, now.tm_min, now.tm_sec, 0)` 

where `now` is computed by CPython's `datetime.datetime.now().timetuple()`. Hopefully this matches your implementation :)

kadamski, 2021-02-18:
@aivarannamaa wait, do you use 8 separate arguments? My implementation expects 1 argument that is a tuple of 8 integers, just like the esp32 port..

aivarannamaa, 2021-02-18:
> @aivarannamaa wait, do you use 8 separate arguments? My implementation expects 1 argument that is a tuple of 8 integers, just like the esp32 port.

Sorry, I got it wrong. Thonny actually also sends these components as an 8-tuple. If your method is like the one in esp32 port, then Thonny will be happy with it.

kadamski, 2021-02-18:
@aivarannamaa good. I've created a precompiled version of my branch with this change, if you would like to easily test it with Thonny. See https://github.com/kadamski/micropython/releases/tag/v1.14-67-gcf6a01588%2Brp2_rpc_7cc6412

kadamski, 2021-02-19:
> For the time being, machine.RTC() doesn't seem to be implemented on the Pico. This means that you need to access the registers directly. There is a forum post on doing this: https://forum.micropython.org/viewtopic.php?f=15&t=9706 and I just updated rshell (in the pico branch) to take do this as well: [dhylands/rshell@cd1b552](https://github.com/dhylands/rshell/commit/cd1b5522a9991bcc83e01eb4ddc6f5146b0a980c)

@davehylands you're also welcome to test it for your case. I would be glad to get some feedback about it. See my comment above for the details.

aivarannamaa, 2021-02-19:
@kadamski, with your uf2, Thonny seems to set UTC without problems. Good work!

kadamski, 2021-02-19:
@aivarannamaa thanks, could you put this info to my pull request https://github.com/micropython/micropython/pull/6928 to show it was verfied by somebody else than me? :)

The-Bow-Hunter, 2021-06-17:
hey, just asking but did someone put it in the documentation befor it was on the corresponding stable build? https://docs.micropython.org/en/latest/rp2/quickref.html#real-time-clock-rtc
states taht it is there, but i can not import it there

robert-hh, 2021-06-17:
Try the daily update of the firmware.

The-Bow-Hunter, 2021-06-19:
in 1.16. it works fine now. Still shouldnt be in the documentation of the 1.15

The-Bow-Hunter, 2021-06-19:
Maybe this issue should get closed then

zzzbatmand, 2023-01-24:
The RP2 `machine.RTC.datetime()` function, returns the wrong order of time.
The function returns `(year, month, day, day_of_the_week, hour, minute, second, 0)`
Where normally "day_of_the_week" is second-to-last.
The correct format should be as follows `(year, month, day, hour, minute, second, day_of_the_week, day_of_the_year)`
As shown in the [RP2/modutime.c](https://github.com/micropython/micropython/blob/master/ports/rp2/modutime.c#L36) module.

robert-hh, 2023-01-24:
The format of rtc.datetime() and time.localtime() is indeed different, and that's true for all ports. That may be irritating. But there is an inconsistency to the documentation, which does not mention the day-of-week field,and instead a tz-info field, which is not handled by the firmware.

zzzbatmand, 2023-01-24:
> The format of rtc.datetime() and time.localtime() is indeed different, and that's true for all ports. That may be irritating. And there are quite a few inconsistencies between the ports and documentation with the RTC module.

Is there a reason for this?
Otherwise I have made the changes as a pull request [here](https://github.com/micropython/micropython/pull/10577).

robert-hh, 2023-01-24:
> Is there a reason for this?

No. Just history. I made an issue report, and that should be fixed for all ports. But  the only consistent thing between all ports is the rtc.datetime() format as it is now. And therefore it should be kept.

jonnor, 2024-08-18:
I believe this issue can be closed, as the associated PR have been merged.

closed 2024-08-18 by robert-hh