micropython/micropython #6833: ADC reading incorrectly when others are created in RP2 port

benevpi, 2021-02-02:
In the below code, whether or not you created ADC3 affects the reading of ADC4. With the line commented out, I get a reading of about 17 degrees. With it there I get a reading of about 80.

```
import machine
import time

sensor_temp = machine.ADC(4)
blah = machine.ADC(3) # comment or uncomment this line to trigger / untrigger

conversion_factor = 3.3 / 65535

min_temp = 0
max_temp = 30

while True:
    reading = sensor_temp.read_u16() * conversion_factor
    temperature = 27 - (reading - 0.706)/0.001721
    print(temperature)
    time.sleep(0.5)
```

lurch, 2021-02-02:
Just out of curiosity, I tried having a look at https://github.com/micropython/micropython/blob/master/ports/rp2/machine_adc.c but it seems to be using `channel` in confusing ways to refer to both the ADC channel _and_ the GPIO number, so I couldn't really follow what it was doing :confused:

dpgeorge, 2021-02-03:
Thanks for the report, I can confirm the issue.  It's because `adc_init()` is being called each time an ADC object is constructed, which turns off the temperature sensor switch.

Should be fixed by #6834.

dpgeorge, 2021-02-04:
Fixed by 7f7b4f2bc68a06c9f47d6b99e953cac0b21c26c9

closed 2021-02-04 by dpgeorge