micropython/micropython #6837: rp2 port no module named array

benevpi, 2021-02-03:
In the latest version of MicroPython from the getting-started page (version 1.14 dated 2nd Feb), there doesn't appear to be the array module. The following fails

```
import array
```

I *think* this also stops PIO code from running -- at least, it seems to stop mine.

robert-hh, 2021-02-03:
There is uarray. So it's the alias which is missing. You can use

import uarray as array

benevpi, 2021-02-03:
Just to add -- here's some code that was working until this version: https://gist.githubusercontent.com/benevpi/845e5dc432f05b6160ed56fc6f3b4032/raw/e151508ae724fcfbf0b08be33ebce554dff563c2/pwm_pio.py

benevpi, 2021-02-03:
there's no uarray

```
MicroPython v1.14 on 2021-02-02; Raspberry Pi Pico with RP2040

Type "help()" for more information.
>>> import uarray
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'uarray'
>>> 
```

aallan, 2021-02-03:
I'm going to back out, and point the [Getting Started](https://www.raspberrypi.org/documentation/pico/getting-started/) page at the launch version of MicroPython (dated stamped 20210121) unless this is resolved.

robert-hh, 2021-02-03:
The code which you are pointed at works fine. So there may by a problem in the public firmware image. I am using this one which I have build myself: https://github.com/robert-hh/micropython/blob/rp2_fat/ports/rp2/build/firmware.uf2

benevpi, 2021-02-03:
yep, @robert-h 's firmware works fine. It's the public one with the problem

aallan, 2021-02-03:
> The code which you are pointed at works fine. So there may by a problem in the public firmware image. I am using this one which I have build myself: https://github.com/robert-hh/micropython/blob/rp2_fat/ports/rp2/build/firmware.uf2

Could you quickly download the public binary, https://micropython.org/download/rp2-pico/rp2-pico-latest.uf2, and test?

aallan, 2021-02-03:
Okay then. Looks like it's up to @dpgeorge.

The-Spirit, 2021-02-04:
> I'm going to back out, and point the [Getting Started](https://www.raspberrypi.org/documentation/pico/getting-started/) page at the launch version of MicroPython (dated stamped [20210121](https://www.raspberrypi.org/documentation/pico/getting-started/static/5d8e777377e8dbe23cf36360d6efc727/pico_micropython_20210121.uf2)) unless this is resolved. @dpgeorge can you take a look?

Fixed it for me, thanks 👍 
Was thinking it was me with my inexperience.....

dpgeorge, 2021-02-04:
Thanks for the report.  The reason the public builds don't include it is because they build the firmware in a directory that is outside the source tree, and the user-c-modules feature didn't support this.  That should be fixed by #6845.

dpgeorge, 2021-02-04:
Hopefully fixed by 1f800cac3c4f56418486f24c3824418a890f759f.  Let's see if the daily builds have array.

closed 2021-02-04 by dpgeorge

lurch, 2021-02-04:
Thanks Damien.
If the next daily-build _does_ fix this problem (which prevented PIO code from working), should we link the Getting Started page to (a copy of) that, rather than the 20210121 release (which has the GP19 and ADC temp-sense bugs)? Or should we continue linking to 20210121 until the stable v1.15 is released? ping @aallan

The-Spirit, 2021-02-05:
> Hopefully fixed by [1f800ca](https://github.com/micropython/micropython/commit/1f800cac3c4f56418486f24c3824418a890f759f). Let's see if the daily builds have array.

Feedback: latest unstable build (rp2-pico-20210205-unstable-v1.14-8-g1f800cac3.uf2) has array's 👍

benevpi, 2021-02-05:
Also, latest unstable rp2-pico-20210205-unstable-v1.14-8-g1f800cac3.uf2 version works with PIO again (at least in the simple version tested above)

aallan, 2021-02-05:
Okay, I've pulled last night's unstable and the version of MicroPython you get when you click to download the UF2 from the Getting Started page is now rp2-pico-20210205-unstable-v1.14-8-g1f800cac3.uf2. I can flip the link back to the permanent URL of  https://micropython.org/download/rp2-pico/rp2-pico-latest.uf2 when V1.15 is released I guess?

dpgeorge, 2021-02-05:
Thanks for confirming the fix.

> Okay, I've pulled last night's unstable and the version of MicroPython you get when you click to download the UF2 from the Getting Started page is now [...]

This sounds fine.  But in the future it make come up again that we want a fix to go through to a "stable" release before a MicroPython release.  Maybe there's a better way to manage this?  I don't really have any ideas though.

lurch, 2021-02-05:
In SemVer land you'd make a v1.14.1 release ?

dpgeorge, 2021-02-05:
> In SemVer land you'd make a v1.14.1 release ?

Indeed.  For that we'd need to start doing branches with patch releases.  That can be done, just hasn't been done before here.

ObeidHaidar, 2021-02-22:
Hey,

Had the same issue recently. I believe Thonny Python IDE downloads the stable uf2 file onto the pico.
Problem was fixed when downloading the unstable build https://www.raspberrypi.org/documentation/pico/getting-started/static/f70cc2e37832cde5a107f6f2af06b4bc/rp2-pico-20210205-unstable-v1.14-8-g1f800cac3.uf2

robert-hh, 2021-02-22:
It' called daily build, not unstable build. These daily builds are typically as stable as the others.

aallan, 2021-02-22:
> It' called daily build, not unstable build. These daily builds are typically as stable as the others.

The word "unstable" is right there in the file name.