micropython/micropython #6841: Unable to open SSL / HTTPS socket with ESP8266: OSError from ssl.wrap_socket

joseph-long, 2021-02-04:
*xposted from https://github.com/micropython/micropython-lib/issues/413 because I read this is a better place for micropython issues*

I've just set up an ESP8266 with 4MB of flash using `esp8266-20210202-v1.14.bin` and I'm trying to run some example code to open a socket for https:

```
import usocket
import ssl

hostname = 'xwcl.science'
s = usocket.socket()
ai = usocket.getaddrinfo(hostname, 443)
addr = ai[0][-1]
print(addr)
s.connect(addr)
s = ssl.wrap_socket(s)
```

In the REPL, it all goes well until `ssl.wrap_socket`:

```
>>> hostname = 'xwcl.science'
>>> s = usocket.socket()
>>> ai = usocket.getaddrinfo(hostname, 443)
>>> addr = ai[0][-1]
>>> print(addr)
('104.238.141.27', 443)
>>> s.connect(addr)
>>> s = ssl.wrap_socket(s)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: -40
```

joseph-long, 2021-02-04:
the NodeMCU project also uses mbed TLS and has an in-depth and rather discouraging description of its limitations: https://nodemcu.readthedocs.io/en/release/modules/tls/

Seems like there's just not enough heap memory to speak SSL to arbitrary internet servers. Rats.

ofen, 2021-11-06:
similar to https://github.com/micropython/micropython-lib/issues/400

jonnor, 2024-09-02:
Closing as duplicate of https://github.com/micropython/micropython-lib/issues/400

closed 2024-09-02 by jonnor