micropython/micropython #6846: Need a way to disable Control-C

davehylands, 2021-02-04:
When transferring binary data (perhaps .mpy files) embedded 0x03 characters get interpreted as Control-C characters which interrupts the file transfer.

With the pyboard, the Control-C interrupt can be disabled by using setinterrupt(-1) on the machine.USB_VCP() class. However, there is no other way to do this.

So it's impossible to transfer binary 0x03 characters over serial REPLs, for example.

The rp2 currently doesn't implement machine.USB_VCP(), so there is currently no way to disable Control-C as interrupt.

I'd like to see a way to disable Control-C on stdin regardless of whether USB or serial or whatever is being used as the REPL.

robert-hh, 2021-02-04:
You can disable Ctr-C with micropython.kbd_intr(-1) and re-enable it with micropython.kbd_intr(3). It's been there since long and works.

davehylands, 2021-02-04:
For some reason, I didn't know about that. Thanks.

closed 2021-02-04 by davehylands