micropython/micropython #6847: Error in compilation files

Krasn4ck, 2021-02-04:
```
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/genhdr
GEN build/genhdr/mpversion.h
GEN build/genhdr/moduledefs.h
GEN build/genhdr/qstr.i.last
gcc:gcc:gcc:   fatal error: fatal error: fatal error:                 cannot execute 'cannot execute '
      cc1': CreateProcess: No such file or directory
': CreateProcess: No such file or directory
': CreateProcess: No such file or directory
cccooommmppipililalatatitiono ino nt ettreemrrimmiinnnaaattteeeddd...

Traceback (most recent call last):
  File "../py/makeqstrdefs.py", line 188, in <module>
gcc: fatal error: cannot execute 'cc1': CreateProcess: No such file or directory
    preprocess()
c  File "../py/makeqstrdefs.py", line 62, in preprocess
ompil    for output in p.imap(pp(flags), chunks):
a  File "C:\msys64\mingw64\lib\python3.8\multiprocessing\pool.py", line 868, in next
tion termin    raise value
a  File "C:\msys64\mingw64\lib\python3.8\multiprocessing\pool.py", line 125, in worker
ted.
    result = (True, func(*args, **kwds))
  File "../py/makeqstrdefs.py", line 46, in run
    return subprocess.check_output(args.pp + flags + files)
  File "C:\msys64\mingw64\lib\python3.8\subprocess.py", line 411, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\msys64\mingw64\lib\python3.8\subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['gcc', '-E', '-I.', '-Ibuild', '-I..', '-Wall', '-Werror', '-
Wextra', '-Wno-unused-parameter', '-Wpointer-arith', '-std=gnu99', '-DFFCONF_H="lib/oofatfs/ffconf.h"'
, '-Os', '-fdata-sections', '-ffunction-sections', '-fno-asynchronous-unwind-tables', '-DNO_QSTR', '..
/py/mpstate.c', '../py/malloc.c', '../py/gc.c', '../py/pystack.c', '../py/qstr.c', '../py/vstr.c', '..
/py/mpprint.c', '../py/unicode.c', '../py/mpz.c', '../py/reader.c', '../py/lexer.c', '../py/parse.c',
'../py/scope.c', '../py/compile.c', '../py/emitcommon.c', '../py/emitbc.c', '../py/asmbase.c', '../py/
asmx64.c', '../py/emitnx64.c']' returned non-zero exit status 1.
gcc: fatal error: cannot execute 'cc1': CreateProcess: No such file or directory
compilation terminated.
gcc: gcc: fatal error: cannot execute 'fatal error: cc1cannot execute '': CreateProcess: No such file
or directory
cc1c m': CreateProcess: No such file or directory
pciomlapitiloant tieorminn atteerd.m
inated.
make: *** [../py/mkrules.mk:105: build/genhdr/qstr.i.last] Error 1
make: *** Deleting file 'build/genhdr/qstr.i.last'
```

dlech, 2021-02-04:
It says you don't have a compiler (gcc) installed.

Krasn4ck, 2021-02-04:
> It says you don't have a compiler (gcc) installed.

I don't believe it, I just compiled version 1.13 of micropython and it worked, only with version 1.14 that error appears.

```
CC build-STM32F7DISC/pins_STM32F7DISC.c
LINK build-STM32F7DISC/firmware.elf
   text    data     bss     dec     hex filename
 461704      92   78896  540692   84014 build-STM32F7DISC/firmware.elf
GEN build-STM32F7DISC/firmware.dfu
GEN build-STM32F7DISC/firmware.hex

```

dlech, 2021-02-04:
Maybe related to this then: https://github.com/micropython/micropython/pull/6579

Krasn4ck, 2021-02-04:
> Maybe related to this then: #6579

Thank you, i will check it out.

jimmo, 2021-02-05:
I notice you're compiling with mingw... my guess is that somehow the PATH is not being inherited all the way down to the gcc invocation and so gcc can't find cc1... perhaps makeqstrdefs.py needs to set some extra flags (although I thought the default was to inherit the environment).

Although #6579 is worth investigating as it changes the way we invoke the QSTR preprocessing, I think the relevant change might actually be #6525 which is where it was moved out of the makefile.

JonatanESalinas, 2021-02-05:
I want to compile Micropython for a particular board, but I'm having this same issue when trying to compile Micropython 1.14 in Windows. I have installed MinGW, and I'm using Cygwin to make the compilation. When I try to compile version 1.13 the compilation is successful (and I get the firmware for the board without problems). But if I want to do it with version 1.14, I type in Cygwin the command: make STRIP=echo SIZE=echo in the mpy-cross folder of micropython, and this error message appears.

Am I doing something wrong? Is there any way I can solve this and compile Micropython 1.14 in Windows to generate the necessary firmware? Do I need to install anything particular apart from MinGW and Cygwin? Thanks in advance.

teuler, 2021-02-05:
Under Windows 10 there is the option of building and deploying micropython in the Ubuntu Subsystem. I did this successfully with 1.13; I did not try it with 1.14 though.

stinos, 2021-02-06:
> But if I want to do it with version 1.14, I type in Cygwin the command: 'make STRIP=echo SIZE=echo' in the mpy-cross folder of micropython, and this error message appears

If you run make without specifycing the compiler, it will try to use gcc as compiler. However you say you installed MinGW but if you do that (in Cygwin) that doesn't install gcc but e.g. x86_64-w64-mingw32-gcc and you'd need to run `make CROSS_COMPILE=x86_64-w64-mingw32-`. Or do you have another way to get gcc available as command? What is the output of `gcc -v` on the commandline?

@Krasn4ck @JonatanESalinas can you specify what your build environment looks like exactly. How and what did you install, which version?

JonatanESalinas, 2021-02-06:
I tried to run `make CROSS_COMPILE=x86_64-w64-mingw32-` in Cygwin in the mpy-cross folder, the process began and everything looked good, but suddenly an error appeared:
```
(Some output omitted)
CC ../py/smallint.c
CC ../py/frozenmod.c
CC main.c
CC gccollect.c
CC ../lib/utils/gchelper_generic.c
CC ../ports/windows/fmode.c
LINK mpy-cross
make: x86_64-w64-mingw32-strip: No such file or directory
make: *** [../py/mkrules.mk:186: mpy-cross] Error 127
make: *** Deleting file 'mpy-cross'
```
If I type `gcc -v` in Cygwin, the output is the following:
```
$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-cygwin/10/lto-wrapper.exe
Target: x86_64-pc-cygwin
Configured with: /mnt/share/cygpkgs/gcc/gcc.x86_64/src/gcc-10.2.0/configure --srcdir=/mnt/share/cygpkgs/gcc/gcc.x86_64/src/gcc-10.2.0 --prefix=/usr --exec-prefix=/usr --localstatedir=/var --sysconfdir=/etc --docdir=/usr/share/doc/gcc --htmldir=/usr/share/doc/gcc/html -C --build=x86_64-pc-cygwin --host=x86_64-pc-cygwin --target=x86_64-pc-cygwin --without-libiconv-prefix --without-libintl-prefix --libexecdir=/usr/lib --with-gcc-major-version-only --enable-shared --enable-shared-libgcc --enable-static --enable-version-specific-runtime-libs --enable-bootstrap --enable-__cxa_atexit --with-dwarf2 --with-tune=generic --enable-languages=c,c++,fortran,lto,objc,obj-c++ --enable-graphite --enable-threads=posix --enable-libatomic --enable-libgomp --enable-libquadmath --enable-libquadmath-support --disable-libssp --enable-libada --disable-symvers --with-gnu-ld --with-gnu-as --with-cloog-include=/usr/include/cloog-isl --without-libiconv-prefix --without-libintl-prefix --with-system-zlib --enable-linker-build-id --with-default-libstdcxx-abi=gcc4-compatible --enable-libstdcxx-filesystem-ts
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 10.2.0 (GCC)
```
And if I type this same command in the Windows command prompt, the output is:
```
C:\Users\jonat>gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.2.0/lto-wrapper.exe
Target: x86_64-w64-mingw32
Configured with: ../gcc-10.2.0/configure --prefix=/mingw64 --with-local-prefix=/mingw64/local --build=x86_64-w64-mingw32 --host=x86_64-w64-mingw32 --target=x86_64-w64-mingw32 --with-native-system-header-dir=/mingw64/x86_64-w64-mingw32/include --libexecdir=/mingw64/lib --enable-bootstrap --with-arch=x86-64 --with-tune=generic --enable-languages=c,lto,c++,fortran,ada,objc,obj-c++,jit --enable-shared --enable-static --enable-libatomic --enable-threads=posix --enable-graphite --enable-fully-dynamic-string --enable-libstdcxx-filesystem-ts=yes --enable-libstdcxx-time=yes --disable-libstdcxx-pch --disable-libstdcxx-debug --disable-isl-version-check --enable-lto --enable-libgomp --disable-multilib --enable-checking=release --disable-rpath --disable-win32-registry --disable-nls --disable-werror --disable-symvers --disable-plugin --with-libiconv --with-system-zlib --with-gmp=/mingw64 --with-mpfr=/mingw64 --with-mpc=/mingw64 --with-isl=/mingw64 --with-pkgversion='Rev6, Built by MSYS2 project' --with-bugurl=https://github.com/msys2/MINGW-packages/issues --with-gnu-as --with-gnu-ld --with-boot-ldflags='-pipe -Wl,--dynamicbase,--high-entropy-va,--nxcompat,--default-image-base-high -Wl,--disable-dynamicbase -static-libstdc++ -static-libgcc' 'LDFLAGS_FOR_TARGET=-pipe -Wl,--dynamicbase,--high-entropy-va,--nxcompat,--default-image-base-high' --enable-linker-plugin-flags='LDFLAGS=-static-libstdc++\ -static-libgcc\ -pipe\ -Wl,--dynamicbase,--high-entropy-va,--nxcompat,--default-image-base-high\ -Wl,--stack,12582912'
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 10.2.0 (Rev6, Built by MSYS2 project)
```
In summary, at the very beginning what I did was to follow the instructions of [this](https://www.youtube.com/watch?v=0HD0pqVtsmw) tutorial to install mingw-w64 in my computer. Basically here I downloaded MSYS2, and from the "MSYS2 MSYS" terminal I installed Mysys2 updating packages. Then I opened the "MSYS2 MinGW 64-bits" terminal and from there I installed the GNU Compiler collection using `pacman -S mingw-w64-x86_64-gcc` , the GNU tools for ARM Embedded Processors with `mingw-w64-x86_64-arm-none-eabi-gcc` and the debugger typing `pacman -S mingw-w64-x86_64-gdb`.

After that, I downloaded Cygwin64 terminal, and if I type `uname -a`, Cygwin shows:
```
$ uname -a
CYGWIN_NT-10.0 LAPTOP-01DBBFGQ 3.1.7(0.340/5/3) 2020-08-22 17:48 x86_64 Cygwin
```
I currently have Python 3.9.1 installed on my laptop too.
After all these, I built Micropython 1.13 with success but 1.14 without success.
I hope this information would be useful, thanks in advance for any help.

stinos, 2021-02-07:
Thanks for the explanation!

> I tried to run make CROSS_COMPILE=x86_64-w64-mingw32- in Cygwin in the mpy-cross folder, the process began and everything looked good, but suddenly an error appeared:

You still have to add the `STRIP=echo SIZE=echo` part to `make` as well.

However I'm not entirely sure this actually invokes the gcc installed with MSYS2: by default there's no way Cygwin can find that one. Is it possible you also have a mingw installation in Cygwin? Should have asked earlier, but what does `x86_64-w64-mingw32-gcc -v` output? What does `echo $PATH` output?

> If I type gcc -v in Cygwin, the output is the following:

Note that is not the gcc you installed for MSYS2, but the one from Cygwin. Which produces executables which usually only work properly a Cygwin shell. And can only be used to build the MicroPython unix port, not the windows port.

> And if I type this same command in the Windows command prompt,

You mean the MSYS2 command prompt, or really just cmd.exe (or powershell.exe)? In the latter case, if the gcc command is found that means it is in your PATH environment variable? That's not ideal, as it is easily going to clash with MSYS2 and/or Cygwin. If you reall mean cmd.exe, what does `echo %PATH%` output? (or for powershell `$env:PATH`)

In any case: I don't know why it worked pre v1.14 and not now anymore, but the thing is: you might have no less than 3 different versions of gcc. (MSYS2's mingw-gcc, Cygwin's own gcc and migw-gcc installed in Cygwin). And possibly added to the system-wide PATH.

Now, each of those work with v1.14 (I just tried all 3 situations in clean sandboxed Windows installation) and I cannot reproduce the problem, which probably means ther are subtle installation differences. What I did not try though is to call MSYS2's gcc from within Cygwin, but I also don't know a clean way of doing that.

Is there a reason you want Cygwin and/or different compilers? I'm not saying mixing them will never work, but it easily leads to confusion and/or problems. 

Not mixing things is more straightforward:
- either use Cygwin, and install mingw-gcc (not gcc!) via Cygwin's package manager, using CROSS_COMPILE as above
- or use MSYS2, install make/mingw-gcc etc via pacman as you did, but then also use the MSYS2 shell (that's mingw64.exe to get gcc as a command without extra prefixes), not Cygwin
- or if it somehow suits you, use Cygwin and Cygwin's gcc (but again, micropython nor mpy-cross will run outside of the Cygwin shell)

> I currently have Python 3.9.1 installed on my laptop too

I'm starting to think this might be the culprit (gcc working under `make` but not via `python`). Also cannot reproduce, though there are again multiple ways to install python, some 'better' than others. How did you install Python? In the tests I just did I always installed the Python distibution with the package managre for platform (MSYS2 or Cygwin).

JonatanESalinas, 2021-02-08:
Thank you very much, @stinos . Now it works, I don't know what happened, I just erased the micropython 1.14 folder in my laptop, and I decompressed the micropython 1.14 zip again (The one I downloaded from here). Then, I typed `make STRIP=echo SIZE=echo CROSS_COMPILE=x86_64-w64-mingw32-` in Cygwin in the mpy-cross folder, and the process finished successfully. Next, I typed `make BOARD=STM32F4DISC` (using that board as an example) in the micropython-1.14/ports/stm32 folder, and the building began. The process finished sucessfully and the .hex and .dfu files were generated, as I was searching for. I tried again with another board and again it worked.

Maybe I broke up something in the original micropython-1.14 folder that I was using at the beginning...
For some extra information:

> but what does `x86_64-w64-mingw32-gcc -v` output?

If I type this command in Cygwin, the output is:
```
$ x86_64-w64-mingw32-gcc -v
Using built-in specs.
COLLECT_GCC=C:\msys64\mingw64\bin\x86_64-w64-mingw32-gcc.exe
COLLECT_LTO_WRAPPER=C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.2.0/lto-
wrapper.exe
Target: x86_64-w64-mingw32
Configured with: ../gcc-10.2.0/configure --prefix=/mingw64 --with-local-prefix=/min
gw64/local --build=x86_64-w64-mingw32 --host=x86_64-w64-mingw32 --target=x86_64-w64
-mingw32 --with-native-system-header-dir=/mingw64/x86_64-w64-mingw32/include --libe
xecdir=/mingw64/lib --enable-bootstrap --with-arch=x86-64 --with-tune=generic --ena
ble-languages=c,lto,c++,fortran,ada,objc,obj-c++,jit --enable-shared --enable-stati
c --enable-libatomic --enable-threads=posix --enable-graphite --enable-fully-dynami
c-string --enable-libstdcxx-filesystem-ts=yes --enable-libstdcxx-time=yes --disable
-libstdcxx-pch --disable-libstdcxx-debug --disable-isl-version-check --enable-lto -
-enable-libgomp --disable-multilib --enable-checking=release --disable-rpath --disa
ble-win32-registry --disable-nls --disable-werror --disable-symvers --disable-plugi
n --with-libiconv --with-system-zlib --with-gmp=/mingw64 --with-mpfr=/mingw64 --wit
h-mpc=/mingw64 --with-isl=/mingw64 --with-pkgversion='Rev6, Built by MSYS2 project'
 --with-bugurl=https://github.com/msys2/MINGW-packages/issues --with-gnu-as --with-
gnu-ld --with-boot-ldflags='-pipe -Wl,--dynamicbase,--high-entropy-va,--nxcompat,--
default-image-base-high -Wl,--disable-dynamicbase -static-libstdc++ -static-libgcc'
 'LDFLAGS_FOR_TARGET=-pipe -Wl,--dynamicbase,--high-entropy-va,--nxcompat,--default
-image-base-high' --enable-linker-plugin-flags='LDFLAGS=-static-libstdc++\ -static-
libgcc\ -pipe\ -Wl,--dynamicbase,--high-entropy-va,--nxcompat,--default-image-base-
high\ -Wl,--stack,12582912'
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 10.2.0 (Rev6, Built by MSYS2 project)
```

> What does `echo $PATH` output?

I typed this in Cygwin and the output is:
```
$ echo $PATH
/usr/local/bin:/usr/bin:/cygdrive/d/Program Files/RoboDK/Python37/Scripts:/cygdrive/d/Program Files/RoboDK/Python37:/cygdrive/c/Windows/system32:/cygdrive/c/Windows:/cygdrive/c/Windows/System32/Wbem:/cygdrive/c/Windows/System32/WindowsPowerShell/v1.0:/cygdrive/c/Windows/System32/OpenSSH:/cygdrive/c/Program Files/Java/jdk-13.0.1/bin:/cygdrive/c/Program Files (x86)/Microchip/xc8/v1.43/bin:/cygdrive/c/Program Files/Git/cmd:/cygdrive/c/Program Files/PuTTY:/cygdrive/d/Program Files/MATLAB/R2019b/bin:/cygdrive/c/Program Files (x86)/Wolfram Research/WolframScript:/cygdrive/c/Program Files/Microsoft SQL Server/Client SDK/ODBC/110/Tools/Binn:/cygdrive/c/Program Files (x86)/Microsoft SQL Server/120/Tools/Binn:/cygdrive/c/Program Files/Microsoft SQL Server/120/Tools/Binn:/cygdrive/c/Program Files/Microsoft SQL Server/120/DTS/Binn:/cygdrive/c/Program Files (x86)/Windows Kits/8.1/Windows Performance Toolkit:/cygdrive/c/MinGW/bin:/cygdrive/d/ProgramCC/Files:/cygdrive/c/Program Files/Microsoft SQL Server/130/Tools/Binn:/cygdrive/c/Program Files/Microsoft SQL Server/Client SDK/ODBC/170/Tools/Binn:/cygdrive/c/Program Files/dotnet:/cygdrive/c/msys64/mingw64/bin:/cygdrive/d/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin:/cygdrive/c/MinGW/mingw64/bin:/cygdrive/c/MinGW/msys/1.0/bin:/cygdrive/c/MinGW/bin:/cygdrive/c/MinGW/mingw64/x86_64-w64-mingw32/bin:/cygdrive/c/Users/jonat/AppData/Local/Microsoft/WindowsApps:/cygdrive/c/Users/jonat/.dotnet/tools:/cygdrive/d/Users/jonat/AppData/Local/Programs/Microsoft VS Code/bin
```

> You mean the MSYS2 command prompt, or really just cmd.exe (or powershell.exe)?

I mean cmd.exe , and yes, the gcc command is found so it is in my PATH environment variable. If I type `echo %PATH%` in cmd.exe the output is:
```
C:\Users\jonat>echo %PATH%
D:\Program Files\RoboDK\Python37\Scripts\;D:\Program Files\RoboDK\Python37\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Java\jdk-13.0.1\bin;C:\Program Files (x86)\Microchip\xc8\v1.43\bin;C:\Program Files\Git\cmd;C:\Program Files\PuTTY\;D:\Program Files\MATLAB\R2019b\bin;C:\Program Files (x86)\Wolfram Research\WolframScript\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\110\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\DTS\Binn\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\MinGW\bin;D:\ProgramCC\Files\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files\dotnet\;C:\msys64\mingw64\bin;D:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin;C:\MinGW\mingw64\bin;C:\MinGW\msys\1.0\bin;C:\MinGW\bin;C:\MinGW\mingw64\x86_64-w64-mingw32\bin;C:\Users\jonat\AppData\Local\Microsoft\WindowsApps;C:\Users\jonat\.dotnet\tools;D:\Users\jonat\AppData\Local\Programs\Microsoft VS Code\bin
```

Well, in any case, it works now, thank you again!

stinos, 2021-02-09:
> Now it works, I don't know what happened

Is it possible earlier you didn't use the CROSS_COMPILE flag? Could you try rebuilding mpy-cross without it and see what happens? Would be great if we can sort out what happens exactly here.

> x86_64-w64-mingw32-gcc -v
Using built-in specs.
COLLECT_GCC=C:\msys64\mingw64\bin\x86_64-w64-mingw32-gcc.exe

Ok, so that is actually the mingw-w64 installed by MSYS2 which is what you intend to use (I think), and which is why compilation with CROSS_COMPILE  works ok

> echo $PATH

There are some peculiarities here; not saying it's a problem per se, but you should be aware:
- `C:/cygdrive/d/Program Files/RoboDK/Python37` you say you have Python 3.9 installed, but note that it will not be used on the commandline, instead python.exe from RoboDK will be used (assuming Python37 is a Python distribution, which is likely)
- `/cygdrive/c/MinGW/bin` that does not look ok (it's mingw32 probably, not supported by MicroPython), also because of the next one
- `/cygdrive/c/msys64/mingw64/bin` is (probably) what allows to to just run `gcc` on a commandline and get the same as x86_64-w64-mingw32-gcc.exe. But it is tricky because when in a Cygwin shell, there's also the gcc which comes with Cygwin, and the one from mingw32, so depending on what comes first in the PATH or through which application this gets invoked, one or the other will be used. Not recommended.
- `/cygdrive/c/MinGW/mingw64/x86_64-w64-mingw32/bin` also does not look correct and/or might not do what you expect

tldr; I assume items are added to the global PATH environment variable, but instead it is better to not do that and work in a shell in which you only add the correct items (e.g. by adding lines like `export PATH=...:$PATH` in .bashrc) , which then stays local to that shell

JonatanESalinas, 2021-02-09:
> Is it possible earlier you didn't use the CROSS_COMPILE flag?

Yes, at the beginning I was only typing in Cygwin `make STRIP=echo SIZE=echo` in the mpy-cross folder of Micropython, which didn't work. Then I typed `make CROSS_COMPILE=x86_64-w64-mingw32-` without the `STRIP=echo SIZE=echo` part, which didn't work either. When I finally typed `make STRIP=echo SIZE=echo CROSS_COMPILE=x86_64-w64-mingw32-` the build began and finished successfully. Right now I repeated the commands in a new copy of the Micropython-1.14 folder, and the same happened. So `make STRIP=echo SIZE=echo CROSS_COMPILE=x86_64-w64-mingw32-` is the command that make it works.

> * also does not look correct and/or might not do what you expect

Yes, I was supposing something was not quite right there... actually I began adding stuff to the PATH trying to make the whole thing work, but I did strange things.

> work in a shell in which you only add the correct items

That sounds better, I'll try to do that from now and so on, thank you for your recommendations.

stinos, 2021-02-09:
@JonatanESalinas ok thanks! And, sorry to bother you again, but I'm trying to reproduce this without success so far; what's the output of `which python` and `python --version` ? Same question for `which python3` and `python3 --version`?

JonatanESalinas, 2021-02-10:
If I type `which python` in Cygwin the output is:
```
$ which python
/cygdrive/c/msys64/mingw64/bin/python
```
And `python --version`:
```
$ python --version
Python 3.8.7
```
An then: `which python3`:
```
$ which python3
/cygdrive/c/msys64/mingw64/bin/python3
```
And `python3 --version`:
```
$ python3 --version
Python 3.8.7
```

It's interesting... that means Python is coming from... mingw-w64 or something like that? The one I installed from MYSYS2?
It is true that I had installed Python 3.8 before (I downloaded it time ago, from the official page), and when I was trying to build Micropython without success I uninstalled that Python 3.8 and I installed Python 3.9.1 from the Microsoft Store, hoping that could help. But it seems I am not using Python 3.9.1 anywhere here.

stinos, 2021-02-10:
> It's interesting... that means Python is coming from... mingw-w64 or something like that?

Yes: that directory is in your PATH so if it has python3.exe in it and none of the other directories before it do, it will be used (so my assumption that it would be using RoboSDK is wrong, perhaps because D:\Program Files\RoboDK\Python37\ is a Windows Python installation meaning it will only have python.exe, not python3.exe.)

> The one I installed from MYSYS2?

Yes. But, not just any python: if you install python3 with pacman you get the MSYS2 version which ends up in msys64/usr/bin. And that one cannot be called from Cygwin (results in an error about cygwin1.dll).
However if you use `pacman -S  mingw64/mingw-w64-x86_64-python` you get a windows version of python in msys64/mingw64/bin. I didn't know that until now, and it's also why I couldn't reproduce the issue apparently. With this python version I can reproduce the problem, yay :)

So: in a situation like yours, calling `make` in the mpy-cross directory results in calling `python3 makeqstrdefs.py` and that is msys64/mingw64/bin/python3.exe, which executes `os.subprocess(['gcc'])`. However because of standard lookup rules, gcc.exe will be found as msys64/mingw64/bin/gcc.exe i.e. from the same directory where python.exe is. No big deal normally, except something in gcc goes wrong here. I'd wager it is a bug, not sure, point is gcc messes up the paths it uses:

```
>>> import subprocess
>>> subprocess.check_output(['gcc', '-E', '-v', '-###', '../py/mpstate.c'])
...
COMPILER_PATH=C:/cygwin64/bin/../lib/gcc/
LIBRARY_PATH=C:/cygwin64/bin/../lib/gcc/;C:/cygwin64/bin/../lib/gcc/x86_64-w64-mingw32/10.2.0/../../../../lib/;C:/cygwin64/bin
/../lib/gcc/x86_64-w64-mingw32/10.2.0/../../../
```

that is quite wrong, beacuse those paths do not exist and also on my machine gcc is c:/tools/msys2/msys64/mingw64/bin/gcc.exe, so why would it use paths relative to another root directory? For comparison, this does work correctly:

```
>>> subprocess.check_output(['x86_64-w64-mingw32-gcc', '-E', '-v', '../py/mpstate.c'])
...
COMPILER_PATH=C:/tools/msys2/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.2.0/;C:/tools/msys2/msys64/mingw64/bin/../lib
/gcc/;C:/tools/msys2/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.2.0/../../../../x86_64-w64-mingw32/bin/
LIBRARY_PATH=C:/tools/msys2/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.2.0/;C:/tools/msys2/msys64/mingw64/bin/../lib/
gcc/;C:/tools/msys2/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.2.0/../../../../x86_64-w64-mingw32/lib/../lib/;C:/tool
s/msys2/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.2.0/../../../../lib/;C:/tools/msys2/msys64/mingw64/bin/../lib/gcc/
x86_64-w64-mingw32/10.2.0/../../../../x86_64-w64-mingw32/lib/;C:/tools/msys2/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/
10.2.0/../../../
```

tldr; while this specific issue indeed gets introduced by the PR mentioned above, in my opinion it is out of the scope of MicroPython to attempt to fix this

Krasn4ck, 2021-02-11:
> If you run make without specifycing the compiler, it will try to use gcc as compiler. However you say you installed MinGW but if you do that (in Cygwin) that doesn't install gcc but e.g. [...]

Thank you @stinos what @JonatanESalinas has put is what we both had in each computer, we are very grateful for your help.

closed 2021-02-11 by Krasn4ck

QGB, 2022-09-28:
use dockcross

```
[root:/work/ports/windows] # make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build-standard/genhdr
GEN build-standard/genhdr/mpversion.h
GEN build-standard/genhdr/qstr.i.last
i686-w64-mingw32.static-cpp: fatal error: too many input files
compilation terminated.
i686-w64-mingw32.static-cpp: fatal error: too many input files
compilation terminated.
i686-w64-mingw32.static-cpp: fatal error: too many input files
compilation terminated.
Traceback (most recent call last):
  File "/work/ports/windows/../../py/makeqstrdefs.py", line 210, in <module>
    preprocess()
  File "/work/ports/windows/../../py/makeqstrdefs.py", line 76, in preprocess
    for output in p.imap(pp(flags), chunks):
  File "/usr/lib/python3.9/multiprocessing/pool.py", line 870, in next
    raise value
  File "/usr/lib/python3.9/multiprocessing/pool.py", line 125, in worker
i686-w64-mingw32.static-cpp: fatal error: too many input files
compilation terminated.
    result = (True, func(*args, **kwds))
  File "/work/ports/windows/../../py/makeqstrdefs.py", line 60, in run
    return subprocess.check_output(args.pp + flags + files)
  File "/usr/lib/python3.9/subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.9/subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/usr/src/mxe/usr/bin/i686-w64-mingw32.static-cpp', '-I.', '-I../..', '-Ibuild-standard', '-Ivariants/standard', '-Wall', '-Wpointer-arith', '-Wdouble-promotion', '-Werror', '-std=gnu99', '-DUNIX', '-D__USE_MINGW_ANSI_STDIO=1', '-DFFCONF_H="lib/oofatfs/ffconf.h"', '-DMICROPY_USE_READLINE=1', '-Os', '-DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool', '-DMICROPY_MODULE_FROZEN_MPY=1', '-DMPZ_DIG_SIZE=16', '-DNO_QSTR', '../../py/mpstate.c', '../../py/malloc.c', '../../py/gc.c', '../../py/pystack.c', '../../py/qstr.c', '../../py/vstr.c', '../../py/mpprint.c', '../../py/unicode.c', '../../py/mpz.c', '../../py/reader.c', '../../py/lexer.c', '../../py/parse.c', '../../py/scope.c', '../../py/compile.c', '../../py/emitcommon.c', '../../py/emitbc.c', '../../py/asmbase.c', '../../py/asmx64.c', '../../py/emitnx64.c', '../../py/asmx86.c', '../../py/emitnx86.c', '../../py/asmthumb.c', '../../py/emitnthumb.c', '../../py/emitinlinethumb.c', '../../py/asmarm.c', '../../py/emitnarm.c', '../../py/asmxtensa.c', '../../py/emitnxtensa.c', '../../py/emitinlinextensa.c', '../../py/emitnxtensawin.c', '../../py/formatfloat.c', '../../py/parsenumbase.c', '../../py/parsenum.c', '../../py/emitglue.c', '../../py/persistentcode.c', '../../py/runtime.c', '../../py/runtime_utils.c', '../../py/scheduler.c', '../../py/nativeglue.c', '../../py/pairheap.c', '../../py/ringbuf.c']' returned non-zero exit status 1.
make: *** [../../py/mkrules.mk:111: build-standard/genhdr/qstr.i.last] Error 1
make: *** Deleting file 'build-standard/genhdr/qstr.i.last'
```

stinos, 2022-09-29:
> i686-w64-mingw32.static-cpp

That's not what is normally there (would be `gcc -E`), so to be able to reproduce this you should post more information. Would be better to open a new issue though because the error ('too many input files') seems unrelated to this issue.

QGB, 2022-09-29:
@stinos https://github.com/micropython/micropython/issues/7894#issuecomment-1261755929

stinos, 2022-09-29:
Before altering the makefiles, try to modify flags by passing them on the commandline. That's also easier to report 'I'm using `make CC=mingw-`' than having to say 'I changed line XXX in file xyz', also because linenumbers change depending on source code versions.

Anyway: changing $(CPP) into $(CC) is not correct, those are 2 different things, so now you have errors completely unrelated to the first one because the preprocessing stage is now compiling instead of just preprocessing.