micropython/micropython #6852: RP2: Support reading the BOOTSEL button on the Pico board

zmarties, 2021-02-05:
The published C-SDK examples already include code that allows the BOOTSEL button to be read.

Please add support for doing this from MicroPython.

The code has already been proved to work when added as a MircroPython C-Extension, as given at https://www.raspberrypi.org/forums/viewtopic.php?f=146&t=300352#p1813209

Note that there may be one change needed to this code before it is production ready.  Since the code runs with interrupts disabled, it has to use a tight loop to introduce a delay in to the process.  The example code published in C has this loop limit as a hard-coded value, whereas it should really base the limit on the current (possibly overclocked) speed of the processor.

tjvr, 2021-02-05:
What would be the desired API for this feature?

davehylands, 2021-02-05:
MicroPython has a Switch class: http://docs.micropython.org/en/latest/library/pyb.Switch.html which supports a single switch. I'd like to see the constructor extended to take a switch number (much like the LED class: http://docs.micropython.org/en/latest/library/pyb.LED.html) but since there is only a single switch it could use the Switch API as-is.

founfabug, 2021-04-07:
This would be a very useful feature.

CorruptName, 2022-08-13:
This would be very nice. Seems someone has done some work here:
[(https://forums.raspberrypi.com/viewtopic.php?t=321536)]

I can't get it to work. (I guess I don't know what I'm doing)
but I always get"
`AttributeError: 'module' object has no attribute 'button'`
would be better to have this functionality integrated anyways.

pdg137, 2022-12-13:
It would be great to have the PR merged into the Micropython core.  But for now I have been using this inline assembly to do it, which seems to work fine at least with single-core code:

```python
@micropython.asm_thumb
def read_bootsel():
    # disable interrupts
    cpsid(0x0)
    
    # set r2 = addr of GPIO_QSI_SS registers, at 0x40018000
    # GPIO_QSPI_SS_CTRL is at +0x0c
    # GPIO_QSPI_SS_STATUS is at +0x08
    # is there no easier way to load a 32-bit value?
    mov(r2, 0x40)
    lsl(r2, r2, 8)
    mov(r1, 0x01)
    orr(r2, r1)
    lsl(r2, r2, 8)
    mov(r1, 0x80)
    orr(r2, r1)
    lsl(r2, r2, 8)
    
    # set bit 13 (OEOVER[1]) to disable output
    mov(r1, 1)
    lsl(r1, r1, 13)
    str(r1, [r2, 0x0c])
    
    # delay about 3us
    # seems to work on the Pico - tune for your system
    mov(r0, 0x16)
    label(DELAY)
    sub(r0, 1)
    bpl(DELAY)
    
    # check GPIO_QSPI_SS_STATUS bit 17 - input value
    ldr(r0, [r2, 0x08])
    lsr(r0, r0, 17)
    mov(r1, 1)
    and_(r0, r1)
    
    # clear bit 13 to re-enable, or it crashes
    mov(r1, 0)
    str(r1, [r2, 0x0c])
    
    # re-enable interrupts
    cpsie(0x0)
```

Maybe that can help some people who come here looking for a solution.  I would also be happy to have feedback since this is the first thing I've attempted in ARM assembly.

cfreshman, 2023-02-26:
> It would be great to have the PR merged into the Micropython core. But for now I have been using this inline assembly to do it, which seems to work fine at least with single-core code

@pdg137 I packaged this into a git repo here https://github.com/cfreshman/pico-bootsel as a bootsel.py file people can drop into their MicroPython projects for now. Let me know if this is alright or if you'd like different attribution
```python
import time, bootsel
while True:
  print('BOOTSEL', bootsel.pressed())
  time.sleep(1)
```

dpgeorge, 2023-03-09:
`rp2.bootsel_button()` has been implemented in f80d040c038c343b0709eba537014fb52bc8115e

closed 2023-03-09 by dpgeorge

JoGeDuBo, 2025-01-31:
**rp2.bootsel_button() not working @ Pico2** 

```
import sys, time, rp2
print(f"{sys.implementation._machine=}\n  {sys.version=}\n")
print(f"press bootsel_button")
for i in range(5, 0, -1):
    print(f"{i:2} {rp2.bootsel_button()=}")
       # result @ Pico as expected: 1 if button pressed, else 0
       # result @ Pico2: always 1
    time.sleep(1)
```

tested with:

working @ Pico:
sys.implementation._machine=Raspberry Pi Pico with RP2040
  sys.version=3.4.0; MicroPython v1.24.1 on 2024-11-29

not working @ Pico2:
sys.implementation._machine=Raspberry Pi Pico2 with RP2350          
  sys.version=3.4.0; MicroPython v1.24.1 on 2024-11-29

sys.implementation._machine=Raspberry Pi Pico2 with RP2350           
  sys.version=3.4.0; MicroPython v1.25.0-preview.236.gbfb1bee6f on 2025-01-29

sys.implementation._machine=Raspberry Pi Pico2 with RP2350-RISCV     
  sys.version=3.4.0; MicroPython v1.25.0-preview.236.gbfb1bee6f on 2025-01-29