micropython/micropython #6853: PI Pico infinite loop made device unusable

katurov, 2021-02-05:
If main.py contains `while True:` without delays (as an error or an idea to make it hot) device disappears from virtual ports and cannot be used in common ways. 

1. Put micropython .uf2 to Pico
2. Write dummy main.py with `while True : 1 = 1` and save in as "file" in Pico's memory
3. It hangs, Pico lost from devices' list
4. Load in BOOTSEL - works fine with any other .uf2
5. Put micropython .uf2 back and Pico will hangs again on main.py

The found way: to save dump of embedded drive from Pico, edit it in HEX editor, upload it back to Pico

Proposal: may be some watchdog to prevent hanging?

dpgeorge, 2021-02-05:
>  Write dummy main.py with `while True : 1 = 1`

Note that `1 = 1` is invalid Python and raises a syntax error.

I cannot reproduce the above problem.  With a main.py containing that exact code, the device appears correctly as a serial USB device, and the prompt is immediately available because main.py finishes with a SyntaxError.

Changing main.py to `while True: x = 1` and it also works as expected, it runs main.py forever but the USB serial is still there and you can ctrl-C to break the loop and get a REPL.

katurov, 2021-02-06:
sorry, the "full code"

```from machine import Pin, Timer
led = Pin(25, Pin.OUT)

i = 3

while True:
   i = (i * i) - i
   led.toggle()
```

save it as main.py to the board, reboot, starting from this point OSX (10.15.7) cannot find connected board till BOOTSEL mode. It IS possible to put any other .uf2 (e.g. blink) and Pico works fine. But as soon as you will put micropython uf2 back the Pico goes hang until you manage data in embedded drive area ("delete" file with procedure: save bin from, edit in hex, load bin to Pico). By micropython uf2 I mean `rp2-pico-20210205-unstable-v1.14-8-g1f800cac3.uf2` downloaded yesterday from official link (and yes, I see "unstable" in file name).

I have stored .bin file which is dump of embedded drive if you are interrested

dpgeorge, 2021-02-06:
Ah, seeing the full code does help in this case.

The problem is that the `i = (i * i) - i` expression grows exponentially fast.  After 13 loops `i` is already of order `10**3000`.  So the loop quickly grinds to a halt because it allocates exponentially more memory each time to store the new `i`, and also takes exponentially more time to compute the next `i`.  The USB code doesn't get a chance to run.

robert-hh, 2021-02-06:
Down that page: https://www.raspberrypi.org/documentation/pico/getting-started/
there is a link to an .uf2 image which wipes out the flash memory. That is good in cases where autostarted code blocks the device. Easier than hex-editing the image file.

katurov, 2021-02-06:
Many tanks for your link and pointing. This is really great. And my ticket is solved here.

One thing last on the table: when you done with the file and "file save" dialog of Thonny appear, you have to choose one of: "On the computer" or "On the device". This leads to storing all programs on the Pico, so... hard choose before the user: to lost all OR to edit HEX. If this question is on, user is not this kind to edit HEX. I suppose to write something to see files and manage it in "DOS-like mode" (but firstly: I would like to search for it).

closed 2021-02-06 by katurov

dpgeorge, 2021-02-06:
> One thing last on the table ...

Best to discuss that in the Thonny repo: https://github.com/thonny/thonny

katurov, 2021-02-06:
> Best to discuss that in the Thonny repo: https://github.com/thonny/thonny

This is why I've close this ticket, thanks, Damien!