micropython/micropython #6855: rp2 API documentation

tjvr, 2021-02-06:
👋 Is the `rp2` port documented anywhere? In particular, the rp2-specific features like PIO.

I realise there are a bunch of examples provided by the Raspberry Pi people; I'm looking specifically for "API reference"-style documentation, e.g. a list of all possible arguments to `StateMachine`. The only reference I've managed to find so far is the C source code, which while I'm fine with that it's not the most beginner-friendly :)

At [docs.micropython.org](http://docs.micropython.org/en/latest/) I see there is port-specific documentation for some of the chips, but not the RP2040/Pi Pico.

Is this documented somewhere and I missed it? Are there already plans to write documentation like this? Or is this an area where we're seeking contributions?

Cheers!

dpgeorge, 2021-02-06:
> Is the `rp2` port documented anywhere? In particular, the rp2-specific features like PIO.

There are generic docs for modules shared among ports, but at the moment no docs for rp2-specific features.  These need to be added!

Anyone is most welcome to contribute.  I'd suggest starting by adding `docs/library/rp2.rst` to document the `rp2` module.

enesbcs, 2021-03-24:
There are some examples in the Raspberrypi Pi Python SDK:
https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-python-sdk.pdf

tjvr, 2021-03-27:
I've started writing a `docs/library/rp2.rst` today. :)

nameoftherose, 2021-06-14:
In file `https://github.com/micropython/micropython/blob/master/docs/library/rp2.rst` the link for `class: rp2.StateMachine`. is not pointing to `https://github.com/micropython/micropython/blob/master/docs/library/rp2.StateMachine.rst` as it should.
Thank you for providing documentation.

mattytrentini, 2022-03-07:
Given rp2 documentation was added as part of the GSoD can this ticket be closed? (A good start is the [quickref](https://docs.micropython.org/en/latest/rp2/quickref.html))

@nameoftherose, I'm not sure if the link has been updated in the interim but I think you were referring to the reference to `rp2.StateMachine` in ["For running PIO programs, see rp2.StateMachine"](https://docs.micropython.org/en/latest/library/rp2.html#pio-related-functions)? It does look like it's correct now.

dpgeorge, 2022-03-07:
I'll close this.  If there are any particular parts of the rp2 API that are not documented then separate tickets can be opened for them.

closed 2022-03-07 by dpgeorge