micropython/micropython #6856: rp2 I2C start, stop functions not supported

kiwih, 2021-02-06:
In the MicroPython documentation the machine.I2C describes under primitive functions the start() and stop() functions.
https://docs.micropython.org/en/latest/library/machine.I2C.html

However, these are not supported on the RP2:
```
MicroPython v1.14 on 2021-02-05; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>>from machine import I2C
>>> i2c = I2C(1, scl=Pin(19), sda=Pin(18), freq=100000)
>>> i2c.start()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: I2C operation not supported
```

Are these supposed to be excluded? If so, what is the correct way to manage low level I2C transactions without these functions? I am currently writing a driver for an AT24 EEPROM, and in order to manage the chip efficiently I need to be able to control the issuing of start/stop sequences.

kiwih, 2021-02-06:
Okay, I don't love it, but I solved my problem without the low level functions:

(relevant snippet)
```python
class AT24():
    def __init__(self, A2_value, i2c):
        self.A2_value = A2_value
        self.i2c = i2c
        
    def write_bytes(self, address, value):
        """Writes up to 16 bytes starting at 10-bit address [address]. All bytes must be on the same page boundary (that is, A[9]-A[4] should be constant)."""
        i2c_addr = AT24_ROOT_ADDR << 3 | self.A2_value << 2 | (address & 0x300) >> 8
        mem_addr = address & 0xFF
        
        #The proper way to check if the AT24 chips are ready is to poll the write address and check the NACK.
        #Unfortunately, this isn't possible in the RP2040 port of MicroPython as the low-level I2C functions are not supported
        #Instead, we have to do this using exceptions (an OSError is thrown if the AT24 NACKS)
        #So, this works, but is less efficient than it could be
        attempts = 0
        while attempts < 50: 
            attempts = attempts + 1
            try:
                self.i2c.writeto_mem(i2c_addr, mem_addr, value)
                break
            except OSError:
                pass
        
        if attempts >= 50:
            raise AT24Exception("AT24: Write operation timed out (check the address?)")
```

I still feel that the low level functions should be supported, especially since the documentation doesn't say that they're optional.
Alternatively, the documentation could be updated to say that they are optional.

dpgeorge, 2021-02-06:
> Are these supposed to be excluded?

Yes.  The docs state that these methods are only available on `machine.SoftI2C` instances.  The reason is that most MCU I2C hardware peripherals cannot issue independent start/stop conditions (they require at least one byte to be read/written).

> If so, what is the correct way to manage low level I2C transactions without these functions?

It should be possible to do almost all I2C operations using the "standard bus operations" and their `stop` arguments.  Note that the write methods return the number of ACKs that were received.

If you can't write an efficient driver for the AT24 using these standard bus operations, please report here, that would be interesting to know.

kiwih, 2021-02-07:
> The docs state that these methods are only available on machine.SoftI2C instances. 

My apologies, I somehow missed that sentence! I'll have a further play around.

closed 2021-02-07 by kiwih