micropython/micropython #6858: UART error with MicroPython 1.14 (ESP32)

teuler, 2021-02-07:
Hi,

I am running code that uses two UARTs (channels 1+2), which worked with 1.13 (EPS32, TinyPICO) w/o any problems. When I switched to 1.14, I got the following errors:

> E (32173) uart: uart_wait_tx_done(1040): uart driver error
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4011d3ff  PS      : 0x00060430  A0      : 0x800f8a9d  A1      : 0x3ffce4e0  
A2      : 0x000000c6  A3      : 0x3ff4904c  A4      : 0x00000010  A5      : 0xffffffff  
A6      : 0xffffffff  A7      : 0x000000c7  A8      : 0x0004904c  A9      : 0x00013ffc  
A10     : 0x3ff44574  A11     : 0x000000c6  A12     : 0x0ffd114c  A13     : 0x00000000  
A14     : 0x3ffbdc38  A15     : 0x000000c7  SAR     : 0x0000000f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000b  LBEG    : 0x40097c61  LEND    : 0x40097c71  LCOUNT  : 0xfffffffc  
>
>ELF file SHA256: 766b6f6e732a22222222222222227367ffffffffffffffffffffffffffffffff
>
>Backtrace: 0x4011d3ff:0x3ffce4e0 0x400f8a9a:0x3ffce510 0x400e94ed:0x3ffce560 0x400df799:0x3ffce580 0x400ed945:0x3ffce5a0 0x400e3b38:0x3ffce640 0x400df799:0x3ffce670 0x400e2081:0x3ffce690 0x400df799:0x3ffce6e0 0x400df8d5:0x3ffce700 0x400ed9cb:0x3ffce720 0x400e3b38:0x3ffce7c0 0x400df799:0x3ffce840 0x400e2081:0x3ffce860 0x400df799:0x3ffce8b0 0x400df8d5:0x3ffce8d0 0x400e98fe:0x3ffce8f0 0x400e94ed:0x3ffce940 0x400df799:0x3ffce960 0x400ed945:0x3ffce980 0x400e3b38:0x3ffcea20 0x400df799:0x3ffcea50 0x400df7c2:0x3ffcea70 0x4011a882:0x3ffcea90 0x4011abf5:0x3ffceb20 0x4011ac53:0x3ffceb40 0x400f72db:0x3ffceb60 0x400930ba:0x3ffceb90
>
>Rebooting...

Sometimes, I see this here:

>E (62318) uart: uart_wait_tx_done(1040): uart driver error
assertion "mp_obj_is_dict_or_ordereddict(MP_OBJ_Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x400f7713  PS      : 0x00060034  A0      : 0x800f7ade  A1      : 0x3ffbe980  
A2      : 0x00000000  A3      : 0x3f844db0  A4      : 0xfffffff8  A5      : 0x3ffce000  
A6      : 0x00000000  A7      : 0xffffffff  A8      : 0x40095094  A9      : 0x3ffbe980  
A10     : 0x3ffba0a0  A11     : 0x3ffbe980  A12     : 0x00060023  A13     : 0xffffffcd  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000001e  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40097c61  LEND    : 0x40097c71  LCOUNT  : 0xfffffffc  
Core 0 was running in ISR context:
EPC1    : 0x40093d09  EPC2    : 0x4000bff0  EPC3    : 0x4012b7b7  EPC4    : 0x400f7713
>
>ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000
>
>Backtrace: 0x400f7713:0x3ffbe980 0x400f7adb:0x3ffbe9b0 0x40086e35:0x3ffbe9d0 0x4012b7b4:0x3ffce060 0x4012c1a5:0x3ffce080 0x4009a199:0x3ffce0b0 0x4000bd83:0x3ffce0d0 0x400961cb:0x3ffce0f0 0x40096cc9:0x3ffce110 0x40096d45:0x3ffce130 0x4009793d:0x3ffce150 0x4018c69d:0x3ffce180 0x4018eae5:0x3ffce1b0 0x40187125:0x3ffce470 0x401870ad:0x3ffce4c0 0x400e8e0b:0x3ffce4f0 0x400e9152:0x3ffce510 0x400dfb05:0x3ffce570 0x400ecfae:0x3ffce5a0 0x400e3b38:0x3ffce640 0x400df799:0x3ffce670 0x400e2081:0x3ffce690 0x400df799:0x3ffce6e0 0x400df8d5:0x3ffce700 0x400ed9cb:0x3ffce720 0x400e3b38:0x3ffce7c0 0x400df799:0x3ffce840 0x400e2081:0x3ffce860 0x400df799:0x3ffce8b0 0x400df8d5:0x3ffce8d0 0x400e98fe:0x3ffce8f0 0x400e94ed:0x3ffce940 0x400df799:0x3ffce960 0x400ed945:0x3ffce980 0x400e3b38:0x3ffcea20 0x400df799:0x3ffcea50 0x400df7c2:0x3ffcea70 0x4011a882:0x3ffcea90 0x4011abf5:0x3ffceb20 0x4011ac53:0x3ffceb40 0x400f72db:0x3ffceb60 0x400930ba:0x3ffceb90
>
>Core 1 register dump:
PC      : 0x401e2d16  PS      : 0x00060634  A0      : 0x801287c6  A1      : 0x3ffbc430  
A2      : 0x3ffc2de9  A3      : 0x00060620  A4      : 0x800932b3  A5      : 0x3ffbc360  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x801296e0  A9      : 0x3ffbc420  
A10     : 0x00000003  A11     : 0x00060623  A12     : 0x00060620  A13     : 0x0000abab  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001d  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
>
>ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000
>
>Backtrace: 0x401e2d16:0x3ffbc430 0x401287c3:0x3ffbc450 0x4009448d:0x3ffbc470 0x400930ba:0x3ffbc4a0
>
>Rebooting...

I think it happens when initiating the second UART, no matter if it is channel 1 or 2. 
I define the UART objects using :
`
self._uart = UART(ch, baudrate=baud, tx=_tx, rx=_rx)
`
with _tx=21, _rx=22, baudrate=115200 for the first UART and
with _tx=14, _rx=4, baudrate=115200  for the second UART.

Any help is appreciated.

Thanks
Thomas

tve, 2021-02-07:
Are you using the published v1.14 firmware for TinyPico?
Damien, can you publish the elf files so one can do a backtrace?

teuler, 2021-02-07:
No, I built them as I need to include _ulab_. But I can test the prebuilt firmware.

teuler, 2021-02-10:
I tested a bit more - same firmware and conditions as described above. 
The issue seems to occur when a timer is used as well:
```
from machine import UART, Timer
import time
use_timer = True

def _cb(value):
  print(value)

try:
  u1 = UART(2, baudrate=115200, tx=21, rx=22)
  u2 = UART(1, baudrate=115200, tx=14, rx=4)
  if use_timer:
    t1 = Timer(1)
    t1.init(period=50, mode=Timer.PERIODIC, callback=_cb)
  print(u1)
  print(u2)
  print("Wait ...")
  time.sleep_ms(3000)
finally:
  t1.deinit()
  print("done")
```
... results in 
```
Traceback (most recent call last):
  File "<stdin>", line 16, in <module>
OSError: (-261, 'ESP_ERR_NOT_FOUND')
```
With `use_timer = False`, the code runs fine, resulting in:
```
UART(2, baudrate=115201, bits=8, parity=None, stop=1, tx=21, rx=22, rts=-1, cts=-1, txbuf=256, rxbuf=256, timeout=0, timeout_char=1)
UART(1, baudrate=115201, bits=8, parity=None, stop=1, tx=14, rx=4, rts=-1, cts=-1, txbuf=256, rxbuf=256, timeout=0, timeout_char=1)
Wait ...
done
```
Defining the Timer seems fine; initializing the Timer causes the error.
After a hard reset, the code runs also with the Timer activated, producing the expected:
```
UART(2, baudrate=115201, bits=8, parity=None, stop=1, tx=21, rx=22, rts=-1, cts=-1, txbuf=256, rxbuf=256, timeout=0, timeout_char=1)
UART(1, baudrate=115201, bits=8, parity=None, stop=1, tx=14, rx=4, rts=-1, cts=-1, txbuf=256, rxbuf=256, timeout=0, timeout_char=1)
Wait ...
Timer(3f817f20; alarm_en=1, auto_reload=1, counter_en=1)
Timer(3f817f20; alarm_en=1, auto_reload=1, counter_en=1)
...
```
Then, at some point (possibly if the code is interrupted at some point or when I run my actual code, the OSError reappears.
A soft reboot does not help.

Runing my actual code after a hard reboot with the same sequence of events (defining the two UARTs, then the Timer) produces the same OSError as shown above, However, the sequence Timer, then the two UARTs results in these errors: 
```
E (18540) uart: uart_wait_tx_done(1040): uart driver error
E (22270) uart: uart_get_buffered_data_len(1266): uart driver error
E (22270) uart: uart_read_bytes(1208): uart driver error
```
... but w/o the crash I described above.
I am utterly confused.

teuler, 2021-02-13:
'''
from machine import Timer
t = Timer(12345)
t.init(period=500, mode=Timer.PERIODIC, callback=lambda t:print(0))
'''
... does not cause an error - which Timer is it using?

teuler, 2021-02-13:
I think the UART problems I experience are related to the Timers. If I don't use Timer, the UARTs work as expected.

mraess, 2021-02-20:
@teuler I think this has been discussed/solved here

https://forum.micropython.org/viewtopic.php?t=9746&p=54541

I had the same issue with my ESP32 and the suggested code switch fixed the problem for me. Now everything runs fine with MicroPython 1.14

In my code, I switched

```
from machine import Pin, UART, I2C
import time
import urandom as random
import uasyncio

uart = UART(1)
uart.init(baudrate = 9600, bits = 8, parity = None, stop = 1, tx = 25, rx = 26)
```

to

```
from machine import Pin, UART, I2C
import time
import urandom as random
import uasyncio

uart = UART(1, baudrate = 9600, tx = 25, rx = 26)
uart.init(bits = 8, parity = None, stop = 1)
```

dpgeorge, 2021-02-20:
I tested this on a TinyPICO using both `tinypico-idf3-20210202-v1.14.bin` and `tinypico-idf4-20210202-v1.14.bin`.  I used the test code above, namely:
```

from machine import UART, Timer
import time
use_timer = True

def _cb(value):
  print(value)

try:
  u1 = UART(2, baudrate=115200, tx=21, rx=22)
  u2 = UART(1, baudrate=115200, tx=14, rx=4)
  if use_timer:
    t1 = Timer(1)
    t1.init(period=50, mode=Timer.PERIODIC, callback=_cb)
  print(u1)
  print(u2)
  print("Wait ...")
  time.sleep_ms(3000)
finally:
  t1.deinit()
  print("done")
```
I pasted that using ctrl-E paste mode at the REPL and it ran without error, printing out the timer in `_cb`.  I could paste it multiple times and it would run the same each time.  And also after a soft reset I could run it again and again.  On both idf3 and idf4 v1.14 builts it ran without any error.

@teuler are you able to provide code that fails on that pre-built v1.14 firmware?

teuler, 2021-02-20:
@dpgeorge Thank you for taking the time and testing the code.
@Matthias Raess: Thank you for your suggestion; it did not solve my problem.

As I wrote, it is difficult to reproduce the error reliably. I suggest to close this issue for now and I get back when I found out more.
Thanks again.

dpgeorge, 2021-02-20:
You might want to try using the very latest builds from https://micropython.org/download/tinypico/ that use IDF 4.2

teuler, 2021-02-20:
I need ulab included, therefor I need to build it myself. I will try this with IDF 4.2 later today.

teuler, 2021-06-10:
A quick update after several months: I revisited the UART problem I described above. It involved using two UARTs and a Timer. Initiating the Timer caused diverse problems (see above); apparently depending on the IDF version. With IDF 4.2, the result was the error message `OSError: (-261, 'ESP_ERR_NOT_FOUND')`, which pointed at missing resources.
I tested my code again with IDF 4.3-rc, and now the issues seem to be gone.

closed 2021-06-10 by teuler

dpgeorge, 2021-06-10:
Thanks for the update.