micropython/micropython #6859: esp32: Hardware SPI not working with non-standard Pins

kevinkk525, 2021-02-07:
I recently came across the issue that the Hardware SPI (HSPI/VSPI) on the esp32 don't work with non-standard Pins. Accordings to documentations the esp32 should have no problem with non-standard ports but it doesn't work. Standard ports however do work and the SoftwareSPI worked on the non-standard ports I tried.
Firmware: micropython 1.14

tve, 2021-02-07:
Can you provide an example, what you observe, steps to reproduce, ... ?

kevinkk525, 2021-02-07:
1. Create a hardware SPI object on non-standard pins, e.g. 

```
_miso = 23
_mosi = 14
_sck = 18
spi = machine.SPI(2, baudrate=8000000, mosi=Pin(_mosi), miso=Pin(_miso), sck=Pin(_sck))
```
or use id 1 for HSPI, same result.

2. Try using SPI devices 
3. Nothing works, no response, no communication arriving at the destination.
4. Use the same pins and configuration with SoftSPI and everything works

dpgeorge, 2021-03-12:
Using a TinyPICO board with pin 14 connected to pin 23, I ran the following test code:
```python
from machine import Pin, SPI
def test(n):
    for _ in range(n):
        for id in (1, 2):
            spi = SPI(id, baudrate=8000000, sck=Pin(18), mosi=Pin(14), miso=Pin(23))
            print(spi.read(4, id))
            buf = bytearray(4)
            spi.write_readinto(b'test', buf)
            print(buf)
test(4)
```
and the output is (after a clean reset):
```
b'\x01\x01\x01\x01'
bytearray(b'test')
b'\x02\x02\x02\x02'
bytearray(b'test')
b'\x00\x00\x00\x00'
bytearray(b'\x00\x00\x00\x00')
b'\x02\x02\x02\x02'
bytearray(b'test')
b'\x00\x00\x00\x00'
bytearray(b'\x00\x00\x00\x00')
b'\x02\x02\x02\x02'
bytearray(b'test')
b'\x00\x00\x00\x00'
bytearray(b'\x00\x00\x00\x00')
b'\x02\x02\x02\x02'
bytearray(b'test')
```
So it works the first time around, but then SPI(1) stops working.  It could be something to do with reinitialisation of the SPI, or the fact that pins are claimed by the other peripheral.

@kevinkk525 can you confirm that after a hard reset, if the SPI is only initialised once and nothing else is using the pins, then it does work?

jonnor, 2024-08-18:
This issue is missing information needed to reproduce. Proposing to close.

dpgeorge, 2024-08-19:
A lot of things have changed and  improved in the esp32 since this report was opened, so it may have been fixed in the meantime.

I'll close this issue, but feel free to reopen with a report of how things are broken in the latest esp32 firmware, if they are still broken.

closed 2024-08-19 by dpgeorge