micropython/micropython #6862: ESP32 UART 0 changing the baudrate

kdschlosser, 2021-02-07:
I am in the process of writing a replacement file manger for an ESP32 running MicroPython. The new file manager will be cross platform but for the time being I am working on getting it up and running on Windows. Th file manager I am writing leverages the Windows API using a kernel mode driver so the ESP32 will be seen as a removable drive to the OS. This has a slew of benefits like in place file editing and drag and drop using the OS's file manager. IDE's will be able to access the device directly. It will function the same way when I add Linux support to it.

Now that you know what I am doing this is the problem I am having..

using a 115200 bitrate to communicate over is simply not going to cut it. I need to bump that bitrate up to 460800. I also need to get rid of using the repl and sending python command over it to carry out tasks. This simply has to much overhead and I am not able to use multiple threads on the ESP32 very easily. I need to be able to send byte data to instruct the program on the ESP32.

How the file manager works is it backup up the current boot.py file and replace it with one that I made and contains all the relevant code needed for it to work. when the esp32 gets ejected/unmounted the program will place the original boot.py file back into place.

My issues are 

* I am not able to set the bitrate on UART 0. 
* UART 0 always gets bound to the repl

this is the default boot.py file and there are no lines in it that set the repl to UART 0 using uos.dupterm.
```python
# This file is executed on every boot (including wake-boot from deepsleep)
# import esp
# esp.osdebug(None)
# import webrepl
# webrepl.start()
```
has this been removed for some reason? and if so why????

kdschlosser, 2021-02-08:
OK so after some digging the ability to do pretty much anything with UART 0 has been disabled. This was done back in 2017

```c
// Attempts to use UART0 from Python has resulted in all sorts of fun errors.
    // FIXME: UART0 is disabled for now.
    if (uart_num == UART_NUM_0) {
        mp_raise_msg_varg(&mp_type_ValueError, MP_ERROR_TEXT("UART(%d) is disabled (dedicated to REPL)"), uart_num);
    }
```

mapping UART 1 to pins 1 and 3 does not work, Nor can stdin and stdout be shanghaied in a manner that would allow me to send raw byte data.. so no characters outside of the 32 to 127 range (or whatever it is), I cannot send 0x04 or 0x05. If I attempt to pass them to stdout they get converted to null characters. I have not been able to collect anything from stdin ever.. I am sending data to the ESP like gang busters but I get not one byte by using stdin.read(1)

I have seen it may time already about there being file managers available. OK so yeah. for linux. and there may be a command line based one that will run on Windows. ampy is a fine example of one, it only works about 5% of the time. I have not been able to locate a GUI based file manager that will run on Windows, so I am writing one. Because of the number of queries that Windows is going to do I have to shorten up the data that is being sent and speed it up huge amounts. Windows will query some 300+ times just to list the contents of a folder that has 20 files in it. Having it take minutes just to load the contents of a single folder that has 20 files in it is not going to work.

I need to be able to do this via the single USB connection, I need to have the baudrate  turned up as high as possible (Expressif states 5,000,000bps is the max) and I need to be able to send any character within the 0 - 255 range and not have the characters get converted into null characters.

If there is a way to do this without having to wiggle around the REPL and it capturing incoming data or anything like that, please help me out and share how to go about doing it.

ProtoSD, 2021-04-28:
I know this is a little late, but have you seen the section in here about 'dupterm'?
[https://docs.micropython.org/en/latest/library/uos.html](url)

kdschlosser, 2021-04-29:
I need to use UART 0. And there is no way to disconnect it from the repl and  change the baudrate on an esp32. I would have to have a cist firmware done for it to work. This is what I am trying to avoid having to do.

jabelone, 2021-08-02:
Would just like to say I came here looking for how to do this too. I have a PCB with a serial sensor hooked up to the hardware uart0. This sensor operates at 9600 baud so I need a way to change the baud rate the REPL operates at, or disable it so I can use the uart0 separately.

QGB, 2022-09-25:
@jabelone @kdschlosser \
How to do this in esp32c3 without recompile micropython firmware?

kdschlosser, 2022-09-26:
you can't

dpgeorge, 2022-09-30:
The ability to change the baudrate on UART(0) was added in 919e586e46af67e7eaa819b0f187b600a3165526

closed 2022-09-30 by dpgeorge

kdschlosser, 2023-02-28:
@dpgeorge 

I wanted to say TY for doing this. I am pretty sure I speak for the majority when I say that we appreciate you and all the effort you have put in to MicroPython. If people don't say this to you I want to make sure you understand it has not gone unnoticed. 

This change has made development a whole lot nicer. I am working with LVGL and MicroPython and uploading almost a megabyte of source code to the ESP32 has become tolerable. I was starting to have flash backs of downloading over a 14.4 modem. Being able to set the baudrate to the maximum the CP IC is able to handle has made for a HUGE improvement in the speeds. Going from115200 to 921200 is quite the jump. I do occasionally have an error when things are being sent maybe one out of every 50 times of uploading which is not bad.

Is there a way to set the default baudrate for the repl at compile time? It would make for one less step having to upload the boot file at the slower speed in order to get the higher speed to work.

lerouxb, 2023-03-03:
It would be really nice if `mpremote` could be made to operate at higher baud rates. Even with all the tricks [outlined here](https://github.com/micropython/micropython/discussions/9096) (ie. read most stable files from flash and only changing files from the remote mount) the "change a line of code, reload" loop still has room left for improvement.

george-hawkins, 2024-02-21:
Sorry, to comment on an issue that's been closed for some time now but it seems the most direct way to contact people like @kdschlosser who seem to have got the changes introduced in commit 919e586 to work for them. Sorry that the following comment is so long but I wanted to show I'd "done my homework" and also wanted to provide information for anyone else who runs across this issue.

First off - I have been able to change the baud rate for UART0 but things don't work quite as I expect.

The first issue is essentially cosmetic. I'm assuming I can only change the baud rate for UART0 by creating a new `machine.UART` instance that corresponds to it. I.e. there's no way to retrieve an existing `UART` instance, e.g. by doing something like:

```
uart0 = os.dupterm(None, 0)
```

`dupterm` just returns `None`.

OK. So, I create a new instance like this (leaving all the characteristics of the UART unchanged in this first step):

```
from machine import UART

UART(0)
```

This works, but creating the `UART` instance always results in:

```
E (738) uart: uart_wait_tx_done(1115): uart driver error
```

However, this appears to be cosmetic as this little program successfully changes the baud rate to 9600:

```
import os
import sys
import time
import micropython

from machine import UART

# Give myself a chance to bail and upload a new program.
print('Press ctrl-C now to exit')
time.sleep(3)
print("Taking control of the USB UART")
time.sleep_ms(200)

os.dupterm(None, 0)
UART(0, 9600, timeout=0)

buffer = memoryview(bytearray(1024))

micropython.kbd_intr(-1)

while True:
    read_count = sys.stdin.buffer.readinto(buffer[:1])
    if read_count is not None and read_count > 0:
        sys.stdout.buffer.write(buffer[:read_count])
    else:
        sys.stdout.buffer.write(b'.')
```

So, there are a couple of problems:

* As I'd set `timeout` to 0, I had hoped I'd see a stream of `.` characters when there wasn't anything to read but the `readinto` is blocking and I can't find any way to make it non-blocking.
* You don't see the `uart driver error` in this situation (I presume it somehow becomes "invisible" to `mpremote` in REPL mode due to the change in baud rate). But you see it e.g. if you use a terminal emulator, like the UNIX `screen` command, that, unlike `mpremote`, can be started with the baud rate set to 9600 and you then press the boards reset button:

```
$ screen /dev/ttyUSB0 9600
��B�����J1�9� �m�����h�ښ�j�(3769) uart: uart_wait_tx_done(1115): uart driver error
```

The initial bytes of jibberish are just the initial output after I pressed the reset button and things are still using 115200 baud.

I can get around the blocking issue like so using `select.poll`:

```
poller = select.poll()
poller.register(sys.stdin.buffer, select.POLLIN | select.POLLERR | select.POLLHUP)

while True:
    for (s, event) in poller.ipoll(0):
        if event != select.POLLIN:
            print(f"got unexprected event {event}")
        else:
            read_count = s.readinto(buffer[:1])
            if read_count is not None and read_count > 0:
                b = buffer[0]
                if b < b' '[0] or b > b'~'[0]:
                    print("?")
                else:
                    print(str(b))
            else:
                print("failed to write character")
```

But the `readinto` is still blocking so, I can only read one byte at a time, using `ipoll` to essentially tell me "yes, there's at least one byte available so you won't block if you call `readinto` for a single byte".

Really, I'd like to read as many bytes as are available in one go without blocking but I don't think I can get a byte count using `ipoll` and I don't see an equivalent of `UART.any()`.

That brings me to my final point - the above does get me where I want (despite the apparently harmless `uart driver error` when creating the `UART` instance).

But I had actually been hoping I could use the `UART` instance I'd created rather than having to work through `sys.stdin` and `sys.stdout`. But while creating the `UART` has achieved something - changing the baud rate of UART0 - the resulting `UART` instance doesn't seem to be useable. In the following program, I get:

**1.** `uart: uart_wait_tx_done(1115): uart driver error` on creating the `UART` (as before this seems to be harmless).
**2.** More lethally, when I call `uart.any()`, I get:

> ```
> E (3979) uart: uart_get_buffered_data_len(1340): uart driver error
> Traceback (most recent call last):
>   File "main.py", line 18, in <module>
> OSError: [Errno 1] EPERM: ESP_FAIL
> ```

```
import os
import time

from machine import UART

# Give myself a chance to bail and upload a new program.
print('Press ctrl-C now to exit')
time.sleep(3)
print("Taking control of the USB UART")
time.sleep_ms(200)

os.dupterm(None, 0)
uart = UART(0, baudrate=9600, timeout=0)

buffer = memoryview(bytearray(1024))

while True:
    available = uart.any()
    if available > 0:
        read_count = uart.readinto(buffer[:available])
        if read_count is not None and read_count > 0:
            write_count = uart.write(buffer[:read_count])
            if write_count != read_count:
                uart.write(b'?')
```

If I simply call `uart.write(b'?')`, it doesn't work either but it's not as lethal as calling `uart.any()`, it results in an error but doesn't terminate the program:

```
E (3959) uart: uart_write_bytes(1245): uart driver error
```

kdschlosser, 2024-02-22:
I made a file called boot.py and I have this in it.

```python
import machine

uart = machine.UART(0, baudrate=921600)
```

I use something like APY to upload the file to the ESP32. you have to remember at this point the uart speed is still set to 115200 so you need to upload the boot.py file at that speed. Once you do that and you restart the ESP32 then you will have the higher speed  available. In my example it's 921600. I am able to use ampy and upload the rest of my project or when I am connecting to use the repl.

it being in boot.py might be the trick to getting it done. Not sure.

george-hawkins, 2024-02-22:
Thanks for your reply, Kevin 👍 

When MicroPython and its REPL start-up, the UART associated with stdin and stdout only exists at the C level underlying the ESP32 MicroPython implementation.

Creating a `machine.UART` instance corresponding to UART0 (as you've done in `boot.py` and I've done in my examples) doesn't currently create a valid `machine.UART` object that can then be used directly in ones code. Instead, calling the `machine.UART` constructor for UART0 essentially just communicates a set of changes to its configuration (such as baud rate) to the underlying C code (which is then reflected in how stdin and stderr behave).

I wish I knew if this was deliberate when @pi-anl made his changes in commit 919e586, i.e. that it's just about communicating configuration changes to the C level and not about creating a valid `machine.UART` object that one can interact with in ones MicroPython code. If so, fair enough, I'm just hoping for some clarity.

So, in summary, I'm curious to find out:

1. Is it expected that the resulting `machine.UART` instance should be unusable in terms of interacting with it in MicroPython code?
2. If one must instead work with stdin/stdout, can one use stdin in a non-blocking manner? Or is there an equivalent of [`machine.UART.any()`](https://docs.micropython.org/en/latest/library/machine.UART.html#machine.UART.any). Posts like this [one](https://forum.micropython.org/viewtopic.php?t=12162) seem to suggest the best one can do is a poll and read **per** character (rather than a non-blocking read of as many as are currently available). 
3. Least important - what's causing the apparently harmless error `(3769) uart: uart_wait_tx_done(1115): uart driver error` when one creates a `machine.UART` instance corresponding to UART0.

Minimal examples of each issue...

**1.** Created `machine.UART` UART0 instance is not directly useable in MicroPython code:

```
import os

uart = UART(0, baudrate=9600, timeout=0)

uart.write(b'.')  # Results in `uart_write_bytes` error`.
uart.any()  # Results in `uart_get_buffered_data_len` error and exception `OSError: [Errno 1] EPERM: ESP_FAIL`.
```

**2.** Best equivalent of a non-blocking read is `poll`/`ipoll` and reading _a byte at a time_:

```
import os
import sys

from machine import UART

UART(0, 9600, timeout=0)

buffer = memoryview(bytearray(1024))

poller = select.poll()
poller.register(sys.stdin.buffer, select.POLLIN)

while True:
    for (s, event) in poller.ipoll(0):
        # There seems to be no way to do a non-blocking read of as many bytes as are currently available.
        # Instead, one must read a byte at a time (with `ipoll` telling you that at least one byte is available).
        read_count = s.readinto(buffer[:1])
        # ...
```

**3.** Creating a `machine.UART` instance (whether with a `baudrate` or other change or not) always results in a _harmless_ `E (738) uart: uart_wait_tx_done(1115): uart driver error`:

```
from machine import UART

UART(0)
```

---

Note: for reference, see [`machine_uart.c:366`](https://github.com/micropython/micropython/blob/f1d6af9/ports/esp32/machine_uart.c#L366) - one can see there's a special case for UART0 (i.e. `MICROPY_HW_UART_REPL` at this level) - it isn't deleted and recreated (but is later reconfigured by the code following this block):

```
if (uart_num != MICROPY_HW_UART_REPL)
//  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
{
    check_esp_err(uart_driver_delete(self->uart_num));

    check_esp_err(uart_param_config(self->uart_num, &uartcfg));

    check_esp_err(uart_driver_install(uart_num, self->rxbuf, self->txbuf, 0, NULL, 0));
}
```

kdschlosser, 2024-02-22:
This issue was not about the repl releasing control of UART 0. It was about changing the bit rate for UART 0. that is able to be done as you know.

Here is an example of how to hijack the UART from the repl.

https://github.com/micropython/micropython/blob/master/examples/bluetooth/ble_uart_repl.py

george-hawkins, 2024-02-22:
> This issue was not about the repl releasing control of UART 0. It was about changing the bit rate for UART 0.

Yes, I understand that. But to change the baud rate (or any other characteristic of the UART associated with the REPL), one creates a `machine.UART` instance that corresponds to UART0.

Given that it's done this way, one might naively assume that the constructed `machine.UART` instance would then be useable.

This does not seem to be the case. _Maybe_ adding a function into the `esp32` library would have been clearer, e.g. something like:

```
esp32.reconfigure_repl_uart(baudrate=9600)
```

This would avoid the creation of a `machine.UART` instance that confusingly turns out to be unusable, i.e. its sole purpose was to pass the constructor arguments down to the C layer in order to reconfigure the REPL UART and doing anything with the resulting instance (like calling `any()` or `write()`) will result in an error.

kdschlosser, 2024-02-23:
The returned instance buy creating a UART using the same port number is not going to return an working instance. It is only going to reconfigure it.

You are not able to overwrite the buffers used when the UART has been created when altering UART 0. That is because of the special handling that is involved with the REPL UART.

Now it appears that there might be coding issue with respect to the ESP32 port regarding os.dupterm. the issue is that dupterm doesn't allow for stepping in between and stopping input and output from the REPL. It only appears to do what the function name suggests and it duplicates the input and output.

george-hawkins, 2024-02-25:
Thanks for all the time you took over the last reply, @kdschlosser. I think I'm much clearer about what one can and cannot do with UART0 now (and have now been able to successfully implement what I originally wanted to achieve) :+1:

kdschlosser, 2024-02-25:
I shouldn't have said that it cannot be done. It can be done but you would have to alter the MicroPython C code if you wanted to keep both the REPL active and use UART0 or you would have to turn off the REPL all together.

0wwafa, 2025-01-19:
How about a simple:

micropython.stop_repl()

then use UART0 as we like and then

micropython.restart_repl()

?
that would be great!

kdschlosser, 2025-01-19:
unfortunately with MicroPython there is no really easy way to start and stop the REPL at runtime. I too wish there was. This would allow the pins to be configured for multiple purposes so long as there was a way to change between the purpose of the pins. Something like an attached LCD with a touch display where it could be selected for what is connected.

0wwafa, 2025-01-19:
> I too wish there was.

It's also crazy that there is not a possibility to change the port speed at runtime. Sometimes data need to flow faster than 115200...

kdschlosser, 2025-01-20:
The speed is runtime changeable.

robert-hh, 2025-01-20:
> The speed is runtime changeable.

@kdschlosser May you tell the world how you do that, probably without using the machine.UART class?

kdschlosser, 2025-01-20:
```python
import machine

machine.UART(0, baudrate=921600)
```

That's all there is to it. just set the  baudrate to whatever you want when you want it set.

MicroPython internally caches the UART instances it has created so if you use the same UART id it changes the configuration of it because that UART is already initialized. You don't need to hold a reference to it at all because the instance has a reference held internally in MicroPython.

robert-hh, 2025-01-20:
Doing that produces an error, but the baud rate is changed. However, UART gets pretty unreliable if the new rate is higher than 230400. I used 115200 in the example below to show the output.

```
>>> import machine
>>> machine.UART(0, 115200)
E (97363) uart: uart_wait_tx_done(1231): uart driver error
E (97363) uart: call uart_driver_install API first
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: (-259, 'ESP_ERR_INVALID_STATE')
```

jimisola, 2025-07-19:
Was hoping to increase the rate as I'm copying many large files but I'm getting the following error:

```
MicroPython v1.25.0 on 2025-04-15; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
Type "help()" for more information.
>>> import machine
>>> 
>>> machine.UART(0, baudrate=921600)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: (-259, 'ESP_ERR_INVALID_STATE')
```

kdschlosser, 2025-07-19:
You need to do it from a script you upload to the esp32. You cannot change the bitrate when you are connected to the ESP32 using that UART channel. 

I will typically and the code to boot.py. Upload that single script at 115200 and then when you restart the ESP32 you will need to connect to it using the higher speed.

kdschlosser, 2025-07-19:
There is also another way to do it as well. You can add `-DMICROPY_HW_UART_REPL_BAUD=921600` to the compile command and that will also change the UART speed without needing to mess about with changing the UART speed from python code.

jimisola, 2025-07-19:
@kdschlosser Ok. How do I know which UART(s) that to change? I have an ESP32-S3 with two USB-C.

kdschlosser, 2025-07-20:
By default MicroPython is set to use the USB CDC for the repl. So if your board uses the UART instead then you will need to tell MicroPython to use the UART instead.

jimisola, 2025-07-20:
> By default MicroPython is set to use the USB CDC for the repl. So if your board uses the UART instead then you will need to tell MicroPython to use the UART instead.

Ok. How do I check this?

dmascord, 2025-07-22:
> Was hoping to increase the rate as I'm copying many large files but I'm getting the following error:
>
> ```
> MicroPython v1.25.0 on 2025-04-15; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3 [...]

I received the same issue, even when used in boot.py, I had to work around the issue by catching the OSError and just printing it to the console (so I remember I've done the hack).  My use case is slightly different, as I am needing to connect to system which sets baud rate to 9600, and I am using sys.stdin.readline() to read the data in main.py.

```
import machine
try:
    machine.UART(0, baudrate=9600)
except OSError as e:
    print(e)
```

Then at a later stage, if I need to change the baud rate back to default 115200 (for the next boot), I do:
```
import os
os.remove('boot.py')
```