micropython/micropython #6866: rp2 More USB woes

peterhinch, 2021-02-09:
I raise this in case it sheds any more light on the USB issues.
I am using `rp2-pico-20210209-unstable-v1.14-11-gc7aaee2b2.uf2`.

I have a substantial `uasyncio` application developed on Pyboards and ported to the Pico. The test script outputs regular messages via print statements. The application maintains a heartbeat LED via a coroutine on the Pico.

It runs fine if I suppress all print output.

With print statements enabled it also runs so long as I periodically press a key on the keyboard (in rshell or in miniterm). If I don't, it runs for a while then stops. The heartbeat stops flashing until I press a key, when running resumes. The code has evidently stalled.

Unfortunately my attempts to create a trivial test case have so far failed. I have never encountered this behaviour on Pyboards, ESP8266 or ESP32.

peterhinch, 2021-02-09:
I should add that this application, while fairly involved, uses no "advanced" features of the chip. It communicates with an ESP8266 by bit banging, so doesn't use UART, I2C, SPI. It uses the Pin object but not interrupts. There is no use of native or viper emitters.

Here are the MicroPython modules imported by the various code modules:
```python
import uasyncio as asyncio
from utime import localtime, time
import sys  # sys.platform disables import of pyb and access to pyb.RTC
from micropython import const
import ujson
from machine import Pin  # The only "hardware" object
```

peterhinch, 2021-02-09:
I should also add what happens when I press a key as I think this must provide a clue.

The print statements comprise complete lines of text terminated with newlines. These are emitted at a low rate, a line every few seconds.

When the heartbeat stops flashing and I press Enter, the program resumes and many lines of text are emitted. Given the number of lines output it's clear that the program continued to run for several minutes before it stalled, with the text accumulating in a buffer. On receipt of a character the buffer emptied and the program resumed.

dpgeorge, 2021-02-09:
Is there are fixed time (always the same) for how long it takes before it pauses?  It may be that your PC is suspending the USB device so it goes to sleep, and only wakes when you type a key.

dhalbert, 2021-02-09:
We had a similar-sounding problem in CircuitPython recently. Copying multiple large files to the MSC drive would stall, then there would be a 30-second timeout, and USB would re-enumerate. The problem did not occur if there was a CDC REPL connection open to the board, even if there was no activity on that connection.

The problem was that `tud_task()` was not being scheduled to be called. We used to call `tud_task()` on every 1ms tick. Usually there was no work to do, but it got called promptly all the time. (It's cheap to call.) Then we started enqueuing an event to call it only after servicing a USB interrupt. If we were idle, we would do `WFI` to save power and just wait for an interrupt to trigger more work. This scheme worked most of the time, but we missed scheduling a `tud_task()` call in a particular circumstance. The [details](https://github.com/adafruit/circuitpython/pull/4122) are not really relevant to MicroPython because they have to do with our auto-reload mechanism. But they are related to scheduling `tud_task()`  properly when there is work to do, as is being discussed in #6835.

dpgeorge, 2021-02-09:
Thanks @dhalbert , that's very interesting info (and relates to calling `tud_task()` from the USB IRQ handler).  Currently our rp2 port calls `tud_task()` all the time (eg every 1ms) so not sure that is the issue here.

peterhinch, 2021-02-09:
I think this is unlikely as I have used this PC extensively with other MicroPython targets and never seen this before.

It also doesn't explain the symptoms (in my opinion).

While the program is running, no output is printed until I press Enter (or another key). Then, whatever output is pending is printed. Aside from the heartbeat, there is other evidence that the program is running normally while producing no printed output. If I don't press a key, it eventually stalls after nearly three minutes. Then pressing a key releases a torrent of (correct) output and the program resumes.

Another observation. The first time this happened I assumed my code had hung so I hit ctrl-c. To my surprise the Pico emitted many lines of output before producing a prompt. The code doesn't trap the keyboard interrupt. Presumably the code stopped immediately with the data being output by the underlying runtime.

dhalbert, 2021-02-09:
I spoke too casually and perhaps misleadingly. In CircuitPython 5.x and before, we used to _schedule_ a call to `tud_task()` on every 1ms tick. We did not call it from the IRQ handler. We use `MICROPY_VM_HOOK_LOOP` to run background tasks, and running `tud_task()` (and many other background tasks) happened if 1ms had passed.

We subsequently revised the mechanism so that not every background task was called when 1ms had passed. Instead the tasks are enqueued when there is work to do. We no longer have a 1ms tick scheduling the overall set of background task. We still use `MICROPY_VM_HOOK_LOOP` (renamed to `RUN_BACKGROUND_TASKS`) and it does the queue checking, but only runs what is enqueued.

(Perhaps I should be repeating this in #6835.)

peterhinch, 2021-02-09:
OK, here is a trivial test case. Paste at the REPL. I left this running for many minutes before going back to examine progress,
```python
import time
while True:
    print('got here')
    time.sleep(30)
```
It printed output once only, only emitting queued output when I press a key. The slow rate seems to be crucial, my other attempts at test cases emitted output at high speed. These ran without error.

peterhinch, 2021-02-09:
This test case behaves in exactly the same way as my application. I added the obligatory LED flasher:
```python
import time
from machine import Pin
led = Pin(25, Pin.OUT)
while True:
    print('got here')
    led(not led())
    time.sleep(30)
```
which shows that the script continues to run while producing no output, but it eventually stalls. Pressing a key causes it to emit output and to resume running.

robert-hh, 2021-02-09:
How long usually until it gets stuck? I'm now at 123 iterations.

peterhinch, 2021-02-09:
The script produces only one line. If I hit a key, it produces as many lines as are buffered at that time. I didn't time how long it took to actually stall, I'd guess at about 15 minutes (30 iterations).

I've tried reducing the delay. At 2 seconds or below it works perfectly. At 3 seconds it behaves as described, emitting only one line until I press a key.

robert-hh, 2021-02-09:
At my set-up, the script produces a line every 30 seconds. It goes on and on.

robert-hh, 2021-02-09:
What I noticed and verified again now: the RP2040 does not work at a USB2 port of my debian 10 computer, which is like 7 years old (time flies). It needs a USB3 port. Same on a small windows computer. Same on an old MAC.

robert-hh, 2021-02-09:
@peterhinch Do you use an USB hub? I tried once but only had problems.

peterhinch, 2021-02-09:
OK, this prompted me to try other PC's. On my three other machines the fault does not appear even with 30s timing. The two desktops are elderly so USB2. The other laptop is USB3. One of the desktops uses a hub.

So it's just this laptop.

There must be something odd about the USB implementation on this laptop. It's a two year old Dell Precision 5530 Core i5-8300H - a fairly high-end machine. It runs Ubuntu 20.04. I use it heavily for MicroPython on Pyboards, ESP8266 and ESP32 and have had no issues with USB. I have changed nothing in its setup and the same script ran with this machine on a Pyboard 1.1.

I therefore do believe there is an issue with the rp2 USB, but I've clearly struck unlucky with my usual PC.

robert-hh, 2021-02-09:
I agree that the rp2 USB interface needs rework. It should work an any USB2 or USB3 port.

robert-hh, 2021-02-09:
I re-checked. Using the Circuitpython firmware the device will be detected on USB2 ports.

lurch, 2021-02-09:
This sounds _very_ similar to the buffering / flow-control issues seen with RP2040 in #6803 ?
I'll mention it to the team tomorrow to see if anyone has some ideas :slightly_smiling_face: 

> I've clearly struck unlucky with my usual PC.

Perhaps it's specific to the different USB controller chipsets on the motherboard? :shrug:

dpgeorge, 2021-02-10:
I ran the led-blink-sleep-30-seconds test above on rp2 via a USB 2.0 hub, and it seems to work ok for me.  It has been running for about 2 hours and the LED is still blinking every 30 seconds.

But the USB 2.0 hub did fail after about 1 hour.  The pico board is still powered via the hub and running but the serial output is gone.  Probably this is a problem with the USB 2.0 hub (it's seen a lot of activity...).

liamfraser, 2021-02-10:
I also can't reproduce this. What program are you using to view the serial console?

peterhinch, 2021-02-10:
On this PC you don't need to wait long: failure occurs after the first delay. It outputs one message and then stays silent until you press a key.

It looks as if there is some subtlety in USB implementations: if so what's needed is to try a number of PC's.

liamfraser, 2021-02-10:
@peterhinch can you try and capture a usb trace on the failing pc with https://desowin.org/usbpcap/ It should be viewable in wireshark. Please submit it here

peterhinch, 2021-02-10:
Sorry but I don't own a Windows PC: all Linux here. I'm also a total wireshark n00b... 

@robert-hh has a Windows box which sounds as if it's failing.

liamfraser, 2021-02-10:
I think @robert-hh is actually hitting a USB errata (RP2040-E5) where an RP2040 can fail to enumerate if you have a serial port open. CircuitPython compiles this fix in but nobody else does by default. Would explain the difference between the two. 

I didn't realise it was Linux specific... @lurch can replicate on his Linux machine so we can investigate from there.

robert-hh, 2021-02-10:
As far as I can tell, I do not have a serial port open on my Windows device. It is the difference between USB2 and USB3.

peterhinch, 2021-02-10:
In case this is of any help, here is the relevant output of **lshw** on my failing machine.
```
        *-usb
             description: USB controller
             product: Cannon Lake PCH USB 3.1 xHCI Host Controller
             vendor: Intel Corporation
             physical id: 14
             bus info: pci@0000:00:14.0
             version: 10
             width: 64 bits
             clock: 33MHz
             capabilities: pm msi xhci cap_list
             configuration: driver=xhci_hcd latency=0
             resources: irq:126 memory:cc400000-cc40ffff
           *-usbhost:0
                product: xHCI Host Controller
                vendor: Linux 5.4.0-65-generic xhci-hcd
                physical id: 0
                bus info: usb@1
                logical name: usb1
                version: 5.04
                capabilities: usb-2.00
                configuration: driver=hub slots=16 speed=480Mbit/s
              *-usb:0
                   description: Bluetooth wireless interface
                   vendor: Intel Corp.
                   physical id: 4
                   bus info: usb@1:4
                   version: 0.02
                   capabilities: bluetooth usb-2.00
                   configuration: driver=btusb maxpower=100mA speed=12Mbit/s
              *-usb:1
                   description: Video
                   product: Integrated_Webcam_HD
                   vendor: CN0VGCNHLOG008C4A4TFA00
                   physical id: c
                   bus info: usb@1:c
                   version: 81.31
                   capabilities: usb-2.01
                   configuration: driver=uvcvideo maxpower=500mA speed=480Mbit/s
           *-usbhost:1
                product: xHCI Host Controller
                vendor: Linux 5.4.0-65-generic xhci-hcd
                physical id: 1
                bus info: usb@2
                logical name: usb2
                version: 5.04
                capabilities: usb-3.10
                configuration: driver=hub slots=10 speed=10000Mbit/s
```

peterhinch, 2021-03-07:
This problem keeps cropping up in the forum, with people thinking their script has failed when in fact it is the USB communication which has stopped. 

The defining characteristic is that increasing the message rate makes the fault go away. [This](https://forum.micropython.org/viewtopic.php?f=21&t=9988) is the most recent instance.

lurch, 2021-03-07:
I'm afraid this dropped off the bottom of my priority-list, but now that the latest pico-sdk release is done I'll make sure to do those Wireshark-captures next week :+1:

dpgeorge, 2021-03-31:
Should be fixed by ca3d51f12206a88463f2b2a2d27756a1b519cb6a.  Please reopen if not.

closed 2021-03-31 by dpgeorge

robert-hh, 2021-03-31:
Still no response on an USB2 port.

liamfraser, 2021-03-31:
@robert-hh looking at past comments I think this is because you are hitting the USB enumeration errata (RP2040-E5). You said CircuitPython works, and CircuitPython compiles in a workaround to handle this. MicroPython could potentially do the same. Usually the issue is triggered by having a serial port open connected to the same hub Pico is plugged into but a similarly busy USB device on the same hub could cause it.

lurch, 2021-03-31:
@robert-hh I agree with Liam that your issue sounds different to the one that was being discussed here. Please open a separate issue and include as much detail as you can. Also, does this only affect MicroPython or does it also affect the "'Hello World' over a USB serial connection." UF2 at https://www.raspberrypi.org/documentation/rp2040/getting-started/#getting-started-with-c ?

liamfraser, 2021-03-31:
@lurch @robert-hh an even easier test would be if the bootsel mode works or not

robert-hh, 2021-03-31:
bootsel mode also does not work on the USB2 port.

liamfraser, 2021-03-31:
Okay, so can you try CircuitPython and see if that works on the USB2 port?

peterhinch, 2021-03-31:
This has fixed the problem in my initial reports.

robert-hh, 2021-03-31:
adafruit-circuitpython-raspberry_pi_pico-en_US-6.2.0-beta.4 works
adafruit-circuitpython-raspberry_pi_pico-en_US-6.2.0-beta.1 did not not work

lurch, 2021-03-31:
@tannewt @dhalbert  I have no familiarity with the CircuitPython codebase - do you have any idea what might be going on here?

@robert-hh Your report that it works in a USB3 port but not a USB2 port is quite strange, because Pico doesn't implement the [extra connectors](https://en.wikipedia.org/wiki/USB_3.0#Connectors) used by USB3, and RP2040 itself only supports USB1.1, so regardless of whether it's plugged into a USB2 port or a USB3 port, Pico should appear identically to your computer :confused: 

As I mentioned above, it's probably better to create a new issue for this, rather than continuing discussion in this now-closed issue.

liamfraser, 2021-03-31:
Does a pico work in the USB2 port without any hub connected?

dhalbert, 2021-03-31:
> @tannewt @dhalbert I have no familiarity with the CircuitPython codebase - do you have any idea what might be going on here?

We have a couple of recent TinyUSB-related PR's that may or may not be relevant. We are ahead of MPy in terms of TinyUSB fixes:
https://github.com/adafruit/circuitpython/pull/4365
https://github.com/adafruit/circuitpython/pull/4327

robert-hh, 2021-03-31:
@liamfraser : There is no external huv connected. I made a new PR at #7089.