micropython/micropython #6867: MicroPython version 1.14 (ESP-IDF v4.x) with PPP support on the ESP32 port.

beyonlo, 2021-02-09:
@dpgeorge Hello. 

Just a clarification:

I see in the announced 1.14 version ( https://github.com/micropython/micropython/releases/tag/v1.14 ) that PPP is now working in IDF >= 4, for ESp32 port:
"modnetwork: re-enable PPP support for IDF-SDK >=v4"

But in the download area ( http://micropython.org/download/esp32/) show that is not working yet:
"
Firmware with ESP-IDF v4.x
Firmware built with ESP-IDF v4.x, with support for BLE, but no LAN or PPP:
"
Please, could you clarify that?

Thank you so much.

mirko, 2021-02-09:
`micropython$ git log v1.14 | grep -q 'esp32/modnetwork: Re-enable PPP support' && echo "yes" || echo "no"`
 -> `yes`

So, yes, it should be part of the release. If it's not, it's probably a bug.
Re "Firmware built with ESP-IDF v4.x, with support for BLE, but no LAN or PPP:" I assume nobody adjusted that text line. At least I wasn't aware of that text when I re-enabled PPP support in above mentioned commit.

mirko, 2021-02-09:
@jimmo @dpgeorge anybody willing to alter respective textline on http://micropython.org/download/esp32/ accordingly?

jonnor, 2024-08-18:
I believe PPP and LAN are now included in the builds, and that this issue can be closed.

closed 2024-08-18 by robert-hh