micropython/micropython #6871: pyb.USB_VCP().isconnected() stuck to True when physically disconnected

Arrowana, 2021-02-10:
**Problem**:
pyb.USB_VCP().isconnected() remains stuck to True when physically disconnected

**Use case**:
If unlucky we can get in this bad state when the USB connection has some issue
Windows USB device descriptor failed...

So the cable can remain attached but we have abruptly lost the COM port

**Software and hardware**:
micropython release v1.12 with STM32F439 on custom board

**How to reproduce**:
Run the following snippet in the REPL
```
import pyb

vcp = pyb.USB_VCP()

result = []
def test():
	delay_ms = 5000
	iterations = 5
	global result
	result = []
	for _ in range(iterations):
		isconnected = vcp.isconnected()
		print(isconnected)
		result.append(isconnected)
		pyb.delay(delay_ms)
		vcp.send(b'Potato salad')
		
test()  # Disconnect USB physically while this is running
```
Then
>>> result
[True, True, True, True, True]

**Possible cause**:
There are only a few spot that can change the condition
https://github.com/micropython/micropython/blob/0fff2e03fe07471997a6df6f92c6960cfd225dc0/ports/stm32/usbd_cdc_interface.h#L74-L76

isconnected() behaviour is correct if I disconnect putty while the snippet above is correct so it does seem like we can hit
usbd_cdc_control case CDC_SET_CONTROL_LINE_STATE

However, something is missing for when the USB connection is completely lost.
Happy to work on this issue but I might need some guidance as to how to resolve this issue.

dpgeorge, 2021-02-10:
This sounds very similar to #6672, and see also a fix for that particular issue in #6827

dpgeorge, 2021-02-10:
On F4's I think we need to intercept the `HAL_PCD_DisconnectCallback()` event/callback to handle the case of abrupt disconnection.

Arrowana, 2021-02-10:
There is already something being called on this event/callback
https://github.com/micropython/micropython/blob/0fff2e03fe07471997a6df6f92c6960cfd225dc0/ports/stm32/usbd_conf.c#L351-L353

which sets pdev->dev_state = USBD_STATE_DEFAULT;
https://github.com/micropython/micropython/blob/0fff2e03fe07471997a6df6f92c6960cfd225dc0/ports/stm32/usbdev/core/src/usbd_core.c#L531-L538

Are you saying we need to add something to it so that we can change cdc->connect_state to disconnected?

dpgeorge, 2021-02-10:
> So it does calls USBD_CDC_MSC_HID_DeInit through DeInit. This calls usbd_cdc_deinit...
> Is the callback simply not firing?

Indeed it does look like the code is calling `usbd_cdc_deinit()` on this event.  So I'm not sure what is wrong, it'll need to be investigated.

Arrowana, 2021-02-11:
So I added printfs and usbd_cdc_deinit is called just fine and sets for the right usbd_cdc_itf_t* (I checked pointer addresses compared to the one in isconnected).

So still investigating, but now the suspicion is that another callback sets the connect_state to USBD_CDC_CONNECT_STATE_CONNECTING then it evolves into USBD_CDC_CONNECT_STATE_CONNECTED

Arrowana, 2021-02-15:
So after going deeper, the board definition mpconfigboard.h does not set MICROPY_HW_USB_VBUS_DETECT_PIN!
And we don't have the pin connected on the board, so there is no bug there.

Coming back to the second part, if the bus is corrupted "Windows USB device descriptor failed...", then we never recover. 
I have explored HAL_PCD_SuspendCallback, the bus does go to suspended on disconnection. So it seems like in user code i could detect that to reset the bus when required.
We also checked VBUS pin electrically (multimeter) and windows does not do anything to it when we get into "Windows USB device descriptor failed"

Arrowana, 2021-02-22:
I think the interaction between the micropython usb driver and windows confused me as we have a USB hub in our hardware.
After further testing, I don't believe micropython has a is_connected problem. Potentially only a problem when the connection enters device descriptor failed windows side, but i don't have strong proof for now.

closed 2021-02-22 by Arrowana

dpgeorge, 2021-03-11:
Ok, thanks for the update.  Feel free to reopen if there is a bug in the code that can be fixed.