micropython/micropython #6873: rp2: time epoch differs from Pyboard and docs.

peterhinch, 2021-02-10:
The docs state that for embedded ports the epoch is 2000.

I retrieve time from an NTP server and apply an offset of 3155673600 to convert from 1900 to 2000. This was calculated from
```python
>>> from datetime import date
>>> (date(2000, 1, 1) - date(1900, 1, 1)).days * 24*60*60
3155673600
```
This produces correct results on Pyboards when passed to **time.localtime()**. On the rp2 it produces correct dates and times except that the year is 1991.

Is this intentional?

davehylands, 2021-02-10:
From my investigation, the RPi Pico firmware uses an epoch of 1970 (and I seem to recall that pyboard D may be the same). So I would say that the documentation is incorrect.

The way that rshell determines the epoch is to do:
```
>>> import time
>>> time.gmtime(0)
(1970, 1, 1, 0, 0, 0, 3, 1)
>>> 
```

peterhinch, 2021-02-11:
Thanks, that's a good workround.

It seems perverse that epoch, on embedded chips, varies. Y2K was chosen to constrain word length. But if it must vary randomly the fact should be documented. On a Pyboard 1.1:
```python
>>> utime.gmtime(0)
(2000, 1, 1, 0, 0, 0, 5, 1)
>>> 
```

davehylands, 2021-12-13:
I think that boards which comes with WiFi will want to use the 1970 epoch because that's what NTP uses.

I believe that choosing the 2000 epoch for micropython was related to being able to use a smallint to store the number of seconds since the epoch. A 30-bit number worth of seconds is 34 years (so picking 2000 epoch allows years from 1966 thru 2034
```
>>> 2**30 / 86400 / 365
34.048129883307965
```

jonnor, 2024-08-18:
The method to determine epoch time is now included in the documentation. Proposing to close this issue.

closed 2024-08-18 by robert-hh