micropython/micropython #6874: Micropyton 1.14 doesn't work on Lolin esp8266 Pro v2

fe328, 2021-02-10:
After flashing 1.14 the board doesn't work (can't enter REPL), blue led continuously blinks.
Same board works fine with 1.13.

I'm a bit of new to this, so I lack knowledge to debug this problem.
What info would be helpful?

Flashing process:

```
+ esptool.py --port /dev/ttyUSB0 --baud 460800 erase_flash
esptool.py v3.0
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: ****
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Erasing flash (this may take a while)...
Chip erase completed successfully in 39.7s
Hard resetting via RTS pin...
+ esptool.py --port /dev/ttyUSB0 --baud 460800 write_flash -fm dio --flash_size=detect 0 esp8266-20210202-v1.14.bin
esptool.py v3.0
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: ****
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 16MB
Flash params set to 0x0290
Compressed 632064 bytes to 415345...
Wrote 632064 bytes (415345 compressed) at 0x00000000 in 9.4 seconds (effective 538.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

robert-hh, 2021-02-11:
Try setting the flash size during flash to 4 MB. Using 16MB requires a separate step.
esptool.py --port /dev/ttyUSB0 --baud 460800 write_flash -fm dio --flash_size=4MB 0 esp8266-20210202-v1.14.bin

about 16MB Flash: https://github.com/micropython/micropython/issues/2335

fe328, 2021-02-11:
Ah. Thank you. I tried the trick form #2335 before writing this issue. It worked for 1.13, but not 1.14, so I thought that the issue was in something else.

But setting 4M fs solved the problem. Thank you again)

closed 2021-02-11 by fe328

robert-hh, 2021-02-11:
Did you manually specify 16MB flash size when uploading both the esp_init_data_default.bin file and the firmware? I just tried the 1.14 firmware on an old ESP01 modded to 16MB, and it worked.