micropython/micropython #6875: adc.h error in compilation

Krasn4ck, 2021-02-11:
Hello everyone, I have been trying to compile a custom version of micropython 1.14 for the STM32F407VGT6 microcontroller, apparently it has failed to compile because it throws the following error:

`CC machine_adc.c
In file included from machine_adc.c:29:
adc.h: In function 'adc_deselect_vbat':
adc.h:50:29: error: 'LL_ADC_PATH_INTERNAL_VBAT' undeclared (first use in this function)
   50 |         adc_common->CCR &= ~LL_ADC_PATH_INTERNAL_VBAT;
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~~
adc.h:50:29: note: each undeclared identifier is reported only once for each function it appears in
make: *** [../../py/mkrules.mk:77: build-OPHYRA/machine_adc.o] Error 1
`
the command I used to compile was 
`make -j2 BOARD=OPHYRA`

Ophyra is the name of the folder where the configuration files are contained.
I am using cygwin on windows 10.

Thanks for reading...

dpgeorge, 2021-02-11:
It should build without error:
- `LL_ADC_PATH_INTERNAL_VBAT` is defined for F4 MCUs in `lib/stm32lib/STM32F4xx_HAL_Driver/Inc/stm32f4xx_ll_adc.h`
- this file is included by `ports/stm32/boards/stm32f4xx_hal_conf_base.h`
- that file should be included by `board/OPHYRA/stm32f4xx_hal_conf.h`
- that's then included by `mpconfigboard_common.h` as `#include STM32_HAL_H`

dpgeorge, 2021-02-11:
Try building the standard PYBV10 board as a test (via `make BOARD=PYBV10` in the stm32 dir).

Krasn4ck, 2021-02-11:
> Try building the standard PYBV10 board as a test (via `make BOARD=PYBV10` in the stm32 dir).

I just compiled PYBV10 and had no problems.

`CC build-PYBV10/pins_PYBV10.c
CC build-PYBV10/frozen_content.c
LINK build-PYBV10/firmware.elf
   text    data     bss     dec     hex filename
 360256      16   27272  387544   5e9d8 build-PYBV10/firmware.elf
GEN build-PYBV10/firmware.hex
GEN build-PYBV10/firmware1.bin
GEN build-PYBV10/firmware0.bin
GEN build-PYBV10/firmware.dfu
`

Krasn4ck, 2021-02-11:
> * stm32f4xx_hal_conf_base.h
I have added  `#include "boards/stm32f4xx_hal_conf_base.h"` to the `stm32f4xx_hal_conf.h` file in the OPHYRA folder and when I compiled again I got the following error:
```
CC ../../py/mpstate.c
CC ../../py/nlr.c
make: *** [../../py/mkrules.mk:77: build-OPHYRA/py/mpstate.o] Error 1
make: *** Waiting for unfinished jobs....
make: *** [../../py/mkrules.mk:77: build-OPHYRA/py/nlr.o] Error 1
```

dpgeorge, 2021-02-11:
> I just compiled PYBV10 and had no problems.

Then it must be your OPHYRA board configuration that is wrong.  I suggest starting from PYBV10 and adjusting it until it matches OPHYRA.  That will show the problem.

Krasn4ck, 2021-02-11:
> > I just compiled PYBV10 and had no problems.
> 
> Then it must be your OPHYRA board configuration that is wrong. I suggest starting from PYBV10 and adjusting it until it matches OPHYRA. That will show the problem.

I almost got it to compile, it just gave me this error, thank you very much for your help @dpgeorge .
```
LINK build-OPHYRA/firmware.elf
C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin\arm-none-eabi-ld.exe: build-OPHYRA/extmod/vfs.o: in function `mp_vfs_mount':
vfs.c:(.text.mp_vfs_mount+0x118): undefined reference to `mp_type_vfs_lfs2'
C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin\arm-none-eabi-ld.exe: build-OPHYRA/main.o: in function `stm32_main':
main.c:(.text.stm32_main+0x264): undefined reference to `_sstack'
C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin\arm-none-eabi-ld.exe: build-OPHYRA/stm32_it.o: in function `HardFault_C_Handler':
stm32_it.c:(.text.HardFault_C_Handler+0x150): undefined reference to `_sstack'
C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin\arm-none-eabi-ld.exe: build-OPHYRA/modmachine.o: in function `machine_info':
modmachine.c:(.text.machine_info+0x1a0): undefined reference to `_sstack'
make: *** [Makefile:689: build-OPHYRA/firmware.elf] Error 1

```

davehylands, 2021-02-11:
_sstack comes from one of the linker script files.

Your mpconfigboard.mk file should specify LD_FILES which is a list of linker script files to use. The stm32f405/407 one is this one: https://github.com/micropython/micropython/blob/df85e4881396d0ebb5942a111fba974ea5c80b77/ports/stm32/boards/PYBV10/mpconfigboard.mk#L10

and the _sstack symbol is defined here: https://github.com/micropython/micropython/blob/df85e4881396d0ebb5942a111fba974ea5c80b77/ports/stm32/boards/stm32f405.ld#L23

You might also want to try doing a clean followed by a rebuild just to make sure that there aren't any files that didn't get rebuilt when you changed some of your config options.

Krasn4ck, 2021-02-12:
> _sstack comes from one of the linker script files.
>
> Your mpconfigboard.mk file should specify LD_FILES which is a list of linker script files to use. The stm32f405/407 one is this one: [...]

Thank you @davehylands, I just adjusted the frequency of the card crystal and exchanged the 407 file for the 405 and it worked, thank you all.
```
LINK build-OPHYRA/firmware.elf
   text    data     bss     dec     hex filename
 333428      16   27112  360556   5806c build-OPHYRA/firmware.elf
GEN build-OPHYRA/firmware0.bin
GEN build-OPHYRA/firmware1.bin
GEN build-OPHYRA/firmware.hex
GEN build-OPHYRA/firmware.dfu

```

closed 2021-02-12 by Krasn4ck