micropython/micropython #6891: I2C writes fail sporadically with OS Error 5

romilly, 2021-02-13:
Multiple I2C writes to an MCP23008 8-bit I/O expander on the Raspberry Pi Pico fail sporadically with OS Error 5.
[image]

The problem occurs with multiple versions of micropython on the  Pico, including
rp2-pico-20210205-unstable-v1.14-8-g1f800cac3.uf2

Code at https://github.com/romilly/pico-code/blob/master/src/pico_code/pico/mcp23008.py and
https://github.com/romilly/pico-code/blob/master/src/pico_code/pico/mcp3008-demo.py

I know the i2c read code won't work, as it uses repeat=True, but it's never invoked in this test.

I am using the I2C pins supported by the Pimoroni Pico explorer base (20 and 21) with its built-in 10K pull-ups, but the same problem arises using pins 0 and 1 with 6K8 pull-ups.

I've seen this behaviour before on the bbc microbit.

sxpert, 2023-01-01:
I have this issue while reading each register of a PCA9685. sometimes it goes all the way from 0 to 255, at some other time, it fails at a random location in the scan with OSError 5. does not occur when using SoftI2C.

robert-hh, 2023-01-01:
The difference between SoftI2C and hard I2C: Soft I2C has an effective slower clock, and SoftI2C supports clock stretching, hard I2C it differs between ports. You could try to run hard I2C at slower clock rates, like 100kHz or less.

jonnor, 2024-09-13:
Sporadic I2C read errors are not uncommon in the wild, unforatunately. EMI and general signal integrity influences a lot. Can anyone else reproduce this specific issue?

jonnor, 2025-03-03:
Missing information to reproduce.

closed 2025-03-03 by jonnor