micropython/micropython #6895: Wrong pull up/pull down on esp32

honza-klu, 2021-02-13:
I updated my very old micropython (unfortunatly unnown version) on esp32 to esp32-idf3-20210206-unstable-v1.14-9-g9dedcf122.bin and I noticed that unconnected pin setup as following 
```
hall_down = machine.Pin(10, machine.Pin.IN | machine.Pin.PULL_UP)  
hall_down.value()  
```
returned 0 instead of expected 1. When I switched machine.Pin.PULL_UP for machine.Pin.PULL_DOWN value vas changed to 1. It looks like PULL_UP  and PULL_DOWN are swaped.

I also tried to downgrade micropython to esp32-idf3-20210203-unstable-v1.14.bin and it fixed issue.
I did not found any changes to the micropython gpio code, but maybe, there is problem in espressif idf?

dpgeorge, 2021-02-14:
> hall_down = machine.Pin(10, machine.Pin.IN | machine.Pin.PULL_UP)

This should be (note comma):
```python
hall_down = machine.Pin(10, machine.Pin.IN, machine.Pin.PULL_UP)  
```

honza-klu, 2021-02-14:
Well, that is embarrassing mistake. Thank you, it solve issue. (-:

closed 2021-02-14 by honza-klu