micropython/micropython #6898: rp2 port: Value of MICROPY_QSTR_BYTES_IN_HASH

robert-hh, 2021-02-14:
Not an issue, more a question:
What is the reason for MICROPY_QSTR_BYTES_IN_HASH being set to 1 in the rp2 port, but left on the default of 2 in the other ports?

lurch, 2021-02-17:
A quick look at https://github.com/micropython/micropython/blob/master/py/qstr.c#L46 suggests that MICROPY_QSTR_BYTES_IN_HASH = 1 can hash 256 different QSTRs before getting a collision, whereas MICROPY_QSTR_BYTES_IN_HASH = 2 can hash 65536 different QSTRs before getting a collision. So _maybe_ it's just a vague rule-of-thumb thing, depending upon how many QSTRs are included by default in a particular port? :shrug:  Or perhaps it's just a memory-usage vs. speed trade-off?

robert-hh, 2021-02-17:
The probablitly of a collision depends on the size of the hash. So you can find 2**n QSTR with different hash value, but any two different QSTR may have the same hash value, regardless of the hash size. So when using a hash function one has always consider an collision for different values. For QSTR use, this should be covered.
But since this hash functions is also used for the hash() method of micropython, fewer bytes are worse. So I'm just asking why it's different to the other ports.

dpgeorge, 2021-02-18:
The rp2 port was originally based on the bare-arm/minimal ports, which use 1 for this config value, and I guess I just never revisited this setting... probably makes sense to put it to 2 to improve performance.

closed 2022-01-26 by robert-hh