micropython/micropython #6899: Working Pico thread messes up the next REPL command

aivarannamaa, 2021-02-14:
When the background thread is working in Pico, then submitting the next REPL command fails because of random syntax error.

The problem was initially reported under https://github.com/thonny/thonny/issues/1586. Here I'm giving a test-case which doesn't involve Thonny.

```python
import pyboard

thread_script = """
import machine
from machine import Pin
import time
import _thread

thread_done = False

def ltask(blinks,delay):
    global thread_done
    led = Pin(25, Pin.OUT)
    for i in range(blinks):
        led.high()
        time.sleep(delay)
        led.low()
        time.sleep(delay)
    print("CPU1 Done")
    thread_done = True

_thread.start_new_thread(ltask, (10, 0.2))

# The problem can be avoided by joining the thread before returing:
# while not thread_done:
#     pass

"""

direct_script = """
import machine
from machine import Pin
import time
import _thread

def ltask(blinks,delay):
    led = Pin(25, Pin.OUT)
    for i in range(blinks):
        led.high()
        time.sleep(delay)
        led.low()
        time.sleep(delay)
    print("CPU1 Done")

ltask(10, 0.2)
"""

second_script = """
print("let's just take up some space")
print("let's just take up some space")
print("let's just take up some space")
print("let's just take up some space")
print("let's just take up some space")
print("let's just take up some space")
print("let's just take up some space")
print("let's just take up some space")
print("let's just take up some space")
print("let's just take up some space")
print("let's just take up some space")
print("let's just take up some space")
print("let's just take up some space")
print("let's just take up some space")
print("let's just take up some space")
print("let's just take up some space")
print("let's just take up some space")
print("let's just take up some space")
print("let's just take up some space")
print("let's just take up some space")
print("let's just take up some space")
print("let's just take up some space")
print("let's just take up some space")
"""

pb = pyboard.Pyboard("/dev/ttyACM0")
print("Got connection")
pb.enter_raw_repl()
pb.exec_(thread_script, data_consumer=print)
#pb.exec_(direct_script, data_consumer=print)
#pb.enter_raw_repl()
pb.exec_(second_script, data_consumer=print)
pb.exec_("print('__thonny_helper' in dir())", data_consumer=print)
```

Context:

* The imported `pyboard` is from current master (https://github.com/micropython/micropython/blob/d128999938d6b44f52f4d0c6e1f5169ab3c1e7a5/tools/pyboard.py). 
* Pico: the release version (not prototype)
* MicroPython 1.14 release
* OS: Ubuntu 20.04

When the test script is run as is, the first command in `thread_script` seems to be submitted successfully, but the next command  (`second_script`) always gets a syntax error on a random line, ie. it looks like the input gets messed up somewhere. If the test script is attempted again (even after the thread is supposed to have finished), the `thread_script` now gives `OSError: core1 in use` and the only way to cure this is to reset the device.

When I restart Pico and try `direct_script` instead of `thread_script` (ie. avoid using threads) or when I join the thread in `thread_script` before returning, then everything runs properly.

I assume `pyboard` is using raw-paste mode, but the problem remains if I change `self.use_raw_paste` to `False` in `Pyboard`'s constructor. In Thonny I was able to avoid this problem by using regular paste mode, but I'm not confident it would always work in this situation.

aivarannamaa, 2021-02-14:
Made another experiment -- it looks like if I wait with second command until the thread is done, then everything works properly (ie. things get wrong when the second script is submitted while thread is working, not when the first script returns while the thread is working):

```python
pb = pyboard.Pyboard("/dev/ttyACM0")
print("Got connection")
pb.enter_raw_repl()
pb.exec_(thread_script, data_consumer=print)
#pb.exec_(direct_script, data_consumer=print)

time.sleep(15)
pb.serial.read_all() # Othewise exec_ gets confused by the output of the thread
pb.enter_raw_repl()

pb.exec_(second_script, data_consumer=print)
pb.exec_("print('__thonny_helper' in dir())", data_consumer=print)
```

tonygo, 2021-02-18:
```
# Trying to understand threads
from sys import exit
import time, _thread, machine

flag = False
def task(n, delay):
    global flag
    led = machine.Pin(25, machine.Pin.OUT)
    for i in range(n):
        led.high()
        time.sleep(delay)
        led.low()
        time.sleep(delay)
    print('done')
    flag = True

_thread.start_new_thread(task, (10, 0.25))

while not flag:
    print("Waiting")
    time.sleep(1)
```
Got some help from Pi Forum with this suggestion which worked but sometime intermixes up the two prints if you reduce the delays.

I now need to work out how to lock/unlock.