micropython/micropython #6903: STM32: Can't Factory Reset

braiden, 2021-02-15:
Expected Behavior:
  1. While Holding User Button
  2. Press Reset Button
  3. Release User Button after 3 flashes
  4. Devices echos back 3 flashes
  5. Files-system should be restored to default state.

?

Or am I misunderstanding how it's supposed to work?

I had to change this line:
https://github.com/micropython/micropython/blob/bffb71f523e4bcc21b913af291deeb67091bed88/ports/stm32/main.c#L534

to: `mounted_flash = init_flash_fs(state.reset_mode);`

... since https://github.com/micropython/micropython/blob/bffb71f523e4bcc21b913af291deeb67091bed88/ports/stm32/main.c#L475 only updates `state.reset_mode` not `reset_mode`.

chrismas9, 2021-02-15:
`Release User Button after 3 flashes`

That depends on the board and how many LEDs are configured. Boards with one LED will flash it three times for FS reset. Boards with three or more LEDs (eg pyboard) count in binary on LED2 and LED3 so FS reset is when both are on. Boards with two LEDs also count in binary but you only see bit 0. When LED2 comes on count is 1. When it goes out count is 2. When it comes on again count is 3. So FS reset is second flash. Many Nucleo boards only have two LEDs.

braiden, 2021-02-15:
My board, [WeAct_F411CE](https://github.com/WeActTC/WeAct_F41), has 1 led, active low. So three flash, reset_mode = 3 I believe, and PR should resolve.

Maybe the bug exists only for boards where `MICROPY_HW_USES_BOOTLOADER`  is false.

dhylands, 2021-02-15:
I can confirm that using the latest master than the file system reset no longer works on my PYBV11 board.

I'm releasing when both orange and green LEDs are lit and the files listed in my filesystem before the reset are still there after the reset.

dhylands, 2021-02-15:
I can confirm that the filesystem reset doesn't work on my PYBV11 with the latest master and that the filesystem reset works properly on my PYBV11 with PR #6904

dpgeorge, 2021-03-10:
Thanks for the report, and sorry it took a bit of time to get to it.

commit 85ea4ac0 (2021-03-10): stm32/main: Fix passing state.reset_mode to init_flash_fs.

state.reset_mode is updated by `MICROPY_BOARD_BEFORE_SOFT_RESET_LOOP` but
not passed to `init_flash_fs`, and so factory reset is not executed on
boards that do not have a bootloader.  This bug was introduced by
4c3976bbcaf58266fdcaab264fee5b7a94a682e5

Fixes #6903.
closed 2021-03-10 by dpgeorge