micropython/micropython #6908: rp2 ucollections module: expected methods are missing

peterhinch, 2021-02-15:
`OrderedDict` and `dequeue` are present on Pyboard and ESP32 but absent on rp2.

Are these differences teething problems or do they represent deliberate design decisions?

tjvr, 2021-02-15:
Which version of the rp2 firmware are you using? I think there are some modules missing from the stable builds that are present in the later unstable ones.

peterhinch, 2021-02-16:
I was running
```
MicroPython v1.14 on 2021-02-09; Raspberry Pi Pico with RP2040
```
However I've now installed the latest build
```
MicroPython v1.14 on 2021-02-16; Raspberry Pi Pico with RP2040
```
and I'm getting the same result. Also other issues I've reported: **cmath** and **machine.Signal** missing. It would be good if the rp2 port supported the same modules as the Pyboard (subject to hardware differences).

dpgeorge, 2021-03-13:
Enabled in 8010b15968dece822aa5318888cb6d969f36ef3b

closed 2021-03-13 by dpgeorge