micropython/micropython #6913: Problems with /genhdr/pins_af_defs.h

Krasn4ck, 2021-02-16:
Hello again, I had this error a few days ago, but I solved it easily, editing the file with respect to the following macros, but I don't understand very well to what this error is due because every time I compile cleanly it appears again.
```
In file included from pin_static_af.h:31,
                 from qspi.c:33:
build-OPHYRA/genhdr/pins_af_defs.h:405:9: error: ISO C99 requires whitespace after the macro name [-Werror]
  405 | #define STATIC_AF_JTMS-SWDIO_NULL(pin_obj) ( \
      |         ^~~~~~~~~~~~~~
build-OPHYRA/genhdr/pins_af_defs.h:409:9: error: ISO C99 requires whitespace after the macro name [-Werror]
  409 | #define STATIC_AF_JTCK-SWCLK_NULL(pin_obj) ( \
      |         ^~~~~~~~~~~~~~
cc1.exe: all warnings being treated as errors
In file included from pin_static_af.h:31,
```
what I do is to edit the macros in this way:

```
#define STATIC AF_JTMS_SWDIO_NULL(pin_obj)

#define STATIC_AF_JTCK_SWCLK_NULL(pin_obj)
```

dpgeorge, 2021-02-16:
I assume this is a custom board?  What is in your pins.csv file, does it contain these JTAG pins?

Krasn4ck, 2021-02-17:
> I assume this is a custom board? What is in your pins.csv file, does it contain these JTAG pins?

Yes, is a custom board.
the contain of my pins.csv is:

```
PA0,PA0
PA1,PA1
PA2,PA2
PA3,PA3
PA4,PA4
PA5,PA5
PA6,PA6
PA7,PA7
PA8,PA8
PA9,PA9
PA10,PA10
PA11,PA11
PA12,PA12
PA13,PA13
PA14,PA14
PA15,PA15
PB0,PB0
PB1,PB1
PB2,PB2
PB3,PB3
PB4,PB4
PB5,PB5
PB6,PB6
PB7,PB7
PB8,PB8
PB9,PB9
PB10,PB10
PB11,PB11
PB12,PB12
PB13,PB13
PB14,PB14
PB15,PB15
PC0,PC0
PC1,PC1
PC2,PC2
PC3,PC3
PC4,PC4
PC5,PC5
PC6,PC6
PC7,PC7
PC8,PC8
PC9,PC9
PC10,PC10
PC11,PC11
PC12,PC12
PC13,PC13
PC14,PC14
PC15,PC15
PD0,PD0
PD1,PD1
PD2,PD2
PD3,PD3
PD4,PD4
PD5,PD5
PD6,PD6
PD7,PD7
PD8,PD8
PD9,PD9
PD10,PD10
PD11,PD11
PD12,PD12
PD13,PD13
PD14,PD14
PD15,PD15
PE0,PE0
PE1,PE1
PE2,PE2
PE3,PE3
PE4,PE4
PE5,PE5
PE6,PE6
PE7,PE7
PE8,PE8
PE9,PE9
PE10,PE10
PE11,PE11
PE12,PE12
PE13,PE13
PE14,PE14
PE15,PE15
PH0,PH0
PH1,PH1
LED_RED,PE0
LED_GREEN,PE1
LED_BLUE,PE2
SW,PC2
USB_DM,PA11
USB_DP,PA12

```

dpgeorge, 2021-02-17:
What MCU is it?  What af.csv file do you use?

Krasn4ck, 2021-02-17:
> What MCU is it? What af.csv file do you use?

The MCU is STM32F407VG

I used a modified version of stm32f405_af.csv provided by the board manufacturer called stm32f407_af.csv, I want to think that something is wrong with that file.

dpgeorge, 2021-02-17:
It probably has lines similar to this:
```
PortA,PA13,JTMS-SWDIO,,,,,,,,,,,,,,,EVENTOUT,
PortA,PA14,JTCK-SWCLK,,,,,,,,,,,,,,,EVENTOUT,
```
You need to replace them with:
```
PortA,PA13,JTMS/SWDIO,,,,,,,,,,,,,,,EVENTOUT,
PortA,PA14,JTCK/SWCLK,,,,,,,,,,,,,,,EVENTOUT,
```

Krasn4ck, 2021-02-17:
> It probably has lines similar to this:
> 
> ```
> PortA,PA13,JTMS-SWDIO,,,,,,,,,,,,,,,EVENTOUT,
> PortA,PA14,JTCK-SWCLK,,,,,,,,,,,,,,,EVENTOUT,
> ```
> 
> You need to replace them with:
> 
> ```
> PortA,PA13,JTMS/SWDIO,,,,,,,,,,,,,,,EVENTOUT,
> PortA,PA14,JTCK/SWCLK,,,,,,,,,,,,,,,EVENTOUT,
> ```

Thank You @dpgeorge I just tried and the error did not appear, I really appreciate your help with this.

closed 2021-02-17 by Krasn4ck