micropython/micropython #6918: rp2 asm_pio : no module named 'array'

fragmuffin, 2021-02-17:
Error when running demo code in [ports/rp2/README.md : PIO Blinky](https://github.com/micropython/micropython/blob/469345e7285128739e2934e7934e107ffda79fc1/ports/rp2/README.md#pio-blinky)

    MPY: soft reboot
    MicroPython v1.14 on 2021-02-03; Raspberry Pi Pico with RP2040
    Type "help()" for more information.
    >>> import rp2
    >>> def foo():
    ...     pass
    >>> rp2.asm_pio(set_init=rp2.PIO.OUT_LOW)(foo)
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
      File "rp2.py", line 228, in asm_pio
      File "rp2.py", line 36, in __init__
    ImportError: no module named 'array'

stinos, 2021-02-17:
Possible duplicate of https://github.com/micropython/micropython/issues/6837

lurch, 2021-02-17:
> MicroPython v1.14 on 2021-02-03; Raspberry Pi Pico with RP2040

https://www.raspberrypi.org/documentation/pico/getting-started/ has `rp2-pico-20210205-unstable-v1.14-8-g1f800cac3.uf2` available for download, which fixes this problem.

fragmuffin, 2021-02-17:
Confirmed fixed in `rp2-pico-20210217-unstable-v1.14-67-gcf6a01588.uf2`, thanks @stinos & @lurch

closed 2021-02-17 by fragmuffin