micropython/micropython #6921: same manufacturer collision i2c 0 vs i2c 1

srbpavel, 2021-02-18:
rpi pico
rp2-pico-20210217-unstable-v1.14-67-gcf6a01588.uf2
>>> os.uname()
(sysname='rp2', nodename='rp2', release='1.14.0', version='v1.14 on 2021-02-17 (GNU 10.2.0 MinSizeRel)', machine='Raspberry Pi Pico with RP2040')

testing two original adafruit bme280 and one clone, each original paired with different i2c it looks like two adafuit originals  interact. after one jumpered with GND to SDO it looks ok. my mistake?

INVALID:
INPUT:
i2c_0=I2C(0, scl=Pin(13), sda=Pin(12))
i2c_0_scan=i2c_0.scan()
print('i2c_0_scan: ', i2c_0_scan, [hex(x) for x in i2c_0_scan])

i2c_1=I2C(1, scl=Pin(19), sda=Pin(18))
i2c_1_scan=i2c_1.scan()

OUTPUT:
i2c_0_scan:  [118] ['0x76'] #CLONE ONLY
i2c_1_scan:  [] []

VALID [hw setting: i2c_1 bme280 GND -> SD0  / 0x77 -> 0x76]
INPUT:
i2c_0=I2C(0, scl=Pin(13), sda=Pin(12))
i2c_0_scan=i2c_0.scan()
print('i2c_0_scan: ', i2c_0_scan, [hex(x) for x in i2c_0_scan])

i2c_1=I2C(1, scl=Pin(19), sda=Pin(18))
i2c_1_scan=i2c_1.scan()
print('i2c_1_scan: ', i2c_1_scan, [hex(x) for x in i2c_1_scan])

OUTPUT:
i2c_0_scan:  [118, 119] ['0x76', '0x77'] #CLONE + ORIGINAL
i2c_1_scan:  [118] ['0x76'] #ORIGINAL JUMPERED

pavel

dpgeorge, 2021-03-11:
I'm not sure what the issue or bug report is here.  Can you explain in more detail and give steps to reproduce?  (And remember that you may need 4.7k pull up resistors on I2C lines.)

srbpavel, 2021-03-18:
hallo, ok 
- if i understand that correctly i2c bus 0 a i2c bus 1 are independent. 
so let's say on my esp32 mh et, i use

i2c_0=I2C(0, scl=Pin(25), sda=Pin(32)) 
i2c_1=I2C(1, scl=Pin(22), sda=Pin(21)) 

each bus provided with let's say bme280 (sda+scl each with pull up resistor)
the result via scan()

BME sensor [bus: 0] [118]
BME sensor [bus: 1] [118]

same settings with RPI PICO is not possible, as there is collision between buses(the i2c_1 do not detect device).

to solve this, i had to jumper one BME280 to change address. then both sensors are visible, each on it's own i2c bus.

thank you, pavel

Fede-VR73, 2022-07-19:
If this is still an issue, i would like to report that I2C is working perfectly for me with 2 identical devices with same hw addr. on different I2C buses (VCC-GND RP2040 with fw 8.0.0-alpha.1-54-g145cce93d for WeAct Pico)  with this code : 

`
from board import GP14,GP15,GP16,GP17
from busio import I2C
from adafruit_ssd1306 import SSD1306_I2C

i2c1 = I2C(GP17, GP16, frequency=1_000_000)
i2c0 = I2C(GP15, GP14, frequency=1_000_000)
display1 = SSD1306_I2C(128, 64, i2c0, addr=0x3C)
display2 = SSD1306_I2C(128, 64, i2c1, addr=0x3C)
display1.contrast(10)
display2.contrast(10)
display1.fill(0)
display2.fill(0)
display1.text("YD-RP2040", 10, 2, 1, size=2)
display1.text(str(memfree * 0.0008) + " Kb free", 1, 56, 1, size=1)
display2.text("Midi App", 10, 0, 1, size=2)
display1.show()
display2.show()
`
as seen here : 

[image]

F.

robert-hh, 2022-07-19:
There is no GPIO25 or GPIO32 accessible at a RPi Pico, and GPIO22 and GPIO21 cannot be used for I2C. You probably used board pin numbers instead of GPIO numbers in the i2c instantiation. Which is a common misunderstanding. In the first post, the numbers look right.