micropython/micropython #6922: Tests failing on current code.

kadamski, 2021-02-18:
Hi,

When I'm running tests for micropython, I'm getting failures. In case of a unix port, I get:

`2 tests failed: urandom_basic import_pkg7`

The import_pkg7 test also prints:

`Traceback (most recent call last):           
  File "/home/k/BUILD/micropython/tests/import/import_pkg7.py", line 2, in <module>
    import pkg7.subpkg1.subpkg2.mod3                    
  File "/home/k/BUILD/micropython/tests/import/pkg7/subpkg1/subpkg2/mod3.py", line 9, in <module>
    from .... import mod1                               
ImportError: attempted relative import beyond top-level package`

extmod/urandom_basic.py fails silently. The http://docs.micropython.org/en/latest/develop/index.html does not provide any info about debuging the test cases so I would be glad if anyone could help me with that. I would happily add the information to the documentation, once I get into that.

The question is whether it is failing because my environment is broken or its the code that has problems. I'm on `v1.14-67-gcf6a01588` and cpython 3.9.1.

stinos, 2021-02-18:
Current CI builds succeed and also no problem here on different machines so could be environment-related. What's the output of `run-tests --print-failures` ?

kadamski, 2021-02-18:
hmm.. for the urandom_basic, it seems to be caused by:

```
# check that it throws an error for zero bits
try:
    random.getrandbits(0)
except ValueError:
    print("ValueError")
```

While on CPython 3.8.6 I do get a ValueError when running `random.getrandbits(0)`, on CPython 3.9.1 I get:
```
Python 3.9.1 (default, Jan 20 2021, 00:00:00) 
[GCC 10.2.1 20201125 (Red Hat 10.2.1-9)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import random
>>> random.getrandbits(0)
```

So... well, how should such a problem be fixed?

kadamski, 2021-02-18:
@stinos oh, the --print-failures is useful! 

```
$ ./run-tests --print-failures

FAILURE /home/k/BUILD/micropython/tests/results/extmod_urandom_basic.py
--- /home/k/BUILD/micropython/tests/results/extmod_urandom_basic.py.exp 2021-02-18 11:06:17.268730908 +0100
+++ /home/k/BUILD/micropython/tests/results/extmod_urandom_basic.py.out 2021-02-18 11:06:17.268730908 +0100
@@ -1,2 +1,3 @@
 True
 True
+ValueError

FAILURE /home/k/BUILD/micropython/tests/results/import_import_pkg7.py
--- /home/k/BUILD/micropython/tests/results/import_import_pkg7.py.exp   2021-02-18 11:00:36.259398333 +0100
+++ /home/k/BUILD/micropython/tests/results/import_import_pkg7.py.out   2021-02-18 11:00:36.259398333 +0100
@@ -1 +1,8 @@
-CPYTHON3 CRASH
\ Brak znaku nowej linii na końcu pliku
+pkg __name__: pkg7
+pkg __name__: pkg7.subpkg1
+pkg __name__: pkg7.subpkg1.subpkg2
+mod1
+mod2
+mod1.foo
+mod2.bar
+ValueError
```

This confirms the reason for the urandom_basic i described above. For the pkg7, it also is caused by the fact that CPython on my computer is not behaving as expected. It does work on python 3.8.6 but on a system with 3.9.1, I'm getting:

```
$ python3 import/import_pkg7.py 
pkg __name__: pkg7
pkg __name__: pkg7.subpkg1
pkg __name__: pkg7.subpkg1.subpkg2
mod1
mod2
mod1.foo
mod2.bar
Traceback (most recent call last):
  File "/home/k/BUILD/micropython/tests/import/import_pkg7.py", line 2, in <module>
    import pkg7.subpkg1.subpkg2.mod3
  File "/home/k/BUILD/micropython/tests/import/pkg7/subpkg1/subpkg2/mod3.py", line 9, in <module>
    from .... import mod1
ImportError:  attempted relative import beyond top-level package
```

stinos, 2021-02-18:
> So... well, how should such a problem be fixed?

Are you asking in general, or this particular case? The former: the Python version MicroPython currently aims to support is roughly 3.7 (I think, and definitely not everything is implemented) so 3.9 is a bit too recent i.e. solution is to not use it for running tests. The latter: not sure in this particular case.

kadamski, 2021-02-18:
> Are you asking in general, or this particular case? 

I was asking in general. I assumed it is not the first time that the tests stop working with later CPython version and the project has some way of solving this issue. If not, maybe its time to  think about it.

Also maybe the CPython version one should use for testing should be documented somewhere (or maybe it is?) ?

dpgeorge, 2021-02-18:
CPython does evolve and so these tests that fail now have different behaviour to pre-3.9 CPython.

Usually this is dealt with by providing a .exp file to test against, instead of testing against CPython.

See also `tests/basics/python34.py` and `tests/basics/python36.py`

kadamski, 2021-02-18:
@dpgeorge should I add the .exp files for those tests, then? Or should I close this issue for now?

dpgeorge, 2021-02-18:
I would be good to fix these issues, but it's not a high priority.

I also see warnings on CPython 3.9 running some other tests like `basics/is_isnot.py`.

For the `random` test, I'd actually like to support `getrandbits(0)` in uPy, to match CPy 3.9.

For the pkg7 test, it'd be good to track down the CPy change (is there a PEP for it, or is it a bug?) and make uPy also behave the same (it's simply changing a ValueError to ImportError).

dpgeorge, 2021-05-30:
`getrandbits(0)` is implemented/fixed in c3199f56494429ba44f7e722597945dc91f3589c

This issue is also a duplicate of #6750, so closing.

closed 2021-05-30 by dpgeorge