micropython/micropython #6924: unix port socket sendto - "TypeError: object with buffer protocol required"

ozimchuk, 2021-02-18:
MicroPython v1.14-68-gd4b45898f on 2021-02-18; linux version

The following code produces "TypeError: object with buffer protocol required":

> from socket import *
> s = socket(AF_INET, SOCK_DGRAM)
> s.sendto(b'\x00', ('localhost', 5000))

dpgeorge, 2021-02-18:
The second argument to sendto must be an address as returned by getaddrinfo.  The following may work:
```
s.sendto(b'\x00', getaddrinfo('localhost', 5000)[0][-1])
```

ozimchuk, 2021-02-18:
> The second argument to sendto must be an address as returned by getaddrinfo. The following may work:
> 
> ```
> s.sendto(b'\x00', getaddrinfo('localhost', 5000)[0][-1])
> ```

This worked for me.
Thank you!

closed 2021-02-18 by ozimchuk