micropython/micropython #6926: micropython : port/esp32 ==> fatal error: soc/spinlock.h: No such file or directory

picpic020960, 2021-02-18:
Bonjour ,
try to build micropython for ESP32
install esp-idf and git clone micropython. fine.
(linux UBUNTU 16 no change in make menuconfig )
when 'make' the issue : 

In file included from /home/xxxx/esp-idf/components/freertos/include/freertos/portable.h:52,
                 from /home/xxxx/esp-idf/components/freertos/include/freertos/FreeRTOS.h:64,
                 from /home/xxxx/micropython/ports/esp32/mpconfigport.h:234,
                 from /home/xxxx/micropython/py/mpconfig.h:62,
                 from /home/xxxx/micropython/py/mpstate.h:31,
                 from /home/xxxx/micropython/py/nlrx64.c:27:
/home/xxxx/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:49:10: fatal error: soc/spinlock.h: No such file or directory
 #include "soc/spinlock.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.

many times ...

some action for control ?
tricks or workarounds  ?
Thank's for help and regards

picpic020960, 2021-02-18:
``

picpic020960, 2021-02-18:
``

picpic020960, 2021-02-18:
oooups

wogum, 2021-02-18:
There may be something wrong with the new ESP32 Makefiles. There are no new daily builds from 1.14-9 (210206).

dpgeorge, 2021-02-18:
You'll need to use a supported IDF version to build, one of: v4.0.2, v4.1.1, v4.2.

dpgeorge, 2021-02-19:
This should now be fixed by 143372ab5e41a609b5b7d8bb60af1a5fd478e7ea, you should be able to build with the latest IDF.

closed 2021-02-19 by dpgeorge

picpic020960, 2021-02-19:
work fine !
Thank's