micropython/micropython #6927: stm32: powerctrl_check_enter_bootloader() bypass

peter-conalgo, 2021-02-18:
My STM32L496 based system has its own bootloader, and so the logic in `powerctrl_check_enter_bootloader()` is inappropriate, and `branch_to_bootloader()` cannot work and is not wanted.

However, there is no `#define` to disable this code, and it runs very early.

I'm happy to provide a PR, but what should I call the `cpp` variable to disable this code? Thanks!

dpgeorge, 2021-03-10:
I'm happy to make this configurable, but as the code currently stands it should not affect your board: if you don't call `machine.bootloader()` then it should never try to enter the bootloader.  Do you see adverse behaviour even without calling `machine.bootloader()`?

peter-conalgo, 2021-03-10:
It calls `branch_to_bootloader()` which cannot work on my board, depending on reset reason.

dpgeorge, 2021-03-10:
> Depending on reset reason, it calls `branch_to_bootloader()`

It shouldn't be calling that function, because:
- on power on the reset reason does not match the if-logic
- after that the `BL_STATE[1]` variable is set to 1 so does not match the if-logic

Anyway, does #7010 work for your case?

peter-conalgo, 2021-03-10:
Yes, that PR would cover my needs nicely.

dpgeorge, 2021-03-11:
Made configurable in c33c749f64fc664e46529885d84e499ea2c3d461

closed 2021-03-11 by dpgeorge