micropython/micropython #6936: MicroPython external C modules : example KO ?

picpic020960, 2021-02-19:
Bonjour 
(compile micropython UNIX port OK)
then I try compile for UNIX port the example C module as show in the doc/develpo/cmodules.rst : 

make USER_C_MODULES=../../examples/usercmodule CFLAGS_EXTRA=-DMODULE_CEXAMPLE_ENABLED=1 all

display two liens 'Including User C Module from ...' OK
but then errors :

In file included from ../../py/mpstate.h:35:0,
                 from ../../py/runtime.h:29,
                 from ../../examples/usercmodule/cexample/examplemodule.c:2:
../../examples/usercmodule/cexample/examplemodule.c:22:50: error: ‘MP_QSTR_cexample’ undeclared here (not in a function)
     { MP_ROM_QSTR(MP_QSTR___name__), MP_ROM_QSTR(MP_QSTR_cexample) },
                                                  ^
../../py/obj.h:96:56: note: in definition of macro ‘MP_OBJ_NEW_QSTR’
 #define MP_OBJ_NEW_QSTR(qst) ((mp_obj_t)((((mp_uint_t)(qst)) << 3) | 2))
                                                        ^
../../examples/usercmodule/cexample/examplemodule.c:22:38: note: in expansion of macro ‘MP_ROM_QSTR’
     { MP_ROM_QSTR(MP_QSTR___name__), MP_ROM_QSTR(MP_QSTR_cexample) },
                                      ^
../../examples/usercmodule/cexample/examplemodule.c:22:5: error: missing initializer for field ‘value’ of ‘mp_rom_map_elem_t {aka const struct _mp_rom_map_elem_t}’ [-Werror=missing-field-initializers]
     { MP_ROM_QSTR(MP_QSTR___name__), MP_ROM_QSTR(MP_QSTR_cexample) },
     ^
In file included from ../../py/mpstate.h:35:0,
                 from ../../py/runtime.h:29,
                 from ../../examples/usercmodule/cexample/examplemodule.c:2:
../../py/obj.h:405:18: note: ‘value’ declared here
     mp_rom_obj_t value;
                  ^
In file included from ../../py/mpstate.h:35:0,
                 from ../../py/runtime.h:29,
                 from ../../examples/usercmodule/cexample/examplemodule.c:2:
../../examples/usercmodule/cexample/examplemodule.c:23:19: error: ‘MP_QSTR_add_ints’ undeclared here (not in a function)
     { MP_ROM_QSTR(MP_QSTR_add_ints), MP_ROM_PTR(&example_add_ints_obj) },
                   ^
../../py/obj.h:96:56: note: in definition of macro ‘MP_OBJ_NEW_QSTR’
 #define MP_OBJ_NEW_QSTR(qst) ((mp_obj_t)((((mp_uint_t)(qst)) << 3) | 2))
                                                        ^
../../examples/usercmodule/cexample/examplemodule.c:23:7: note: in expansion of macro ‘MP_ROM_QSTR’
     { MP_ROM_QSTR(MP_QSTR_add_ints), MP_ROM_PTR(&example_add_ints_obj) },
       ^
mistake ?
workaround ?
Thank's for help and
Regards
 
config : linux UBUNTU 16.4

picpic020960, 2021-02-19:
thank's a lot ! it's fine.
now the same for esp32.
then st7789 as devbis/st7789_mpy
after latter tests no success but continue to probe.
Thank's

closed 2021-02-19 by picpic020960

ali299792, 2023-08-17:
Hi, how did you resolve this? I've just recursively cloned the micropython repo and am running this from ports/unix and get the same error as you

jimmo, 2023-08-21:
@ali299792 Please provide more information (i.e. build output). It's working for me.

ali299792, 2023-08-21:
> @ali299792 Please provide more information (i.e. build output). It's working for me.

Not sure what was wrong with the initial clone, but when I recursively recloned it in a different folder again I could build successfully (previous clone's build output iirc was pretty much identical to the one posted above).
Sorry for the trouble.