micropython/micropython #6937: Powerpc port versions 1.13 and 1.14 broken on Chiselwatt

carlosedp, 2021-02-19:
I've built latest Micropython versions (1.13 and 1.14) to run on Chiselwatt core but serial is currently broken.

Running on Verilator Chiselwatt, the output is:

```
root@verilator:/work# ./chiselwatt
BITS error 434 384
                  BITS error 434 384
                                    BITS error 434 336
                                                      BITS error 434 384
                                                                        BITS error 434 384
                                                                                          BITS error 434 336
                                                                                                            BITS error 434 384
                                                                                                                              BITS error 434 384
                                                                                                                                                BITS error 434 384
    BITS error 434 288
                      BITS error 434 384
                                        BITS error 434 336
                                                          BITS error 434 336
                                                                            BITS error 434 384
```

I've reverted the changes to serial, and it worked fine on 1.13 and 1.14 (following patch):

```diff
diff --git a/ports/powerpc/Makefile b/ports/powerpc/Makefile
index 1f5ec80d4..cd8215465 100644
--- a/ports/powerpc/Makefile
+++ b/ports/powerpc/Makefile
@@ -33,7 +33,9 @@ LIBS =
 
 SRC_C = \
 	main.c \
-	uart_$(UART).c \
+	uart_core.c \
+	uart_potato.c \
+	uart_lpc_serial.c \
 	lib/utils/printf.c \
 	lib/utils/stdout_helpers.c \
 	lib/utils/pyexec.c \
diff --git a/ports/powerpc/uart_lpc_serial.c b/ports/powerpc/uart_lpc_serial.c
index 760615041..51a55cb1e 100644
--- a/ports/powerpc/uart_lpc_serial.c
+++ b/ports/powerpc/uart_lpc_serial.c
@@ -96,24 +96,16 @@ static int lpc_uart_rx_empty(void) {
     return !(lpc_uart_reg_read(REG_LSR) & LSR_DR);
 }
 
-void uart_init_ppc(void) {
+void lpc_uart_init(void) {
     lpc_uart_base = LPC_UART_BASE;
 }
 
-int mp_hal_stdin_rx_chr(void) {
-    while (lpc_uart_rx_empty()) {
-        ;
-    }
+char lpc_uart_read(void) {
+    while (lpc_uart_rx_empty()) ;
     return lpc_uart_reg_read(REG_THR);
 }
 
-
-void mp_hal_stdout_tx_strn(const char *str, mp_uint_t len) {
-    int i;
-    for (i = 0; i < len; i++) {
-        while (lpc_uart_tx_full()) {
-            ;
-        }
-        lpc_uart_reg_write(REG_RBR, str[i]);
-    }
+void lpc_uart_write(char c) {
+    while (lpc_uart_tx_full());
+    lpc_uart_reg_write(REG_RBR, c);
 }
diff --git a/ports/powerpc/uart_potato.c b/ports/powerpc/uart_potato.c
index 29f21934d..18f89d463 100644
--- a/ports/powerpc/uart_potato.c
+++ b/ports/powerpc/uart_potato.c
@@ -34,12 +34,10 @@
 #include <stdbool.h>
 #include "py/mpconfig.h"
 
-#define SYSCON_BASE     0xc0000000  /* System control regs */
-#define SYS_REG_CLKINFO                 0x20
-
+#define PROC_FREQ                           50000000
 #define UART_FREQ                           115200
 #define POTATO_UART_BASE                    0xc0002000
-static uint64_t potato_uart_base;
+uint64_t potato_uart_base;
 
 #define POTATO_CONSOLE_TX                   0x00
 #define POTATO_CONSOLE_RX                   0x08
@@ -62,7 +60,7 @@ static uint64_t potato_uart_reg_read(int offset) {
     return val;
 }
 
-static void potato_uart_reg_write(int offset, uint64_t val) {
+void potato_uart_reg_write(int offset, uint64_t val) {
     uint64_t addr;
 
     addr = potato_uart_base + offset;
@@ -98,35 +96,26 @@ static unsigned long potato_uart_divisor(unsigned long proc_freq, unsigned long
     return proc_freq / (uart_freq * 16) - 1;
 }
 
-void uart_init_ppc(void) {
-    uint64_t proc_freq;
-
+void potato_uart_init(void) {
     potato_uart_base = POTATO_UART_BASE;
-    proc_freq = *(volatile uint64_t *)(SYSCON_BASE + SYS_REG_CLKINFO);
-    potato_uart_reg_write(POTATO_CONSOLE_CLOCK_DIV, potato_uart_divisor(proc_freq, UART_FREQ));
+    potato_uart_reg_write(POTATO_CONSOLE_CLOCK_DIV, potato_uart_divisor(PROC_FREQ, UART_FREQ));
 
 }
 
-int mp_hal_stdin_rx_chr(void) {
+char potato_uart_read(void) {
     uint64_t val;
 
-    while (potato_uart_rx_empty()) {
-        ;
-    }
+    while (potato_uart_rx_empty());
     val = potato_uart_reg_read(POTATO_CONSOLE_RX);
 
     return (char)(val & 0x000000ff);
 }
 
-void mp_hal_stdout_tx_strn(const char *str, mp_uint_t len) {
-    int i;
+void potato_uart_write(char c) {
+    uint64_t val;
 
-    for (i = 0; i < len; i++) {
-        uint64_t val = str[i];
+    val = c;
 
-        while (potato_uart_tx_full()) {
-            ;
-        }
-        potato_uart_reg_write(POTATO_CONSOLE_TX, val);
-    }
+    while (potato_uart_tx_full());
+    potato_uart_reg_write(POTATO_CONSOLE_TX, val);
 }
```

With this, 1.14 runs fine:

```
root@verilator:/work# ./chiselwatt
MicroPython v1.14-72-gd28dbcd6c-dirty on 2021-02-19; bare-metal with POWERPC
Type "help()" for more information.
```

Cc. @mikey @antonblanchard

mikey, 2021-02-21:
Does upstream micropython work if you do `make UART=potato` @carlosedp ?

mikey, 2021-02-22:
@carlosedp Thanks for the report. 

The issue is chiselwatt doesn't have syscon which is used to determine the processor speed. Either we add syscon to chiselwatt or we force micropython to use a specific processor freq. For example, this patch fixes the issue for me (ping @shenki) 

```diff --git a/ports/powerpc/uart_potato.c b/ports/powerpc/uart_potato.c
index 29f21934dd..bbf66c74f3 100644
--- a/ports/powerpc/uart_potato.c
+++ b/ports/powerpc/uart_potato.c
@@ -102,7 +102,7 @@ void uart_init_ppc(void) {
     uint64_t proc_freq;
 
     potato_uart_base = POTATO_UART_BASE;
-    proc_freq = *(volatile uint64_t *)(SYSCON_BASE + SYS_REG_CLKINFO);
+    proc_freq = 50000000;
     potato_uart_reg_write(POTATO_CONSOLE_CLOCK_DIV, potato_uart_divisor(proc_freq, UART_FREQ));
 
 }
```

shenki, 2021-02-22:
Adding a syscon is a good idea. Here's a really simple implemntation for chiselwatt:

```
diff --git a/src/main/scala/LoadStore.scala b/src/main/scala/LoadStore.scala
index 96a05ad97420..c472ba3e7204 100644
--- a/src/main/scala/LoadStore.scala
+++ b/src/main/scala/LoadStore.scala
@@ -150,6 +150,13 @@ class LoadStore(val bits: Int, val words: Int) extends Module {
         data := d.zeroExtend(length)
       }
 
+      /* Syscon */
+      when (addr(31, 8) === "hc00000".U) {
+        when (addr(7, 0) === "h20".U) {
+          data := "h50000000".U
+        }
+      }
+
       /* UART */
       when (addr(31, 8) === "hc00020".U) {
         when (addr(7, 0) === "h08".U) {
```

However, when testing micropython on it I still see the same BITS error.

shenki, 2021-02-22:
Of course, 0x50000000 is not the same as 50000000. If we use `h2faf080` in place of `h50000000` the micropython runs fine.

shenki, 2021-03-03:
@carlosedp Did you confirm that microypthon now works on chiselwatt for you?

carlosedp, 2021-03-03:
Ah yes, latest version of Chiselwatt worked perfectly.
Thanks @shenki!

closed 2021-03-03 by carlosedp