micropython/micropython #6947: Ble, read more 20 bytes esp32 wroom32d. MTU not work

rdaguerre, 2021-02-21:
Hello, Sorry for my English.
I am making a connection to a blutooth BMS and I need to send a command to receive via notify that I have the data.
The thing is, I only get 20 bytes.
I don't know what else to change. I am not very knowledgeable in esp32 but I have been trying for many days to make it connect and now I have succeeded but I do not get out of these 20 reception bytes.
If anyone knows I would appreciate it.
regards

jimmo, 2021-02-22:
You need to do an MTU exchange to go past the default MTU.

```
_BMS_MTU = const(128)

ble.gattc_exchange_mtu(_BMS_MTU)
```

Then you will get a `_IRQ_MTU_EXCHANGED` event when this succeeds.

See https://docs.micropython.org/en/latest/library/ubluetooth.html#ubluetooth.BLE.gattc_exchange_mtu

dpgeorge, 2021-03-11:
Hopefully the above suggestion helped, so closing the issue.  Feel free to reopen if there is a problem.

closed 2021-03-11 by dpgeorge

jakub-vesely, 2021-09-27:
This issue has been closed but I would like to provide my experience. I faced a similar issue. I was not able to receive more than 20 bytes. Setting MTU did not help. By reverse engineering the code, I found out that to define MP_BLUETOOTH_DEFAULT_ATTR_LEN is necessary, otherwise it is set to 20. Finally, I'm able to receive bigger data packages.