micropython/micropython #6949: There is no ENC28J60 driver in Micropython

wzab, 2021-02-22:
The [ENC28J60](https://www.microchip.com/wwwproducts/en/en022889) chip is a cheap solution suitable for providing the Ethernet connectivity for simple microcontroller devices. 

It would be beneficial to be able to use it with cheap MicroPython-capable boards. Unfortunately, there is no open-source MicroPython driver for that chip. There is a documentation for MicroPython driver at https://software.open-dev.ru/docs/online/micropython/library/network.ENC28J60.html , but the driver itself is not available (the firm says that it is based on a part of the commercial products, and its source can't be published).
On the other hand, there are a lot of ENC28J60 drivers available in C (e.g.,  [for AVR](https://github.com/bprayudha/avr-enc28j60), [from Oryx](https://www.oryx-embedded.com/doc/enc28j60__driver_8c_source.html) ), or for [Arduino](https://github.com/njh/EtherCard). Writing a driver for MicroPython should be relatively easy if there is any existing network driver skeleton or template.

wzab, 2021-02-22:
I've created a bounty for that issue. If there are more interested users, maybe we can sponsor development of the driver? https://www.bountysource.com/issues/96668071-there-is-no-enc28j60-driver-in-micropython

enesbcs, 2021-02-24:
Honestly ENC28J60 is a very unreliable piece of hardware. Wiznet W5500 works stable with STM32.

rippiedoos, 2021-03-03:
The ENC28J60 is abundantly available because it was and is so cheap. The W5500 has come done in price but isn't that cheap and as abundantly available as the ENC28J60, because of the former pricepoint. 

I have some of these ENC28J60 laying around and would love to connect them to a Pi Pico.

dave2, 2021-04-08:
I've used the ENC28J60 in anger on some boards, and the silicon errata are long and frustrating. The later chips (ENC424J600 and ENC624J600) were a lot better for actually being reliable with fewer major issues in the errata. The problem is, both of those are about the same as a W5500 on price and not as available.

raleighlittles, 2022-07-25:
It looks like someone has added support already: https://github.com/przemobe/micropy-ENC28J60

closed 2024-07-04 by mattytrentini